ngardiner/TWCManager #250: Stop charging = loosing the link

Dustovsky, 2021-03-18:
Hello :-)

My TWCManager seems to be working smoothly with my Tesla MX, but the same setup does not work with our Hyundai eKona:
- start charging and adapting the charging power/current works
- stop charging via "send stop command" does not work (maybe because the eKona doesn't know/understand this command)
- stop charging via "stop responding to slave" works initially, the car stops charging after some seconds/minutes.

BUT:
As soon as the charging stops, TWCManager looses the link to the TWC/slave and is unable to regain it again:

15:58:56 TWCManager 1 32.00 amp slave TWC 5883 is ready to link.  Sign: 74

This goes on forever...

And: even a reboot of the raspy doesn't help. It will never relink again...

I'm I missunderstanding the stop method? Or is there anything else I have to configure to bring TWCManager to relink again?

Cheers Dusty

ngardiner, 2021-03-26:
Hi Dusty,

"Send Stop Command" only works on the latest firmware, so it does sound like your TWC is probably not on that firmware version. Do you see VIN numbers of connected Tesla vehicles on your TWC?

The Pi should relink the next time it wants to charge, so issuing a "Charge Now" should re-link for example, as would Scheduled Charging or if you have any particular charge rate set for Non-Scheduled Charging. Green Energy being over the minimum should also start communicating again. 

I'm not quite sure what the best way to make the eKona stop charging would be, but perhaps we can experiment a little in your environment. Keep in mind that we are totally at the mercy of both the Tesla TWC command set and the vehicle itself, for Tesla we can instruct the vehicle to stop charging via the API, and with newer firmware TWCs we can instruct it to stop charging, but if you have neither of these we may not have a good option available unless there's some way to get a message to the eKona to stop charging.

ngardiner, 2021-04-18:
Hi Dusty,

Having thought this over some more, I can see the issue with the behaviour here and it isn't right.

Stop communicating with slaves should stop communicating for the ~30 seconds that it takes to stop the slave from charging, then it should start communicating again. Otherwise, we never hear from the slave again, and TWCManager will not operate correctly.

The problem was that we stopped responding and never made the decision that we should start responding again. I have just committed an update cecc40c to the development branch which should fix this, and should re-enable slave communication.

Dustovsky, 2021-04-19:
@ngardiner thanks alot :-)
Hope to be able to try it out soon with our eKona & I'll keep you adviced

closed 2022-02-05 by ngardiner

FlorianZ, 2022-04-26:
@ngardiner do you remember the details of this issue and the fix?

I have a non-Tesla car that is very eager to charge. It starts as soon as it is plugged in, and the only way for TWCManager to get it to stop is to cease communication with the slaves. After 60 seconds, the manager resumes communication and the car starts charging again. It then forever ping-pongs between these two states, repeatedly stopping and restarting charging.

I was wondering if I could remove the timer that was added with commit https://github.com/ngardiner/TWCManager/commit/cecc40c7b1ec2dd8187a84f572d2f0383cffbe28, and instead resume communication only when amps are offered up again. I must be missing something, though.

Do the slaves enter a non-recoverable state after some time of not receiving communication from the master?