ngardiner/TWCManager #253: New interface

magic7music, 2021-03-22:
Hi Guys,
great job until now.
I have the v1.2.1 running in Berlin without any problems.
The new Webinterface looks realy fancy.

Now to my question: I have a 3phase smart meter, running on a raspi calles "smartpi" made by the company Enerserve in Germany.
It looks a little bit like the SmartMe.

I took a look at the SmartMe interface, and I think that with a little bit of adjustment, it could work out.

The Raspi has an webinterface that can give out the used power as an API stream via http.

To recall the values you just type: http://192.168.???.???:1080/api/all/power/now  

where the output looks like: {"serial":"smartpi170512345","name":"House","lat":52.3667,"lng":9.7167,"time":"2021-03-22 19:36:38","softwareversion":"","ipaddress":"192.168.1.3","datasets":[{"time":"2021-03-22 19:36:38","phases":[{"phase":1,"name":"phase 1","values":[{"type":"power","unity":"W","info":"","data":168.92613}]},{"phase":2,"name":"phase 2","values":[{"type":"power","unity":"W","info":"","data":212.23642}]},{"phase":3,"name":"phase 3","values":[{"type":"power","unity":"W","info":"","data":91.89515}]},{"phase":4,"name":"phase 4","values":[{"type":"power","unity":"W","info":"","data":0}]}]}]}

If there is Solarpower, it will show a negativ value.
all 3 phases has to be added together to get the complete power that is avaliabe. the value will be negativ.
I have a 2,6kW Solaredge SE2200 already in use and another 1,2kW with a no name converter (Mastervolt).

Is it possible to help me with this problem.
If you could implement this, the TWCManager could read the total used power in my home and adjust the exact amounth of power given to the wallbox.

It would be great to hear from you.
Cheers

Koenraad from Berlin.

ngardiner, 2021-04-18:
Hi Koenraad, I've added a SmartPi module to the current development release (main branch) for testing. Could you please enable and configure it per the documentation and let me know how it does?

magic7music, 2021-04-18:
Hi Nathan, 
Thanks for that. I’ll download it tomorrow and let you know if it works. If there are any bugs, I’ll try to solve them and give you an update.

Great 👍 

Koen

Koenraad Verleyen
Mahonienweg 7
12437
Berlin

Send by mobile app on gsm

magic7music, 2021-04-19:
Hi,

there seems to be an error in the new SmartPi modul.

**Debug: with level 9**
pi@TWC-Manager:~/TWCManager $ python3 -m TWCManager
10:08:56 TWCMaster  14 Registered module ConsoleLogging
10:08:56 TWCMaster  14 Released module FileLogging
10:08:56 TWCMaster  14 Released module SentryLogging
10:08:56 TWCMaster  14 Released module CSVLogging
10:08:56 TWCMaster  14 Released module MySQLLogging
10:08:56 TWCMaster  14 Released module SQLiteLogging
10:08:56 TWCMaster  14 Registered module TWCProtocol
10:08:56 TWCMaster  14 Released module Dummy
10:08:56 TWCMaster  14 Registered module RS485
10:08:56 TWCMaster  14 Released module TCP
10:08:56 TWCMaster  14 Registered module Policy
10:08:56 TWCMaster  14 Registered module TeslaAPI
10:08:56 TWCMaster  14 Released module WebIPCControl
10:08:57 HTTPContro 20 Serving at port: 8080
10:08:57 TWCMaster  14 Registered module HTTPControl
10:08:57 TWCMaster  14 Released module MQTTControl
10:08:57 TWCManager 40 ImportError: No module named 'lib.TWCManager.Control.OCPPControl', when importing Control.OCPPControl, not using Control.OCPPControl
10:08:57 TWCMaster  14 Released module Efergy
10:08:57 TWCMaster  14 Released module Enphase
10:08:57 TWCMaster  14 Released module Fronius
10:08:57 TWCMaster  14 Released module HASS
10:08:57 TWCMaster  14 Released module Kostal
10:08:57 TWCMaster  14 Released module OpenHab
10:08:57 TWCMaster  14 Released module SmartMe
10:08:57 TWCMaster  14 Registered module SmartPi
10:08:57 TWCMaster  14 Registered module SolarEdge
10:08:57 TWCMaster  14 Released module SolarLog
10:08:57 TWCMaster  14 Released module TeslaPowerwall2
10:08:57 TWCMaster  14 Released module TED
10:08:57 TWCMaster  14 Released module Volkszahler
10:08:57 TWCMaster  14 Released module HASSStatus
10:08:57 TWCMaster  14 Released module MQTTStatus
10:08:57 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
10:08:57 TWCMaster  13 Send master linkready1
10:08:57 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:57 TWCMaster  13 Send master linkready1
10:08:57 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:57 TWCMaster  13 Send master linkready1
10:08:57 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:57 TWCMaster  13 Send master linkready1
10:08:57 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:57 TWCMaster  13 Send master linkready1
10:08:57 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
10:08:58 TWCMaster  13 Send master linkready2
10:08:58 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:58 TWCMaster  13 Send master linkready2
10:08:58 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:58 TWCMaster  13 Send master linkready2
10:08:58 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:58 TWCManager 12 Rx@: () FD E2 61 60 8F 0C 80 00 00 00 00 00 00 00 00 BE
10:08:58 TWCManager 20 32.00 amp slave TWC 6160 is ready to link.  Sign: 8F
10:08:58 TWCManager 20 Set slave TWC 6160 protocolVersion to 2, minAmpsTWCSupports to 6.
10:08:58 RS485      12 Tx@: C0 FB E0 77 77 61 60 00 00 00 00 00 00 00 00 00 8F C0
10:08:58 TWCMaster  13 Send master linkready2
10:08:58 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:58 TWCManager 9 Ignoring byte FC between messages.
10:08:58 TWCMaster  13 Send master linkready2
10:08:58 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
10:08:58 TWCManager 12 Rx@: (FC) FD E0 61 60 77 77 00 02 58 00 00 00 00 00 00 E9
10:08:58 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
10:08:58 RS485      12 Tx@: C0 FB EB 77 77 61 60 00 00 00 00 00 00 00 00 9A C0
10:08:58 Policy     13 Policy conditions were not matched.
10:08:58 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
10:08:58 Policy     13 Policy conditions were not matched.
[... 5 lines cut ...]
10:08:58 TWCMaster  10 Number of cars charging now: 0
10:08:58 SmartPi    10 Fetching SmartPi EMS sensor values
10:08:58 TWCSlave   10 desiredAmpsOffered: 0 < minAmpsToOffer: 6 (flexAmps: 0)
10:08:58 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0
10:08:58 TWCSlave   10 set_last_amps_offered(TWCID=61 60, desiredAmpsOffered=0)
10:08:58 TWCMaster  10 Total amps all slaves are using: 0.0
10:08:58 TWCSlave   20 SHB 6160: 00 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
10:08:58 TWCSlave   20 Slave power for TWCID b'a`', status: 0
10:08:58 urllib3.co 10 Starting new HTTP connection (1): 192.168.1.3:1080
10:08:58 TWCManager 9 Ignoring byte FC between messages.
10:08:58 urllib3.co 10 http://192.168.1.3:1080 "GET /api/all/power/now HTTP/1.1" 200 578
10:08:58 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 263, in background_tasks_thread
    check_green_energy()
  File "/home/pi/TWCManager/TWCManager.py", line 323, in check_green_energy
    master.setGeneration(module["name"], module["ref"].getGeneration())
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 61, in getGeneration
    self.update()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 126, in update
    self.getGenerationValues()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 108, in getGenerationValues
    for phase in httpResponse.json()["phases"]:
KeyError: 'phases'
, occurred when processing background task
10:08:58 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API

ngardiner, 2021-04-19:
@magic7music thanks for that - I see the reason why

ngardiner, 2021-04-19:
@magic7music I've just updated the code to parse the output properly, could you please fetch my latest update with a ```git pull``` and try again, and let me know if you see any errors or if you see generation values?

magic7music, 2021-04-19:
I#ll check it out!

magic7music, 2021-04-19:
Hi Nathan,
yes, the phase / data was the problem. I saw it in the script but wasn't shure how to change it to the right value.
Thanks for that.

That meens, it works for now.

I´ll write you a mail during the day.
Have a good night.

Koen

magic7music, 2021-04-22:
Hi, today I did a new git pull.
I´m just waiting for the sun to come out.
I think i have to wait until tomorrow.

Koen

magic7music, 2021-04-22:
Hi Nathan,
This happend earlier today after the git pull @15:00 GMT

17:25:14 TWCManager 12 Rx@: (FC) FD E0 61 60 77 77 00 00 00 00 00 00 00 00 00 8F
17:25:14 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
17:25:14 RS485      12 Tx@: C0 FB EB 77 77 61 60 00 00 00 00 00 00 00 00 9A C0
17:25:14 Policy     13 Policy conditions were not matched.
17:25:14 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
17:25:14 Policy     13 Policy conditions were not matched.
17:25:14 Policy     13 Evaluating Policy match (tm_hour [17]), condition (gte), value (6)
17:25:14 Policy     13 Evaluating Policy match (tm_hour [17]), condition (lt), value (20)
17:25:14 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [9]), condition (lte), value (17)
17:25:14 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
17:25:14 TWCMaster  10 Number of cars charging now: 0
17:25:14 SmartPi    10 Fetching SmartPi EMS sensor values
17:25:14 TWCSlave   10 desiredAmpsOffered: 1.82 < minAmpsToOffer: 6 (flexAmps: 0)
17:25:14 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0
17:25:14 urllib3.co 10 Starting new HTTP connection (1): 192.168.1.3:1080
17:25:14 TWCSlave   10 set_last_amps_offered(TWCID=61 60, desiredAmpsOffered=0)
17:25:14 urllib3.co 10 http://192.168.1.3:1080 "GET /api/all/power/now HTTP/1.1" 200 581
17:25:14 TWCMaster  10 Number of cars charging now: 0
17:25:14 TWCMaster  10 Total amps all slaves are using: 0.0
17:25:14 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
17:25:14 TeslaAPI   13 car_api_available returning True
17:25:14 urllib3.co 10 Starting new HTTPS connection (1): owner-api.teslamotors.com:443
17:25:14 TWCManager 12 Rx@: (FC) FD E0 61 60 77 77 00 00 00 00 00 00 00 00 00 8F
17:25:14 TWCMaster  10 Number of cars charging now: 0
17:25:14 TWCSlave   10 desiredAmpsOffered: 3.39 < minAmpsToOffer: 6 (flexAmps: 0)
17:25:14 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0
17:25:14 TWCSlave   10 set_last_amps_offered(TWCID=61 60, desiredAmpsOffered=0)
17:25:14 TWCManager 12 Rx@: (FC) FD EB 61 60 00 59 00 62 00 F4 00 00 00 00 00 00 00 00 00 5B
17:25:14 TWCManager 20 Slave TWC 6160: Delivered 5832802 kWh, voltage per phase: (244, 0, 0).
17:25:14 HTTPContro 10 Starting API GET (Url: /api/getStatus / IP: 192.168.1.8)
17:25:14 TWCMaster  10 Number of cars charging now: 0
17:25:14 TWCMaster  10 Total amps all slaves are using: 0.0
17:25:14 HTTPContro 10 Ending API GET (Url: /api/getStatus / IP: 192.168.1.8)
17:25:15 RS485      12 Tx@: C0 FB E0 77 77 61 60 09 00 00 00 00 00 00 00 00 98 C0

17:25:16 TeslaAPI   13 Tessy Arissa not ready because it wasn't woken in the last 2 minutes.
17:25:16 TWCMaster  10 Total amps all slaves are using: 0.0
17:25:16 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 272, in background_tasks_thread
    update_statuses()
  File "/home/pi/TWCManager/TWCManager.py", line 346, in update_statuses
    if cnnwatts > 0:
NameError: name 'cnnwatts' is not defined**
, occurred when processing background task**
17:25:16 RS485      12 Tx@: C0 FB E0 77 77 61 60 09 00 00 00 00 00 00 00 00 98 C0
17:25:16 TWCManager 12 Rx@: (FD) FD E0 61 60 77 77 00 00 00 00 00 00 00 00 00 8F
17:25:16 TWCMaster  10 Number of cars charging now: 0
17:25:16 TWCSlave   10 desiredAmpsOffered: 3.39 < minAmpsToOffer: 6 (flexAmps: 0)
17:25:16 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0

I think defininng "cnnwatts" at startup with 0 could do the trick, or am I on the wrong path with this?

magic7music, 2021-04-22:
The problem stated above is only accuring when the "subtractChargerLoad": true. When "false" it runs smoothly except for the known problem of charging.

ngardiner, 2021-04-23:
My bad, Koen. It was a typo, my fault for rushing it. Could you please do a ```git pull``` and try again with ```subtractChargerLoad``` enabled?

magic7music, 2021-04-23:
Dear Nathan,
did an update @06:30 GMT.
waiting for the sunto come out here, but for now, no script failure.

magic7music, 2021-04-23:
Hi Nathan,
   
  
  
   
  
  
   was it the typo cnnwatts?
   
  
  
   
  
  
   now during the day I did some tests.
   
  
  
   The on/off issue is still there.
   
  
  
   I think it has something to do with the timing of the reading from the smartPi and the calculation (adding the charging value of the car).
   
  
  
   They are too far appart from eachother, I think.
   
  
  
   The twcm looks too late or too early to see if there is enough energy for charging.
  
  
   
  
  
   I´ve added a 30min log.
   
  
  
   
  
  
   The screenshot shows:
   
  
  
   Upper grafik:
   
  
  
   Green is the Solaredge generation
   
  
  
   
  
  
   Lower grafik:
   
  
  
   red: generation total
   
  
  
   green: Solaredge generation
   
  
  
   Yellow: Mastervolt generation
   
  
  
   Blue: car charging
   
  
  
   
  
  
   Cheers
   
  
  
   Koen

magic7music, 2021-04-27:
this happend all of a sudden.

pi@TWC-Manager:~/TWCManager $ python3 -m TWCManager
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 373, in lex
    for x in l.lex(stream, self.root_lexer.newline_types, self.root_lexer.ignore_types):
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 174, in lex
    raise UnexpectedCharacters(stream, line_ctr.char_pos, line_ctr.line, line_ctr.column, allowed=allowed, state=self.state, token_history=last_token and [last_token])
lark.exceptions.UnexpectedCharacters: No terminal defined for '{' at line 1 col 2

3{
 ^

Expecting: {'RBRACE', 'RSQB', 'TRAILING_COMMA'}

Previous tokens: Token(SIGNED_NUMBER, '3')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/commentjson-0.9.0-py3.7.egg/commentjson/commentjson.py", line 215, in load
    return loads(fp.read(), **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/commentjson-0.9.0-py3.7.egg/commentjson/commentjson.py", line 180, in loads
    parsed = _remove_trailing_commas(parser.parse(text))
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lark.py", line 311, in parse
    return self.parser.parse(text, start=start)
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parser_frontends.py", line 89, in parse
    return self._parse(token_stream, start, *[sps] if sps is not NotImplemented else [])
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parser_frontends.py", line 54, in _parse
    return self.parser.parse(input, start, *args)
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parsers/lalr_parser.py", line 36, in parse
    return self.parser.parse(*args)
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parsers/lalr_parser.py", line 84, in parse
    for token in stream:
  File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 387, in lex
    raise UnexpectedToken(t, e.allowed, state=e.state)
lark.exceptions.UnexpectedToken: Unexpected token Token(LBRACE, '{') at line 1, column 2.
Expected one of: 
	* RBRACE
	* RSQB
	* TRAILING_COMMA

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 126, in <module>
    config = commentjson.load(jsonconfig)
  File "/usr/local/lib/python3.7/dist-packages/commentjson-0.9.0-py3.7.egg/commentjson/commentjson.py", line 217, in load
    raise JSONLibraryException(e)
commentjson.commentjson.JSONLibraryException: JSON Library Exception

Exception thrown by library (json): Unexpected token Token(LBRACE, '{') at line 1, column 2.
Expected one of: 
	* RBRACE
	* RSQB
	* TRAILING_COMMA

    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 373, in lex
        for x in l.lex(stream, self.root_lexer.newline_types, self.root_lexer.ignore_types):
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 174, in lex
        raise UnexpectedCharacters(stream, line_ctr.char_pos, line_ctr.line, line_ctr.column, allowed=allowed, state=self.state, token_history=last_token and [last_token])
    lark.exceptions.UnexpectedCharacters: No terminal defined for '{' at line 1 col 2
    
    3{
     ^
    
    Expecting: {'RBRACE', 'RSQB', 'TRAILING_COMMA'}
    
    Previous tokens: Token(SIGNED_NUMBER, '3')
    
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/commentjson-0.9.0-py3.7.egg/commentjson/commentjson.py", line 215, in load
        return loads(fp.read(), **kwargs)
      File "/usr/local/lib/python3.7/dist-packages/commentjson-0.9.0-py3.7.egg/commentjson/commentjson.py", line 180, in loads
        parsed = _remove_trailing_commas(parser.parse(text))
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lark.py", line 311, in parse
        return self.parser.parse(text, start=start)
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parser_frontends.py", line 89, in parse
        return self._parse(token_stream, start, *[sps] if sps is not NotImplemented else [])
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parser_frontends.py", line 54, in _parse
        return self.parser.parse(input, start, *args)
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parsers/lalr_parser.py", line 36, in parse
        return self.parser.parse(*args)
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/parsers/lalr_parser.py", line 84, in parse
        for token in stream:
      File "/usr/local/lib/python3.7/dist-packages/lark_parser-0.7.8-py3.7.egg/lark/lexer.py", line 387, in lex
        raise UnexpectedToken(t, e.allowed, state=e.state)
    lark.exceptions.UnexpectedToken: Unexpected token Token(LBRACE, '{') at line 1, column 2.
    Expected one of: 
    	* RBRACE
    	* RSQB
    	* TRAILING_COMMA
    
    
And here some more info's about the Start/stop issue of the SmartPi modul.

[journal 27_04_2024.txt](https://github.com/ngardiner/TWCManager/files/6384813/journal.27_04_2024.txt)

ngardiner, 2021-04-27:
That error is complaining that the first character of your ```/etc/twcmanager/config.json``` file is a ```3``` when it should be a ```{```.

I'll have a read through that log file and get back to you

magic7music, 2021-04-27:
stupid me!!! Thanks

magic7music, 2021-04-28:
Hi Nathan,
maybe this is the thing we where looking for.
In the original TWCManager.py there is a command at line 1253  timeTo0Aafter06 = now + 44.
Could this be the reason for the charger to go on/off due to the late calculation of the avaliable power?
Or am I on the wrong path?

cheers Koen

ngardiner, 2021-04-28:
Hi @magic7music,

That same line still exists in TWCManager.py today.

What I see in your logs is:

- Between 14:36:29 and 14:36:39 (10 seconds), 3.24 amps were offered
- At 14:36:39, SmartPi was queried and this changed to 9.13A at 14:36:39 until 14:37:40 (1 minute) when the next fetch from SmartPi gives a generation value of 96W (0A) which stops charging. 

Your notes mention:

```
Until here, the car charged with approx. 9A, and now it stopped.
There was 2,300W available
```

This is correct, the car charged at 9.13 until this point, but the SmartPi value returned (96W) caused the offer to drop to 0A

- The logs end here, so I can't see what the next read from SmartPi was, but even with a higher resolution, we'd still have stopped charging at that time. The key issue is why we read the value of 96W from SmartPi - either that's the value we got, or we parsed it wrong.

ngardiner, 2021-04-28:
I would say, not knowing the specifics of the weather at the time, that it does seem very strange to me that your transition would have been 3.24A -> 9.13A -> 0A over the period of 1.5 minutes - usually solar generation follows somewhat of a curve, maybe there was cloud or other conditions but it seems unlikely that we'd see that sort of flux over 90 seconds - what are your thoughts?

magic7music, 2021-04-28:
Well there was a flux and we had some heavy wind, with thick clouds passing by.
Nathan, can you tell me how to make a log of 30 minutes with debug:13 so you get all the information?
I now copy it from the small terminal window.
That only gives me a small range in time.
Now we have solid sunshine for the next couple of hours, I hope.

ngardiner, 2021-04-28:
Hi Koen,

What I'd recommend is:

- If you could please grab the latest source with a ```git pull```, as I just pushed an update for additional debug and it captures an important value that I think might have something to do with what you're seeing - the data has a unit and we're assuming it's watts for each phase but it's very possible you're getting a result with kW as the unit for example

- Ensure ```debugLevel``` is set to 11 in ```config.json```
- Enable the FileLogger logging module in ```config.json``` and point it somewhere writable, something like this would be perfect:

```
        "FileLogger": {
          "enabled": true,
          "path": "/tmp/twcmanager",
```

And then let it run for a while like that. Once you're happy that you have a decent amount of data logged, could you please do the following:

```
cd /tmp
for file in twcmanager*; do curl bashupload.com -sT $file;done > output.txt
```

The above will upload all of those files to ```bashupload.com``` and then provide a URL for each. The URL can only be used once so if you could provide me the output of ```output.txt``` to ngardiner@gmail.com, I'll download those logfiles and work out exactly what's going wrong.

magic7music, 2021-04-28:
ok, let´s go for it.

magic7music, 2021-04-28:
Oh Oh,

this is the output.

15:11:23 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
15:11:23 RS485      12 Tx@: C0 FB EB 77 77 61 60 00 00 00 00 00 00 00 00 9A C0
15:11:23 Policy     13 Policy conditions were not matched.
15:11:23 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
15:11:23 Policy     13 Policy conditions were not matched.
15:11:23 Policy     13 Evaluating Policy match (tm_hour [15]), condition (gte), value (6)
15:11:23 Policy     13 Evaluating Policy match (tm_hour [15]), condition (lt), value (20)
15:11:23 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [9]), condition (lte), value (15)
15:11:23 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
15:11:23 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 264, in background_tasks_thread
    check_green_energy()
  File "/home/pi/TWCManager/TWCManager.py", line 324, in check_green_energy
    master.setGeneration(module["name"], module["ref"].getGeneration())
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 63, in getGeneration
    self.update()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 132, in update
    self.getGenerationValues()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/SmartPi.py", line 112, in getGenerationValues
    logger.log(logging.INFO8, "Logged " + phase["values"][0]["data"] + " " + phase["values"][0]["unity"] + " for phase " + phase["name"])
TypeError: can only concatenate str (not "float") to str
, occurred when processing background task
15:11:23 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:11:23 TeslaAPI   13 car_api_available returning True

ngardiner, 2021-04-28:
My bad Koen, could you please ```git pull``` and give it another try?

magic7music, 2021-04-28:
did it and it runs, but shit happens. The sun has gone behind the clouds.

ngardiner, 2021-04-28:
That's okay, it's still useful data - there's a few things that I need to verify regarding the SmartPi output, and if we have any luck we may get a bit of direct sun at some point, that would be a bonus

magic7music, 2021-04-28:
back on track.
I think we have to count the amps used by the charger, together with the rest that is left of the green energy (smartpi).
If the car charges, the green energy which is left (smartpi) is ofcourse under the min 6A the car needs.
Example:
smartpi green energy 2200W, car 0W
The car starts charging @1800W. The smartpi has 400W left and this is the green energy at this point according to the TWCM
In the next step, TWCM; looks at the green energy. There are only 400W there, so he kills the charging process.
That means, smartpi shows 2200W again and the car 0W.
It starts over again.
this si what it looks like. You can send me your number. Maybe it´s easier to explain it via whatsapp call.
I´ll send the log in a few minutes.
cheers.

Koen

magic7music, 2021-04-28:
[28_04_2021logfile.txt](https://github.com/ngardiner/TWCManager/files/6392254/28_04_2021logfile.txt)
 Here it is.

ngardiner, 2021-04-28:
Great, thanks for that. Here's what I think the issue is:

```
2021-04-28 15:26:05,117 - SmartPi    13 Logged -159.45404 W for phase phase 1
2021-04-28 15:26:05,118 - SmartPi    13 Logged -54.31996 W for phase phase 2
2021-04-28 15:26:05,119 - SmartPi    13 Logged 184.95547 W for phase phase 3
```

We're logging 2 negative values for Phase 1 and Phase 2, and a positive value for Phase 3. If we add those together (which we do) we get -213.774 + 184.95547 = 28W

Is the polarity different for Phase 3 for any reason - I think the only reason it's charging sometimes is because of a low read on Phase 3:

```
2021-04-28 15:36:15,483 - SmartPi    13 Logged -295.29852 W for phase phase 1
2021-04-28 15:36:15,484 - SmartPi    13 Logged -276.0558 W for phase phase 2
2021-04-28 15:36:15,485 - SmartPi    13 Logged 92.13097 W for phase phase 3
```

magic7music, 2021-04-28:
@15:38:00 I got 2200W as green energy.

My grid is not balaced.
That means on phase 1 the solaredge solar converter. As soon as he generates power, the polarity will be negativ, that means power is flowing back into the grid.
The same thing is the case on phase 2. There is a mastervolt system.
On phase 3 is the car.
In Germany we have a grid meter that counts all the phases together.

ngardiner, 2021-04-28:
Got it, so you're exporting on Phases 1 and 2 and importing on Phase 3. The problem is, by reading all 3 phases we're offsetting your export by your import, but your import is the charger circuit, so we'll always "import" what you export (by increasing the charging to match the export) and therefore zero out your export, which will then cause your car to stop charging, because it will read 0 overall.

Would it make sense for us to ignore Phase 3 if you're not generating on that Phase? It's not entirely ideal as we're only measuring your generation and not your consumption, but if your consumption zeros out your generation we'll only ever read one value and that will be zero, and that won't help TWCManager to make any decisions on how it should charge - so we need to either do one of two things:

 - Read the value as purely generation only, so negative = export and positive = 0 generation. This will never show you a consumption value.

or

 - Read a negative value as generation and a positive value as consumption, but this might give you some inaccurate values for consumption. This is because your consumption value is after offset against generation, ie it's at the export side not the meter side, so we'll only know about true consumption on Phase 3, phase 1 and 2 would just report lower generation values as the consumption is already subtracted from the export on the meter side.

Most clamp/export based systems just read the export/generation value and not the consumption value, but as yours is 3-phase we have 3 values to deal with, so we can offer the option to attribute positive values to consumption, I just don't know how useful the number will be to you.

ngardiner, 2021-04-28:
In fact, we already have a switch for that. So I've modified the module to determine per phase what is a positive or negative value and split those into generation or consumption. This should work for both cases.

Then we have the switch ```showConsumption``` in the config file. If it's disabled (default), it only shows the generation value. If it's set to true, it will show any positive values read from any of the 3 phases as a consumption value. You could experiment with both settings and see which one gives the most accurate output.

I've pushed the change already, if you could do a ```git pull``` and let me know how it goes, that would be great.

magic7music, 2021-04-28:
I'll do a pull. and good news!!!!! It works.
The showConsumption switch isn´t there or I am too ... for this.

Here ara my thoughts.

I Think we need all 3 phases, as I#m not the only one who will use this interface.
In my case, the consumption is the positive count of the smartpi.
This I take from the grid and have to pay for it. Now having the consumpion in there makes no sense, for now.
As soon as you have a Powerwall it´s something complete different.
I'm into a project where I want to give an old electrical car a second life as a powerwall.

car charge
{"serial":"smartpi170512345","name":"House","lat":52.3667,"lng":9.7167,"time":"2021-04-28 16:15:57","softwareversion":"","ipaddress":"192.168.1.3","datasets":[{"time":"2021-04-28 16:15:57","phases":[{"phase":1,"name":"phase 1","values":[{"type":"power","unity":"W","info":"","data":-908.6961}]},{"phase":2,"name":"phase 2","values":[{"type":"power","unity":"W","info":"","data":-640.85736}]},{"phase":3,"name":"phase 3","values":[{"type":"power","unity":"W","info":"","data":1494.473}]},{"phase":4,"name":"phase 4","values":[{"type":"power","unity":"W","info":"","data":0}]}]}]}

total on the smartpi will be:-54w
we have to add the 1494 from the charging so that we come up with:-1548w

car stopped
{"serial":"smartpi170512345","name":"House","lat":52.3667,"lng":9.7167,"time":"2021-04-28 16:16:56","softwareversion":"","ipaddress":"192.168.1.3","datasets":[{"time":"2021-04-28 16:16:56","phases":[{"phase":1,"name":"phase 1","values":[{"type":"power","unity":"W","info":"","data":-961.6047}]},{"phase":2,"name":"phase 2","values":[{"type":"power","unity":"W","info":"","data":-638.56396}]},{"phase":3,"name":"phase 3","values":[{"type":"power","unity":"W","info":"","data":93.2559}]},{"phase":4,"name":"phase 4","values":[{"type":"power","unity":"W","info":"","data":0}]}]}]}

total on the pi will be:-1506w

magic7music, 2021-04-28:
added it myself!!

magic7music, 2021-04-28:
we have a strike!!!

MikeBishop, 2021-04-28:
> The problem is, by reading all 3 phases we're offsetting your export by your import, but your import is the charger circuit, so we'll always "import" what you export (by increasing the charging to [...]

What if you report as generation the sum of all phases which are exporting, and as consumption the sum of all phases which are importing?

ngardiner, 2021-04-28:
Yes, this is what the module will now do. The problem is that the consumption values will be inaccurate, but will be entirely usable (and correct) for the purpose. For example:

- Generating 2kW on Phase 1
- Generating 3kW on Phase 2
- Charger is on Phase 3

Let's say you then start up an Air Conditioner load. You now have 1kW load on all 3 phases, instead of showing as 3kW load, it will show as:

- Generating 1kW on Phase 1
- Generating 2kW on Phase 2
- Charger load + 1kW consumption on Phase 3

That will do the job, though, as we can turn on ```subtractChargerLoad``` and it will be subtracted from the consumption on Phase 3. You'll get accurate usage responses as we will see the load across all 3 phases, but as we see it in different ways (subtracted from generation on P1/P2 vs consumption on P3) the numbers themselves won't be all that useful.

So, the conclusion is:

   - Setting ```showConsumption``` to true in the config of the SmartPi EMS module is recommended, without it you'll only see 2/3 of the consumption because Phase 3 will simply read as 0
   - However if you turn ```showConsumption``` on, the Charger Load will appear as consumption, so the next thing you need to do is enable ```subtractChargerLoad``` in the ```config.json``` file

At that point it should be working as intended.

ngardiner, 2021-04-28:
Unfortunately it raises an entirely different problem.

Imagine that the Charger Load were on P1 or P2 instead. We would see generation drop as charger load increases, and subtractChargerLoad only deals with this if the consumption is 0 (at which point it adds the Charger Load as generation as we don't have a consumption value to subtract from)

This works fine for single phase systems. However the 3 phase systems can both show Generation and Consumption values AND not have consumption of the charger showing, yet have the charger load subtracted from generation due to the measurement being taken at export.

While we haven't encountered it yet, for their scenario we'd probably need an additional setting that explicitly tells us to add charger load to generation and not subtract from consumption.

magic7music, 2021-04-29:
We had a cloudy day, so I couldn´t test it a little bit more.
Maybe during the weekend, but until now it looks good.
Logging is still enabled and I´ll send you a copy of the next charging session.

MikeBishop, 2021-04-30:
If you opt in to subtracting charger load, you could also just allow consumption to be negative.  The rest of the math should still work.  Though I agree, it looks strange.

That could also be addressed by changing the presentation rather than the numbers -- if we know we're monitoring the grid flow, show "importing" and "exporting" rather than generation/consumption.  Maybe the EMS could expose whether it's directly monitoring production or simply metering the grid.

magic7music, 2021-04-30:
we can determin via the cosPi in the API stream if it's positive (importing from the grid) or a negative cosPi (exporting to the grid).
You can find the explenation of all API codes in this link.

https://enerserve.eu/en/service/smartpi/manual/api.html

If we take http://smartPi IP:1080/api/all/all/now, we get all the values from the meter.

ngardiner, 2021-05-18:
Great to hear that Koen! Thank you for the update