ngardiner/TWCManager #254: cancelChargeNow REST call doesn't stop charging

jherby2k, 2021-03-26:
I'm just playing with the REST api for the first time.

My car is not charging right now (tracking green energy, and its night time). If I issue a "chargeNow" call, it begins charging as expected. If I then issue a "cancelChargeNow" it seems to drop back down to 6 amps (my configured minimum) but doesn't stop charging.

cancelChargeNow should use the Tesla API to stop charging if charging was stopped initially. Right? bug?

Great work by the way! Just got this installed this afternoon without any hickups.

ngardiner, 2021-03-26:
Hi @jherby2k!

What happens when you issue a chargeNow command is that the Charge Now policy takes precedence over whatever other policy would normally be running at that point in time, to charge at the specified rate for the specified time.

Once that time runs out or if you end it early with the cancelChargeNow call, it will fall back to whatever the policy would have been at that time. If it happens that the charge policy would charge at 6A (for example, if non-scheduled charging is set to 6A) then it would start charging at 6A afterwards, so it does depend on the policy active at the time.

What policy name does it show you after you execute the cancelChargeNow in the Web Interface or Console, is it Non-Scheduled Charging?

jherby2k, 2021-03-26:
I may have just been impatient... will test again

jherby2k, 2021-03-26:
Yeah you're right - sorry to waste your time. Just took 2 minutes or so to stop entirely.

closed 2021-03-26 by jherby2k