ngardiner/TWCManager #255: Logging to file does not work

Flitzer42, 2021-04-02:
When activating logging with this
```
        "FileLogger": {
          "enabled": true,
          "path": "/etc/twcmanager/log",
          "mute":{
              "ChargeSessions": false,
              "GreenEnergy": true,
              "SlavePower": false,
              "SlaveStatus": false,
              "DebugLogLevelGreaterThan": 1
          }
        },

```

And starting TWC Manager with this
`sudo systemctl start twcmanager.service`

Right after starting the service `systemctl status` promts me this:
```
pi@raspberrypi:~ $ systemctl status twcmanager.service
● twcmanager.service - Tesla Wall Charger Manager
   Loaded: loaded (/etc/systemd/system/twcmanager.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-04-02 09:32:12 CEST; 661ms ago
 Main PID: 1094 (python3)
    Tasks: 1 (limit: 2062)
   CGroup: /system.slice/twcmanager.service
           └─1094 /usr/bin/python3 -u -m TWCManager

Apr 02 09:32:12 raspberrypi systemd[1]: Started Tesla Wall Charger Manager.
```

But after some seconds this
```
pi@raspberrypi:~ $ systemctl status twcmanager.service
● twcmanager.service - Tesla Wall Charger Manager
   Loaded: loaded (/etc/systemd/system/twcmanager.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-04-02 09:32:20 CEST; 4min 57s ago
  Process: 1101 ExecStart=/usr/bin/python3 -u -m TWCManager (code=exited, status=1/FAILURE)
 Main PID: 1101 (code=exited, status=1/FAILURE)

Apr 02 09:32:20 raspberrypi systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Apr 02 09:32:20 raspberrypi systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Apr 02 09:32:20 raspberrypi systemd[1]: Stopped Tesla Wall Charger Manager.
Apr 02 09:32:20 raspberrypi systemd[1]: twcmanager.service: Start request repeated too quickly.
Apr 02 09:32:20 raspberrypi systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Apr 02 09:32:20 raspberrypi systemd[1]: Failed to start Tesla Wall Charger Manager.

```

Do youhave any hint ?
I think the `Start request repeated too quickly` is only a sideeffect, caused by the termiantion of the service butnot the root cause.
I already tried a different path. I already re-installed on a blank system with the leatest 1.2.1.

Many thanks in advance

BTW: Need to enable logging as TWC Manager does not respond after some hour of operation ....

tjikkun, 2021-04-02:
You can try to run `/usr/bin/python3 -u -m TWCManager` to see if that outputs anything useful.

Flitzer42, 2021-04-02:
Thanks for the hint with 

> /usr/bin/python3 -u -m TWCManager

 Ii now shows  

> FileNotFoundError: [Errno 2] No such file or directory: '/etc/twcmanager/log/logfile'

Adding the directory solved the problem.
Thanks

closed 2021-04-04 by ngardiner