ngardiner/TWCManager #256: ImportError: cannot import name 'unicode' from 'past.builtins'

the-dbp, 2021-04-11:
I get:
```
./TWCManager.py
Traceback (most recent call last):
  File "./TWCManager.py", line 42, in <module>
    from ww import f
  File "/usr/local/lib/python3.7/dist-packages/ww-0.2.1-py3.7.egg/ww/__init__.py", line 5, in <module>
    from .wrappers.iterables import IterableWrapper as g  # noqa
  File "/usr/local/lib/python3.7/dist-packages/ww-0.2.1-py3.7.egg/ww/wrappers/iterables.py", line 173, in <module>
    from ww.types import Any, Union, Callable, Iterable, T, T2  # noqa
  File "/usr/local/lib/python3.7/dist-packages/ww-0.2.1-py3.7.egg/ww/types.py", line 3, in <module>
    from past.builtins import unicode as past_unicode
ImportError: cannot import name 'unicode' from 'past.builtins' (/usr/local/lib/python3.7/dist-packages/future-0.18.2-py3.7.egg/past/builtins/__init__.py)
```

when running ./TWCManager.py or python -m TWCManager

I also got:
`ModuleNotFoundError: No module named 'lark'`
but I solved that with pip install lark_parser

ngardiner, 2021-04-11:
Hi @the-dbp, thanks for reporting this.

Could you please provide the output of:

```
python3 -V

lsb_release -a
```

the-dbp, 2021-04-11:
Python 3.7.3

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

the-dbp, 2021-04-11:
both of these issues were due to a faulty sd card

closed 2021-04-11 by the-dbp