ngardiner/TWCManager #260: how does MQTT work?

IT-Economy, 2021-04-17:
I like to controll my Tesla Wallbox via MQTT from ioBroker.

I set up the MQTTconnection: 

` "MQTT": {
        "enabled": true,
        "brokerIP": "192.168.20.20",
        "topicPrefix": "TWC",
        "username": "mqtt",
        "password": "xxx"
      }
`

But I don't see the topics in the MQTT server in ioBroker:

[image]

I can't add the topics manually, they should get created by TWCManager, so that I then can change the values.

The interface seems very limited? What I expected was:

control:
startCharge:                   (trigger to start the charge)
stopCharge:                   (trigger to stop the charge)
chargeMaxAmps: 5...32 (int value to set the charge speed before charging is started or even during charging)

status:
charging : true/false      (reflect the current status)
chargeAmps:                 (current amps when charging)
chargeWatts:                 (current watts when charging)
chargedSeconds:  xx S  (status info)
chargedkWh: xx kWh    (status info)

Do I need to specify a duration in seconds in "chargeNow" or can that be omitted? I want to charge until I send the stopCharge command (or set charging to false, or the car stops by itself).

Do I need to disable any other features to have full control over MQTT and do the green energy tracking in ioBroker myself?

Thanks for your help!

ngardiner, 2021-04-18:
Hi,

> But I don't see the topics in the MQTT server in ioBroker:

Yes, that is because we act as the MQTT subscriber, not publisher. We wait for those topics to be published with a given value, and we act on that value. 

> I can't add the topics manually, they should get created by TWCManager, so that I then can change the values.

That wouldn't work. Simply the creation of the stop topic makes charging stop. If we pre-created these topics, we'd be sending ourselves commands that we would then act on. I think you may be confusing the MQTT Control (within which we are a subscriber, listening for published MQTT topics) module with the MQTT Status (of which we are a publisher, publishing topics containing information which reflects the status of TWCManager). Please refer to the docs below for more.

> The interface seems very limited? What I expected was:

Sorry to hear you aren't impressed, although to be fair a lot of what you are asking for already exists:

https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Control_MQTT.md

https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Status_MQTT.md

For some of the things you have flagged here, it is important to keep in mind that you're communicating with a string of TWCs, not just one, so:

-  charging : true/false (reflect the current status)

This is reported as an integer, as more than one vehicle can be charging at a time if there's more than one slave TWC.

-  chargedkWh: xx kWh (status info)

Depends on the context. If this is per TWC, it is expressed for lifetime kWh charged (which is tracked by the TWC). If it's for a charging session, no such value exists today as we don't get this level of information from the TWC itself. We can track the start lifetime value and end lifetime value per TWC before/after a charging session, but we cannot read this from a vehicle short of implementing API calls on the vehicle side to record charging sessions this way.

> Do I need to specify a duration in seconds in "chargeNow" or can that be omitted? I want to charge until I send the stopCharge command (or set charging to false, or the car stops by itself).

You need to specify duration, as this determines for how long the charge now policy overrides other policies such as tracking green energy, scheduled or non scheduled charging. If you want it to be ongoing, specify a value larger than you'd need to charge for (eg 24 hours) and cancel the charge now policy when you're done. Also, keep in mind you can keep updating the Charge Now rate and duration as you go, ie you could start charging at 6A for 1 hour and then 30 minutes into the session, publish a new value requesting to charge for 2 hours at 32A.

> Do I need to disable any other features to have full control over MQTT and do the green energy tracking in ioBroker myself?

Please make sure that you have not enabled Green Energy tracking, and that Non-Scheduled Charging is set to Do Not Charge. This will ensure that no additional charging occurs outside of the charging you request via MQTT.

IT-Economy, 2021-05-03:
Thank you for your explanation!

After some thoughts all I want to do is set the maximum charge speed over MQTT. I can read out the solar energy surplus with some scripts in ioBroker and control the charge from there.
I can start and stop the charge from the car over the Tesla API. So just setting the charge speed would be sufficient for me.
Do I need to disable any features in TWC to be able to do that reliably?
I did not find a setting to control the charge speed on the web frontend of TWC. Is this disabled?

I am not sure how to add a new MQTT topic to the server manually without the client registering the topic on its own.

MikeBishop, 2026-10-26:
v1.4.0 adds an MQTT control topic for `nonScheduledAmpsMax` (7c63e2b7, #475), which sets the maximum charge rate over MQTT. Could you re-verify on v1.4.0 whether that covers what you needed?