ngardiner/TWCManager #266: Volkszahler API

Saftwerk, 2021-04-24:
Which setting is correct for **"serverPort": "xx"** at Volkszahler API?
Is it port of the Volkszahler Frontend which shows the webpage (port 80 or 8080)
or port of the VZLOGGER which shows json output? (port 8081)

/etc/twcmanager/config.json
```
        "Volkszahler": {
          "enabled": true,
          "serverIP": "pi3.fritz.box",
          "serverPort": "8081",
          "uuid": "00000004-0001-1670-ffff-0123456789ab"
        }
```

TWCManager "logLevel": 13
output: Volkszahler API reports HTTP Status Code 404
```
15:08:54 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
15:08:54 Policy     13 Policy conditions were not matched.
15:08:54 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
15:08:54 Policy     13 Policy conditions were not matched.
15:08:54 Policy     13 Evaluating Policy match (tm_hour [15]), condition (gte), value (6)
15:08:54 Policy     13 Evaluating Policy match (tm_hour [15]), condition (lt), value (20)
15:08:54 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [0]), condition (lte), value (15)
15:08:54 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
15:08:54 Volkszahle 17 Volkszahler API reports HTTP Status Code 404
15:08:54 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:08:54 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1619269734.3728008 or empty carApiBearerToken ''
15:08:54 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
15:08:54 TWCManager 20 Green energy generates 0W, Consumption 0W, Charger Load 0W
15:08:54 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
15:08:54 TWCManager 20 Charge when above 10A (minAmpsPerTWC).
```

My VZLOGGER uses port 8081 in Volkszahler file "etc/vzlogger.conf":
  "local": {
    "enabled": true,
    "port": 8081,
    "index": true,
    "timeout": 0,
    "buffer": 0
  },

Using this VZLOGGER port 8081 in the browser `http://pi3.fritz.box:8081/` will show this output:

`{ "version": "0.8.0", "generator": "vzlogger", "data": [ { "uuid": "00000001-0001-0280-ffff-0123456789ab", "last": 1619267932763, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932763, 767412.20000000007 ] ] }, { "uuid": "00000002-0001-0182-ffff-0123456789ab", "last": 1619267932763, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932763, 1963741.4000000001 ] ] }, { "uuid": "00000003-0001-0181-ffff-0123456789ab", "last": 1619267932763, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932763, 3597606.2000000002 ] ] }, { "uuid": "00000004-0001-1670-ffff-0123456789ab", "last": 1619267932763, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932763, -3485.2000000000003 ] ] }, { "uuid": "00000005-0002-0180-ffff-0123456789ab", "last": 1619267932064, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932064, 58859737.100000001 ] ] }, { "uuid": "00000006-0002-1570-ffff-0123456789ab", "last": 1619267932064, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932064, 3803.8000000000002 ] ] }, { "uuid": "00000007-0003-0017-ffff-0123456789ab", "last": 1619267800774, "interval": -1, "protocol": "s0", "tuples": [ [ 1619267800774, 1 ] ] }, { "uuid": "00000008-0004-0023-ffff-0123456789ab", "last": 1618770284248, "interval": -1, "protocol": "s0", "tuples": [ [ 1618770284248, 1 ] ] } ] }`

The value to use is
{ "uuid": "00000004-0001-1670-ffff-0123456789ab", "last": 1619267932763, "interval": -1, "protocol": "sml", "tuples": [ [ 1619267932763, **-3485.2000000000003** ] ] },

Next try with Volkszahler Frontend port 80 in /etc/twcmanager/config.json
        "Volkszahler": {
          "enabled": true,
          "serverIP": "pi3.fritz.box",
          "serverPort": "80",
          "uuid": "00000004-0001-1670-ffff-0123456789ab"
        }

output:
```
14:56:51 Policy     20 New policy selected; changing to Track Green Energy
14:56:51 TWCSlave   20 SHB 2388: 00 00.00/00.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
14:56:51 TWCSlave   20 Slave power for TWCID b'#\x88', status: 0
14:56:51 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 264, in background_tasks_thread
    check_green_energy()
  File "/home/pi/TWCManager/TWCManager.py", line 324, in check_green_energy
    master.setGeneration(module["name"], module["ref"].getGeneration())
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 63, in getGeneration
    self.update()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 128, in update
    self.getGenerationValues()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 110, in getGenerationValues
    msgMatch = re.search("^(.+) W$", httpResponse.text(), re.DOTALL)
NameError: name 're' is not defined
, occurred when processing background task
14:56:51 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
```

Last try with Volkszahler Frontend port 8080 in /etc/twcmanager/config.json
        "Volkszahler": {
          "enabled": true,
          "serverIP": "pi3.fritz.box",
          "serverPort": "8080",
          "uuid": "00000004-0001-1670-ffff-0123456789ab"
        }

output: HTTP Status Code 400
```
15:05:48 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
15:05:48 Policy     13 Policy conditions were not matched.
15:05:48 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
15:05:48 Policy     13 Policy conditions were not matched.
15:05:48 Policy     13 Evaluating Policy match (tm_hour [15]), condition (gte), value (6)
15:05:48 Policy     13 Evaluating Policy match (tm_hour [15]), condition (lt), value (20)
15:05:48 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [0]), condition (lte), value (15)
15:05:48 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
15:05:48 Volkszahle 17 Volkszahler API reports HTTP Status Code 400
15:05:48 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:05:48 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1619269548.8249192 or empty carApiBearerToken ''
15:05:48 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
15:05:48 TWCManager 20 Green energy generates 0W, Consumption 0W, Charger Load 0W
15:05:48 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
15:05:48 TWCManager 20 Charge when above 10A (minAmpsPerTWC).

```

I do not know why frontend is at port 80 and also at 8080, but port 80 give no error at 1st use.
Another use of Frontend is this syntax:
`http://pi3.fritz.box/api/data.txt?from=now&uuid%5B%5D=00000004-0001-1670-ffff-0123456789ab`
It has a very simple output:
`-3599.2 W`

Saftwerk, 2021-04-25:
TWCManager "logLevel": 10

Based on the URL to Volkszahler it has to be Port:80
Browser is responding
http://pi3.fritz.box:80/api/data.txt?from=now&uuid=00000004-0001-1670-ffff-0123456789ab
"-3314.1 W" (without quotes)

So the URL and port is correct.

14:10:38 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
14:10:38 TWCMaster  10 Number of cars charging now: 0
14:10:38 Volkszahle 10 Fetching Volkszahler EMS sensor values
14:10:38 TWCSlave   10 desiredAmpsOffered: 0 < minAmpsToOffer: 10 (flexAmps: 8)
14:10:38 urllib3.co 10 Starting new HTTP connection (1): pi3.fritz.box:80
14:10:38 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0
14:10:38 TWCSlave   10 set_last_amps_offered(TWCID=23 88, desiredAmpsOffered=0)
14:10:38 urllib3.co 10 http://pi3.fritz.box:80 "GET /api/data.txt?from=now&uuid=00000004-0001-1670-ffff-0123456789ab HTTP/1.1" 200 9
14:10:38 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 264, in background_tasks_thread
    check_green_energy()
  File "/home/pi/TWCManager/TWCManager.py", line 324, in check_green_energy
    master.setGeneration(module["name"], module["ref"].getGeneration())
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 63, in getGeneration
    self.update()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 128, in update
    self.getGenerationValues()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 110, in getGenerationValues
    msgMatch = re.search("^(.+) W$", httpResponse.text(), re.DOTALL)
NameError: name 're' is not defined
, occurred when processing background task
14:10:38 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
14:10:38 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1619352638.8238535 or empty carApiBearerToken ''
14:10:38 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
14:10:38 TWCManager 12 Rx@: (FC) FD E0 23 88 77 77 00 00 00 00 00 00 00 00 00 79

Saftwerk, 2021-04-25:
I have found a missing line in ".../EMS/Volkszahler.py"
`import re`
This is needed for using Regular Expressions.

I added it and now I get another error: "TypeError: 'str' object is not callable"

14:48:53 urllib3.co 10 Starting new HTTP connection (1): pi3.fritz.box:80
14:48:54 urllib3.co 10 http://pi3.fritz.box:80 "GET /api/data.txt?from=now&uuid=00000004-0001-1670-ffff-0123456789ab HTTP/1.1" 200 9
14:48:54 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 264, in background_tasks_thread
    check_green_energy()
  File "/home/pi/TWCManager/TWCManager.py", line 324, in check_green_energy
    master.setGeneration(module["name"], module["ref"].getGeneration())
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 64, in getGeneration
    self.update()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 129, in update
    self.getGenerationValues()
  File "/home/pi/TWCManager/lib/TWCManager/EMS/Volkszahler.py", line 111, in getGenerationValues
    msgMatch = re.search("^(.+) W$", httpResponse.text(), re.DOTALL)
TypeError: 'str' object is not callable
, occurred when processing background task
14:48:54 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API

Saftwerk, 2021-04-25:
I changed line 114
old: ` self.generatedW = msgMatch.group(1)`
new:`self.generatedW = float( msgMatch.group(1) )`

Now I get this log:
```
15:31:00 Volkszahle 10 Fetching Volkszahler EMS sensor values
15:31:00 TWCSlave   10 desiredAmpsOffered: 0 < minAmpsToOffer: 10 (flexAmps: 8)
15:31:00 TWCSlave   10 no cars charging, setting desiredAmpsOffered to 0
15:31:00 urllib3.co 10 Starting new HTTP connection (1): pi3.fritz.box:80
15:31:00 TWCSlave   10 set_last_amps_offered(TWCID=23 88, desiredAmpsOffered=0)
15:31:00 HTTPContro 10 Starting API GET (Url: /api/getStatus / IP: 192.168.178.76)
15:31:00 TWCMaster  10 Number of cars charging now: 0
15:31:00 HTTPContro 10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.76)
15:31:00 TWCMaster  10 Total amps all slaves are using: 0.0
15:31:00 HTTPContro 10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.76)
15:31:00 HTTPContro 10 Ending API GET (Url: /api/getStatus / IP: 192.168.178.76)
15:31:00 urllib3.co 10 http://pi3.fritz.box:80 "GET /api/data.txt?from=now&uuid=00000004-0001-1670-ffff-0123456789ab HTTP/1.1" 200 9
15:31:00 TWCMaster  10 Number of cars charging now: 0
15:31:00 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:31:00 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1619357460.3848352 or empty carApiBearerToken ''
15:31:00 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
15:31:00 TWCMaster  10 Total amps all slaves are using: 0.0
15:31:00 TWCManager 20 Green energy generates 0W, Consumption 2845W, Charger Load 0W
15:31:00 TWCManager 10 Offering 0.00A instead of -12.06A to compensate for inexact current draw
15:31:00 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
```

Since I'm pushing energy to grid right now the "Consumption" and "Generation" must be swapped.
15:36:13 TWCManager 20 Green energy generates 0W, Consumption 2845W, Charger Load 0W

ngardiner, 2021-04-27:
Thanks @Saftwerk for your help with this. I'm closing off this issue, but have updated documentation for this EMS interface with the information you provided on the PR, which will help others out greatly, thanks!

closed 2021-04-27 by ngardiner