ngardiner/TWCManager #271: makefile dependencies - cryptography 2.1.4 on the main branch failing

mikey4321, 2021-04-27:
Hi,

I've tried to build a fresh clone of the current main branch (1.2.2) and its failing on one of the cryptography "eggs", specifically this one.  

`Installed /tmp/easy_install-nbk7nanr/cryptography-2.1.4/.eggs/pycparser-2.20-py3.7.egg
no previously-included directories found matching 'docs/_build'
warning: no previously-included files matching '*' found under directory 'vectors'
build/temp.linux-armv6l-3.7/_openssl.c:570:10: fatal error: openssl/opensslv.h: No such file or directory
 #include <openssl/opensslv.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
make: *** [Makefile:50: setup] Error 1`

full log below, but this appears to be the relevant part.  cryptography itself is installed correctly, according to pip3

ngardiner, 2021-04-27:
Hi @mikey4321,

Thanks for that. This will be fixed shortly in the Makefile, but you can manually fix it now with the following commands:

```
sudo apt-get -y install libffi-dev libssl-dev
```

ngardiner, 2021-04-28:
This has now been added to the ```main``` branch as well, so a ```git pull``` should pull in the updated Makefile with these dependencies. Going to close this off, but please re-open if there's any other errors/missing dependencies during build

closed 2021-04-28 by ngardiner

mikey4321, 2021-04-28:
worked perfectly, thanks!