ngardiner/TWCManager #274: Feature request: new interface should have a button "Check for Updates"

Saftwerk, 2021-05-02:
Isn't it annoying that lots of users are asking: "How to update this software?"
Let's have a button direct on the new interface **Check for Updates**
> Your installed version is ...
> new version available ...
> Do you want to update? [YES] [NO]
> List of changes/new features: ...

Button [YES] will do the updates full automatic script like:
- Backup local settings like "config.json".
- stop Service
- get files from github --hard
- run just new loaded update script to start Service and reload website

Since the "config.json" maybe needs update too, the most important thing is to have a version number in config.json. So in future it is possible to update or convert it without loosing personal settings.

But a much better approach with "config.json" would be: Make the users setting complete Independent from any version.
Now TWCManager always loads config.json and then it handle dynamic settings from "settings.json".
I want to say: Get rid of config.json and make all parameters available to settings.json.
(Yes there is the need of having a web page on the "Settings" tab, the can list all the defined parameters dynamic)

How it works:
On boot time TWCManager load all default values from a fixed file like the old "config.json".
Then TWCManager updates in memory the default values with user defined values coming from settings.json. The "settings.json" is ignored by version control.
If there is an future update, which makes an old parameter obsolete, then the old user parameter is dropped  from "settings.json".
The user can modify only listed parameters in the "Settings" tab with personal values.
As a result the TWCManager is easy to install and there is no more need to `sudo nano anything complicated` to do an update. It's just a click.

ngardiner, 2021-05-03:
Thanks for the feedback @Saftwerk 

As it stands, there should really not be a reason to backup/change ```config.json``` with a version upgrade, and the way you treat settings.json above is the same way we currently treat ```config.json``` - it's effectively a requirement of mine that any config settings that go in for future interfaces or settings must handle non-existence and must provide a sane default. If you find anything that does break after an upgrade with your existing ```config.json``` you'd be most welcome to raise an issue for it as I consider that a bug (and usually a big bug) and I'd fix it on the spot.

You should be able to run with the same config version across many versions of TWCManager - I do, I am probably missing half of what's in the example config because I don't use those interfaces or features in my "production" setup.

There is a concerted effort to avoid tunables in ```config.json``` and to make minor tuning available from the Web Interface, but TWCManager has no way to deal with adding interfaces on the fly and you definitely don't want to be changing the amps available or whatnot while it's running - plus if we default to a given number of amps on start and don't give a way to configure it without the Web UI running we would pop people's breakers every time they start TWCManager to get to the web interface... remember that while in EU/Asia 32 amps is high at 240v (especially 3 phase), 80 amps at 120v might not be unusual and so coming up with sane defaults is hard and potentially dangerous.

I think there will always be a place for ```config.json```, and that place will be to set things that are fundamental to the running of TWCManager:

   * Log level and logging
   * Electrical tolerances and limits (to stop from changing things at runtime in a way that TWC can't support, or accidentally getting TWCManager into a state where it can't be run because the values are incompatible with the circuit)
   * Interfaces to enable and the associated usernames, passwords and IP addresses

For ```settings.json```, this is really where the minor tweaks and tuning should live, as well as functions that are entirely web based such as scheduling and access control. The downsides of ```settings.json``` are:

   * It's neither readable nor editable (easily) so all values should be managed via Web UI
   * Web UI takes a long time to update to keep up with changes and features, hence why some may initially land in ```config,json``` - it's rare for a PR submitted for a new feature to come with the appropriate web controls for managing it and so there's got to be some cross-over. If we made it a requirement that every interface submitted had a web interface to configure it we'd probably not have as much contributed.
 
I'd also love to be able to implement in-place upgrades and updates. It might be possible, it's hard to say. There are many ways to approach the task, but I suspect few of them would be as easy as pressing a button. Different environments work differently and I have my doubts that we could come up with something that would "just work" in different environments without issues with permissions and whatnot - this is why many projects take the approach of containerizing applications so they can be easily swapped out regardless of the underlying OS/platform, and something like Docker could probably give us the best approach to this as we could run freely within the container with minimal restraints around permissions and supporting different environments, but it might be a hard sell to get everyone to move to Docker or virtualenv or whatever the platform is - you can see this under discussion in #99

Saftwerk, 2021-05-03:
Hello Nathan,

Thank you a lot for your detailed information on your config concept.

I like to see, that you want to have such an update button too.
I’m a total noob in Linux, so I can post only the idea and hopes so somebody else with more experience than me could do it.

bikeymouse, 2021-05-08:
> Hello Nathan, Thank you a lot for your detailed information on your config concept. I like to see, that you want to have such an update button too. I’m a total noob in Linux, so I can post only the idea and hopes so somebody else with more experience than me could do it.

I can now fully recommend to do the Docker-install that has just become available. 
Once you have done the initial install as per the instructions, you can simply do a "docker-compose up -d twcmanager" to update to the latest image.

drewpix, 2021-05-15:
Thanks @ngardiner for this great effort. Indeed, I am looking for any kind of discussion/guide on how to update my 1.2.1 to try out the dev version 1.2.2.  Should I re-install everything from scratch? thanks

MikeBishop, 2026-10-26:
TWCManager now checks PyPI for new versions and shows a "vX Available" link with an upgrade prompt in the navbar (4b18c6d5, v1.2.4). Closing as done.

closed 2026-10-26 by MikeBishop