ngardiner/TWCManager #277: Modern Web has no "stop charging" button?

bikeymouse, 2021-05-08:
Hi,

I'm wondering: the Modern Web Theme (which is really an improvement) has a "Start Charging Now" button, but - if charging is active - it does not have a "Stop Charge" button., like there is in the old web GUI. 

Is that a feature or a bug? (setting the charge Amps to 0 seems a bit odd to me).

ngardiner, 2021-05-08:
Yep, this is a target for improvement - let's keep this here for now as the intention is to make Modern the default after the next version (v1.2.2) is released, so this is a priority to resolve.

VIDGuide, 2021-06-26:
Looking at this one, i see the old interface does it by calling sendStopCommand(), however the UI notes this causes the car to disconnect until physically reconnected, which is less than ideal. 

Curious would it be better to send either setChargeNowAmps() with 0 amps (even with a button for "stop" wording/UI), or make a slight variation on setChargeNowTimeEnd() that just sets the end time to "now"? 

Would either/both of those work for a "soft" stop? I'm guessing if we set the end time to "now", then the automated background processes the stop a session anyway when the timer runs out will still do the work and clean it up naturally, so that's the way I'm thinking may be best, unless anyone else has other input to that end?

VIDGuide, 2021-06-27:
Even better, just found resetChargeNowAmps(), which looks like it's called from WebIPControl
Actually, the functions are all there, just missing a button. Add this in, and fixing a small bug in the disabling/enabling logic:

Desktop:
[image]
Mobile:
[image]

Not disabled:
[image]

commit 4cae255f (2021-06-27): fixes #277
Add Cancel Charge Now button
closed 2021-06-27 by ngardiner