ngardiner/TWCManager #278: Running TWCManager.py comes ack to prompt with no errors or log?

OutbackEV, 2021-05-09:
So after getting dracoventions/TWCManager working with my current HW, I loaded up, ngardiner/TWCManager v1.2.1 initially over the top of Raspbian Stretch Lite 3.5 then on a fresh install of 
3.7.3 with v1.2.1 and get the same result every time...no errors on program run and then falls back to prompt after manual run...while I'm expecting it to keep running..
    
pi@raspberrypiTWCManager:~/TWCManager $ sudo nano TWCManager.py
pi@raspberrypiTWCManager:~/TWCManager $ python3 -u -m TWCManager.py
pi@raspberrypiTWCManager:~/TWCManager $ python3 -u -m TWCManager
pi@raspberrypiTWCManager:~/TWCManager $ python TWCManager.py
pi@raspberrypiTWCManager:~/TWCManager $ python --version
Python 3.7.3
pi@raspberrypiTWCManager:~/TWCManager $ python -u -m TWCManager | tee debug.log
pi@raspberrypiTWCManager:~/TWCManager $ python3 -u -m TWCManager.py
Just wondering what I'm missing here!
Regards Hunter

ngardiner, 2021-05-11:
Strange! It isn't console logging being disabled as we'd still print out a few lines before going silent. Any chance you could capture whats happening with an strace:

```
strace python3 ./TWCManager.py > strace.txt
```

I'd probably only need to see the last ~50 lines but that would help greatly with understanding why it's terminating

tjikkun, 2021-05-11:
I think you are missing config.json

OutbackEV, 2021-05-12:
Found where the Program is terminating...
line 117-129:
config = None
jsonconfig = None
if os.path.isfile("/etc/twcmanager/config.json"):
    jsonconfig = open("/etc/twcmanager/config.json")
else:
    if os.path.isfile("config.json"):
        jsonconfig = open("config.json")
if jsonconfig:
    config = commentjson.load(jsonconfig)
else:
    logger.info("Unable to find a configuration file.")
    sys.exit()
# here is the exit!

Fixed: when I removed the "/" from lines...119,120
i.e.
if os.path.isfile("etc/twcmanager/config.json"):
    jsonconfig = open("etc/twcmanager/config.json")
and ran from TWCManager/ Directory Program Ran from then on....

PR #281 (2021-05-12): Make config not found an error.
This soon in the process, default loglevel is still in place, and this
means only warning and higher are printed to stderr.
Fixes #278
commit 241edb4b (2021-05-12): Make config not found an error.

This soon in the process, default loglevel is still in place, and this
means only warning and higher are printed to stderr.
Fixes #278
closed 2021-05-12 by ngardiner