ngardiner/TWCManager #282: Install fails on /main

Flitzer42, 2021-05-11:
Just tried to upgrade to the latest version...

> git pull
> git checkout main

Install failed with
...
```
Run "service lighttpd force-reload" to enable changes
sudo service lighttpd force-reload ; exit 0
# Install TWCManager packages
sudo pip install --upgrade setuptools
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip._internal import main
ModuleNotFoundError: No module named 'pip'
make: *** [Makefile:50: setup] Fehler 1
```
Tried to install pip with 

`sudo apt-get install python-pip`

But did not help.

Do you have any sugestions?
Remark: I am running TWCmanager on Raspberry PI OS LITE

Side effect 1: TWCManager cannot store settings: "Please check the permissions on /etc/twcmanager/settings.json and try saving your settings again."
Side effect 2: TWC cannot obtain the token from Tesla. An empty website opens instead.
Perhaps 2.) depends on 1,) and is casued by the incomplete installation.

Switched back to v1.2.1 (intermediately).

ngardiner, 2021-05-12:
Thanks for reporting this, I've added an additional condition to the Makefile which should stop this from happening for you now - the purpose was to automatically update setuptools on machines where we can detect that pip is installed. This avoids a different error that people get because the default version of setuptools on older systems doesn't support the way we do module naming.

So now it will skip that step if it finds neither pip nor pip3 executables. If you could try a ```git pull``` when you have a moment and run off of ```git checkout main``` branch and then ```make install``` again that would be very helpful.

Not sure what the cause of the permissions issue you see in 1) is, could you please paste the output of:

```
ls -ld /etc/twcmanager
ls -l /etc/twcmanager/config.json
```

Number 2 is likely related to 1, we store the token in settings.json in the same directory so if there's a permissions issue, it would affect both.

Flitzer42, 2021-05-12:
A small step for a man....
Updating my existing system failed again :-(.
```
# Install TWCManager packages
sudo pip install --upgrade setuptools
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip._internal import main
ModuleNotFoundError: No module named 'pip'
make: *** [Makefile:50: setup] Fehler 1
```

Here the requested output
```
pi@TWCManager:~/TWCManager $ ls -ld /etc/twcmanager
drwxrwxr-x 2 root pi 4096 Mai 11 19:48 /etc/twcmanager
pi@TWCManager:~/TWCManager $ ls -l /etc/twcmanager/config.json
-rw-r--r-- 1 root pi 27258 Apr 26 12:56 /etc/twcmanager/config.json
```

So I started from scratch with a blank system (OS LITE) and the installation worked well.
Perhaps one difference: On the previous system I did some power consumption optimizations (turned off wifi, HDMI, ...) but I would not expect that this makes any difference.
But... When installing on a blank system, it did not create the `twcmanager.service` file. So I was not able to just copy it into /systemd. For now, I copied the file from the v1.2.1 installation, hoping that this makes no difference :-)

It works, settings are saved, TWCManager starts after reboot, communication to TWC works, control via (dreprecated) IPC works fine, settings are stored, tresla token is generated, starts stop charging works. No extensive testing as we have no sunshine at night :-).

Nevertheless, here the outputs of the new system (seems to be the same as above)
```
pi@raspberrypi:~ $ ls -ld /etc/twcmanager
drwxrwxr-x 2 root pi 4096 Mai 12 21:24 /etc/twcmanager
pi@raspberrypi:~ $ ls -l /etc/twcmanager/config.json
-rw-r--r-- 1 root pi 28887 Mai 12 21:15 /etc/twcmanager/config.json
```

Observations (No issues)
1: Installing takes some minutes (feel like it was faster in the past), hangs on some steps for a while without progress but keeps going after some time.
2: As with the missing twcmanager.service, I tried to start it with `python3 -m TWCManager`  worked fine, but the output on the console contained the password for the tesla token in readable form.

Many thanks for suporting me.
Think we can close the issue, as it works on a blank system (except the missing twcmanager.system file).

ngardiner, 2021-05-19:
Thanks for the update @Flitzer42! I'm going to keep this open so I can address that printing of the password in plaintext that you've reported, as that shouldn't happen. Once I've found and fixed it, I'll close this off.

Noted on the install time observation, as well. I am not entirely sure why that is, but I'll have a look around and see if I find anything.

ngardiner, 2021-05-19:
Found it - there was an unfortunate left-over debug print statement which printed out the login details - apologies for that. It's fixed now. I'll close this off but I'll look into the install time and see what can be done, thanks!

closed 2021-05-19 by ngardiner