ngardiner/TWCManager #284: Home Assistant integration is weird

jherby2k, 2021-05-18:
Hi all, thanks for the hard work!

Got this all set up with Home Assistant very nicely. I have a concern, however - the current method of integrating with Home Assistant involves creating a user and having TWCManager connect and create / update entities in HA. This is unusual, and it sounds like this isn't even supported as per https://www.home-assistant.io/blog/2021/05/12/integrations-api/. Ideally TWCManager would make all this sensor data available over the REST API (I believe most of it is already - need to investigate further) and a proper TWCManager integration should be created for installation within Home Assistant.

I'd like to take a stab at building said integration, if you guys are interested. Its going to be a learning process for me - just want to ensure this will be accepted by the project!

ngardiner, 2021-05-19:
There's nothing particularly strange about how we integrate with HomeAssistant, we're using HTTP Sensors which are documented here:

https://www.home-assistant.io/integrations/http/

The advice on that page is:

_To use those kind of sensors or binary sensors in your installation no configuration in Home Assistant is needed. All configuration is done on the devices themselves. This means that you must be able to edit the target URL or endpoint and the payload. The entity will be created after the first message has arrived.

Create a Long-Lived Access Tokens in the Home Assistant UI at the bottom of your profile if you want to use HTTP sensors._

You'd be most welcome to create a HomeAssistant integration, but those integrations are entirely within the HomeAssistant project, it wouldn't have much to do with the TWCManager codebase other than using the HTTP API hooks that we expose. Ultimately the options are to either control the flow on our side or on the HomeAssistant side and of course from my perspective I'd prefer to define the behaviour and flow within TWCManager, as maintaining it within HASS means aligning with their standards, release schedules, etc and we haven't encountered anything that we couldn't do using the published HTTP endpoints that HASS makes available.

jherby2k, 2021-05-19:
Thanks. So i think the benefits of a proper component would be:

1. Ability to define things other than sensors (like switches - a "charge now" switch is the first thing that comes to mind)
2. HA entities would persist between reboots of HA (right now they disappear until TWCManager recreates them)

downside being its a separate codebase. We don't have to have it added to the HA release cycle though - I'd like to just publish as a custom component for now. That means either its a file copy or can be installed automatically via HACS. I don't imagine the TWCManager user base is large enough to warrant inclusion in HA proper.

ngardiner, 2021-05-19:
There's a way to implement a switch using a RESTful API endpoint:

https://www.home-assistant.io/integrations/switch.rest/

Which would be pretty easy to implement to call the chargeNow API endpoint that we expose today. We don't currently implement the GET request to get the current charging state but that would be super easy to add.

jherby2k, 2021-05-19:
Are you able to create switches without editing HA's configuration.yaml though? I am doing this right now for a switch.

```yaml
rest_command:
  twcmanager_charge_now:
    url: http://<twc_ip>:8080/api/chargeNow
    method: POST
    content_type: application/json
    payload: '{"chargeNowRate":32, "chargeNowDuration":{{ duration }}}'
  twcmanager_cancel_charge_now:
    url: http://<twc_ip>:8080/api/cancelChargeNow
    method: POST
    content_type: application/json

switch:
  - platform: template
    switches:
      twc_charge_now:
        friendly_name: TWC Charge Now
        value_template: "{{ is_state('sensor.twcmanager_current_policy', 'Charge Now') }}"
        turn_on:
          service: rest_command.twcmanager_charge_now
          data_template:
            duration: 28800
        turn_off:
          service: rest_command.twcmanager_cancel_charge_now
```

KirkKirk, 2021-05-28:
Can you help me please with the setup to get HA sensors.xxx into TWCM?
I followed instructions (created token as well) but no success. 
accessing to HA via browser https://192.168.1.x:8123. I use 2fa as well. Tried without 2fa with no success.
do I have to do something with HA config.yaml?

ngardiner, 2021-05-29:
@KirkKirk let's discuss this in the original issue, #290

jherby2k, 2021-07-23:
I've got a nice Status component working, but would like to be able to push consumption and generation settings to TWCManager as well so that the apiKey is no longer needed. It makes me nervous having it in config.json since HA has a lot of control over our home!

This is going to require an API on the TWCManager side.

Would you prefer a REST API for this, or would MQTT be more suitable? I see MQTT is already roughed-in.