ngardiner/TWCManager #285: (Re-)Startup Time

MikeBishop, 2021-05-18:
Perhaps minor, but I used to be able to kill the running TWCManager instance and start a new one (presumably with different code or config) while a charge was happening without interrupting the charge.  That seems to never happen now, I suspect because it's taking longer to start up.  Is there any work happening at startup that we could rearrange so that the RS485 handling starts sooner, to keep from making the TWCs stop from lack of a master heartbeat?

This is probably pretty minor; I wouldn't expend substantial effort on it, but it's something to keep an eye out for during anything which touches startup -- if we find any easy ability to delay things, that would be good.

ngardiner, 2021-05-19:
We should perhaps just make the RS485 module the first to initialize. I did move logging modules to the top recently, because if there's an error you want to have it logged or printed, but there's no reason we couldn't have console first module and RS485 second, then all the rest below that. We should probably just clearly spell out in TWCManager.py why it's that way so there's no future inadvertent re-ordering.

MikeBishop, 2021-05-19:
It's not just the module order, though.  The loop that sends and receives RS485 messages doesn't begin running until all modules have been loaded in the current design, even if the RS485 module is present.  #286 experiments with moving all module loading into the background thread, but yes, the interfaces need to be early.

ngardiner, 2021-05-20:
Fair enough - my only request is that we target this for the next version, I'm confident there are references to modules that won't cleanly re-initialize if the module wasn't available at the time (carapi, the new protocol module, possibly the status modules), and there's already some instability from some core changes in 1.2.2 that need to be ironed out. I'll open an issue to sync up on a plan for a new release so that MFA login is in a stable release.

MikeBishop, 2021-05-20:
That's fine -- I mostly did it as an experiment and (for the modules I use) found only a few minor things that broke and fixed them.  But obviously, it would need to be tried across lots of modules before we're comfortable with it.

ngardiner, 2021-06-03:
Just a thought in addition to the loading of the modules in the background thread (which we could certainly pursue going forward) - I think a big contributor to this situation was my insistence that we load each module first and then check the module has all of the pre-requisites it needs to operate within the module itself, or unload.

While I still feel this makes sense (because it avoids modules remaining resident if they aren't able to operate) I can also see how inefficient it is as we scale out, so I have a compromise in mind which I think will help.

The compromise is to add some logic which tries to pre-empt the module load by checking if there's an "enabled" flag in config where it should be if the module has a dedicated config section... for example:

```
if module.endswith("Status"} and config["Status"][restofmodulename][enabled] == false then skip
(....)
elif config["EMS"][modulename][enabled] == false then skip
```

Unfortunately module naming isn't perfect and neither is this logic but it will get rid of the low-hanging fruit by pre-emptively looking for an enabled parameter in config for that module's name and only if it exists and if it's false will we skip loading the module. Anything else will go through and get disabled (or enabled) in the current fashion

What are your thoughts?

closed 2022-02-15 by ngardiner