ngardiner/TWCManager #289: Strange Delivered kWh number

mikeljo, 2021-05-20:
i have v1.2.1, two TWC, TWCManager is runing on  Ubuntu on a Atom Laptop, NOT on a Raspi.

All works fine, until some days before i got this strange kWh count: (Cut from log file)
Slave TWC 00xx: Delivered 4194484 kWh, voltage per phase: (236, 240, 236)
Slave TWC 0xxx: Delivered 1587 kWh, voltage per phase: (234, 238, 240).
i think that 4194484 kWh is a little bit high.

I made a pull request today and remove the summ in the settings, reset the Box not change.

ngardiner, 2021-05-21:
The delivery value is a value we get from the TWC, and we parse. It's recorded locally, but that is not the source of the value - we query the TWC over RS485 and get the value from there.

Perhaps it's one of two things, either a parsing error or a geniunely incorrect value being returned by the TWC. The best way to tell would be from the Rx value we get from the TWC, if you could turn your debug all the way up to 10 and capture it for a while, I will reply shortly with the string that you can search for that will show what the TWC sent to us as the lifetime kWh value.

mikeljo, 2021-05-21:
ok, no Problem. 
Tell me the string to search for and i will reply with the TX Value

ngardiner, 2021-05-21:
The message we're looking for looks like this:

17:55:24 TWCManager 12 Rx@: (FC) FD EB XX XX *00 00 03 4A* 00 FA 00 00 00 00 00 00 00 00 00 DA
17:56:25 TWCManager 12 Rx@: (FC) FD EB XX XX *00 00 03 4A* 00 F8 00 00 00 00 00 00 00 00 00 D8

The XX XX is the TWC ID and the rest of the message gives us voltage per phase and lifetime kWh. The most important bytes are the ones I highlighted above, that's the value we derive lifetime kWh from. You can see in the above value:

```
00 00 03 4A
```

03 4A = 842 in Decimal

Therefore I get:

18:11:34 TWCManager 20 Slave TWC XXXX: Delivered 842 kWh

mikeljo, 2021-05-21:
i got this:
2021-05-21 10:22:29,755 - TWCManager 12 Rx@: (FC) FD E0 00 28 77 77 03 0C 80 00 00 00 00 00 00 85
2021-05-21 10:22:29,844 - TWCManager 12 Rx@: (FC) FD E0 04 25 77 77 03 0C 80 00 00 00 00 00 00 86
2021-05-21 10:22:29,902 - TWCManager 12 Rx@: (FC) FD EB 04 25 00 00 06 34 00 EB 00 F0 00 EE 00 17
2021-05-21 10:22:30,014 - TWCManager 12 Rx@: (FC) FD EB 00 28 00 40 00 B8 00 EF 00 EE 00 EE 00 00 00 00 00 D6
2021-05-21 10:22:31,985 - TWCManager 12 Rx@: (F8) FD E0 00 28 77 77 03 0C 80 00 00 00 00 00 00 85
2021-05-21 10:22:32,071 - TWCManager 12 Rx@: (FC) FD E0 04 25 77 77 03 0C 80 00 00 00 00 00 00 86
and this
2021-05-21 10:23:31,105 - TWCManager 12 Rx@: (FC) FD E0 04 25 77 77 03 0C 80 00 00 00 00 00 00 86
2021-05-21 10:23:31,141 - TWCManager 12 Rx@: (FC) FD EB 04 25 00 00 06 34 00 E9 00 EE 00 EE 00 13
2021-05-21 10:23:31,251 - TWCManager 12 Rx@: (FC) FD EB 00 28 00 40 00 B8 00 EF 00 ED 00 EC 00 00 00 00 00 D3
2021-05-21 10:23:33,219 - TWCManager 12 Rx@: (FC) FD E0 00 28 77 77 03 0C 80 00 00 00 00 00 00 85
2021-05-21 10:23:33,328 - TWCManager 12 Rx@: (F8) FD E0 04 25 77 77 03 0C 80 00 00 00 00 00 00 86
2021-05-21 10:23:35,243 - TWCManager 12 Rx@: (F8) FD E0 00 28 77 77 03 0C 80 00 00 00 00 00 00 85

so the ID 0425 reports: 06 34 = 1588
the ID 0028 reports 00 28 00 40 00 B8 this could be 184 or 4194488
For me i looks like that the String got i byte too much.

mikeljo, 2021-05-21:
Is it Possible to reset the Lifetimekwh in the twc? Cause 184 seems a little bit less-
and could it be that the ID is displayed on the website as is? no transform from hex to dec?

ngardiner, 2021-05-21:
There's not a way to reset it, but I don't think the value is wrong. It's consistently sending a longer string, with 3 extra bytes. Seems to me there's a newer revision of the message and we need to parse it properly. Is this a newer TWC by any chance?

ngardiner, 2021-05-21:
Actually, I take that back. One is speaking the V1 protocol and the other the V2, and the V2 is parsed the way we normally do, and produces the number you see (4194488 kWh). Something's wrong with the value stored in the TWC's flash, I would have to guess

mikeljo, 2021-05-21:
ok.
i don't think that this is a newer Hardware. It could be that there is a new Firmware on this one. Cause from this one i got the FIN from my Tesla when it is connected. From the other one (ID 0425) not. i will go outside and check the Hardware.

mikeljo, 2021-05-21:
i Check the hw. the look identical. switches are ok too. 
But i think 4 Million kWh is a "little bit" toooooo much. This is a 22kW charge 24hour per day for 7944 days!! This mean 21 Years full loaded!
Why is one speaking V1 and the other one V2? Normally the Tesla will update them when it is connected.

ngardiner, 2021-05-21:
Tesla doesn't update TWC firmwares - whatever firmware is flashed is what stays on the device. I'm afraid I can't tell you why it is that it's reporting that value, it's just not possible to say. We only represent what we get.

mikeljo, 2021-05-21:
Shure? My Information tells me that Tesla update the Firmware for TMC and UMC over the car when it is connected.
Let me have a look from where i got this info.

mikeljo, 2021-05-21:
I found the Update for the HPWC here:
https://teslamotorsclub.com/tmc/threads/charge-connector-update-in-progress.112886/

ngardiner, 2021-05-21:
They can update it from the vehicle, but they stopped doing it a long time ago (perhaps 2018). That said, there are reports of people getting a targeted update via the service option in the app, but as for a wide-scale update? Unfortunately not, there are still many people out there with older firmwares that can't take advantage of features like querying VINs.

mikeljo, 2021-05-21:
This make sense to me.
A bricked HPWC / TWC is not funny. With a UMC you can got to the Service centre.
I think i wait until the Reading "B8" increase over FF to see where the "1" appears.

mikeljo, 2021-05-21:
I set my log level back to 20 and restart the TWCManager.
And what do you think didi i see in the Log?
----
2021-05-21 12:55:44,449 - TWCManager 20 Set slave TWC 0028 protocolVersion to 2, minAmpsTWCSupports to 6.
2021-05-21 12:55:47,018 - TWCManager 20 Set slave TWC 0425 protocolVersion to 2, minAmpsTWCSupports to 6.
----
And now??

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop