ngardiner/TWCManager #293: [General support] Amps vs. kw when Tesla says "Low quality grid detected"

danielsan1, 2021-05-29:
Hi everyone,
I have three small issues and I hope you can help. 

1. I'm charging solar surplus via KOSTAL solar module which in general woks fine! I noticed one issue which in certain situations leaves a lot of energy unused: My Model 3 is very quick in saying "**Low quality grid detected**" and reducing the charging power. That results in less kw with the same amps e.g. 8A -> ~6kw charging (normal) vs. 8A -> ~4kw charging (low quality). But the TWCM still "thinks" with 8A it's providing ~6kw. (2kw unused) Any chance to fix this?

2. **About "FlexAmps"**: Im using "greenEnergyFlexAmps" (6A) but NOT "useFlexAmpsToStartCharge" but it's always starting anyway with 6A when I plug in the car... Did I miss another setting?

3. (Not crucial) About **"Charge Point" in web interface** (modern theme): I don't get any information on "Charging time counter", "Vehicle", "Current" and "Last". Also nothing in "Vehicle" tab. I there anything I can do?

Thanks in advance to anyone who has an idea or tip for me :)

ngardiner, 2021-05-29:
Hey @danielsan1,

I haven't encountered 1 before - perhaps it's an issue with Grid Voltage. Do you see a grid voltage reading at the time? It would be interesting to see what it reads when you see that. It is true that power is a factor of voltage x amps so if the voltage is low, then your output will be less for the same current, but we should be taking that into account as we convert watts to amps for adjusting power offered, so I'm not sure why you're seeing a mismatch there

For 2 - have you entered your Tesla API credentials into TWCManager? We can't stop and start the charge without API, so your car is going to start charging at 6A until we send it an API message to stop

For 3 - You're right, the modern UI is missing some things. The time counter is one. Vehicle missing means that the firmware version of your TWC doesn't support reading VINs. There's a chance you might be able to get it updated by Tesla by raising a service request via the app but it's a low chance, I haven't seen it yet. If you do go that route I'd avoid mentioning any 3rd party apps and perhaps just say that the charger needs a reset every now and then, that would be about the best excuse I could think of for a legitimate firmware upgrade :)

danielsan1, 2021-05-29:
Hey @ngardiner 
Thanks for your super fast reply! :) I added some screen shots where you can see the issue quite well I guess. It's 11A actually but it shows the "normal" 8kw in TWCM and the (low quality) 5kw in the Tesla app. (and Kostal shows the "left over kw")

For 2: I added the access & refresh token. But is there the need to send "stop"? Because I set "useFlexAmpsToStartCharge" to FALSE. So I expected the behavior that I plug the car in and until the surplus is 6A or more it won't start. But that's not the case. I starts immediately with 6A... even at night ;)

For 3: Thanks for the tip! But how would they do this? Send a ranger out? :) ...maybe I'll give it a try :)

Thanks agian!

[image]
[image]
[image]
[image]

danielsan1, 2021-05-31:
Hey @ngardiner are the screenshots helpful? Any idea on the “low quality grit” issue? Thanks again for your help!

MikeBishop, 2021-06-02:
> For 3: Thanks for the tip! But how would they do this? Send a ranger out? :) ...maybe I'll give it a try :)

The data connection between the car and the TWC carries firmware updates.  They'd push a package to your car OTA, and it would instruct the TWC to install it the next time you plugged it in.

ngardiner, 2021-06-03:
For 2 there's absolutely a need to stop :) The TWC is programmed to just start charging when a vehicle is connected, we work against that natural inclination by telling the vehicle to stop if we see that you are not generating enough power. Is it still not stopping after you log in? If so, try giving the car a reboot (the 3 buttoned salute). If it doesn't stop when we ask it to stop, the only backup function we have at the moment is to set the charge target to 50% SoC - we can't set it any lower. So if it charges to 50% and stops that means it was ignoring our command to stop charging.

ngardiner, 2021-06-03:
Also, I did some research on the error you're seeing and it is interesting. Apparently the source of that error message is that while the car is charging, your TWC is observing an abnormal drop in voltage when compared to the amps drawn, effectively it is detecting an abnormality that suggests that there's potentially some unexpected resistance or voltage drop on one or more of the phases, so it scales back charging.

To fix your issue, you may need to check the connections within the TWC and perhaps check for high temperatures or any signs of abnormality. Here's some threads where others have had the same issue:

https://teslamotorsclub.com/tmc/threads/low-quality-grid-detected-while-sitting-in-car-after-its-charged.147658/

https://www.reddit.com/r/TeslaModel3/comments/k7wp3z/low_quality_grid_detected_on_a_wall_connector/

https://forums.tesla.com/discussion/175615/low-quality-grid-detected

danielsan1, 2021-06-04:
Thanks @MikeBishop for your reply! I’ll give it a try with the Tesla SC as I will have an appointment next Tuesday anyway :) @ngardiner thanks for your research - that’s really interesting as I’m 99% sure that this relates to my solar installation. Also before I installed TWCM I saw this warning/behavior on cloudy days when the solar power was very “unevenly”. I’ll read through the links you shared.
And I think now I got how no.2 works :) So TWCM actively sends a “stop” command if surplus is below 5 amps when I plug in the car? Unfortunately it is still not working. I’ll try reentering all credentials and reset the car. How can I get back to enter my log in credentials instead of the tokens (what I did in the first place)?

bikeymouse, 2021-06-09:
Your issue with charging at lower power levels may be the same I am having when tracking solar and offering less then 8Amps to the car. The car (Model 3) then seems to switch off (after spiking) one of the phases. See[ here](https://github.com/ngardiner/TWCManager/issues/298).

ngardiner, 2021-06-09:
> Your issue with charging at lower power levels may be the same I am having when tracking solar and offering less then 8Amps to the car. The car (Model 3) then seems to switch off (after spiking) one of the phases. See[ here](https://github.com/ngardiner/TWCManager/issues/298).

I'd say that is unlikely - the issue here is pretty well documented if you search for it, if the vehicle detects abnormal voltage loss when drawing from the TWC, it reduces the charge rate entirely and shows a message on the MCU. Search on the error message provided and you'll find plenty of threads on the forums.

ngardiner, 2021-06-09:
> But nevertheless - why does TWCM shows the correct amps but wrong kw in these “low quality grit” moments? 🤔 (and as a result does not use the full available solar power?)

That error on the MCU is informing you that the vehicle has decided to draw less power from the TWC due to the "excessive" (in its view) voltage loss as the power draw increases. There's no working around it, as we offer a set amount of amps to the TWC for the vehicle to draw from but it decides to draw less power than we allow for. Posters on the forums theorise that it is because the car is concerned about heat/fire risk if there's excessive resistance or loose connections within the TWC and so backs off power draw to reduce risk.

Unfortunately I am not in a position to comment on how/why a solar install might relate to this. Generally your inverter will be in sync with grid voltage and so I wouldn't expect the generation to be behind it, but I am not sure of the logistics of your setup and how it may differ from others. If it is related to solar it's probably a false positive, ie the inverter probably just isn't able to supply stable voltage, but a TWC/Tesla interprets that as an issue with the power feed and subsequently throttles down the power draw on the vehicle side. There's a lot of things to consider such as the regulatory spec for voltages in your region and how much the voltage drop actually is under load, so it's very hard to estimate without a lot more data.

I'd call up Tesla service and see what they think. Maybe they've dealt with this issue before? If you have a contact on your solar inverter side, maybe ask them as well?

ngardiner, 2021-06-09:
> And I think now I got how no.2 works :) So TWCM actively sends a “stop” command if surplus is below 5 amps when I plug in the car? Unfortunately it is still not working. I’ll try reentering all [...]

That's right. It's strange that it's not working, because if you're not seeing a prompt for your credentials, it means we logged in successfully and we should be able to start/stop your vehicle's charging. If you have a look in the log output (and if you have debugLevel set to 8), you should see something like:

```
21:28:57 TeslaAPI   13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/xxxx <Response [200]>
21:28:57 TeslaAPI   13 Car Name not ready because it wasn't woken in the last 2 minutes.
```

Or similar, which is TWCManager communicating with your vehicle via Tesla API.

If you need to reset the current credentials so the login page appears again, you can do that on the Settings tab, by putting in anything in the token boxes and pressing save. This will overwrite your current token and bring back the login prompt. Restart TWCManager after that and you'll get the Tesla login prompt again.

bikeymouse, 2021-06-09:
> I'd say that is unlikely - the issue here is pretty well documented if you search for it, if the vehicle detects abnormal voltage loss when drawing from the TWC, it reduces the charge rate entirely [...]

Well my point is that - at least in my case - I think the "grid" error is not correct.  
I'm logging the voltage independently every 10 seconds and the voltage is really within all acceptable ranges. 
Also the error only occurs when charging at low amps, which is completely the opposite of what you'd expect when you have voltage losses due to e.g. a loose connection or incorrect cabling. If I look at the links to these errors, you can also see that none of the 3 actually discovered anything wrong with the cabling/voltage.

Because @danielsan1  also experienced the same error (and mentions this happening in lower range of charging between 4 - 8 kw) where this occurs) and a car that charges with less power then offered, my guess was that also in his case this is caused by  1 of the phases being switched off.

danielsan1, 2021-06-10:
Hey guys thanks for all the replys!
@ngardiner I’ll check the logs and will try to resent the credentials - let’s see if it helps :)
@bikeymouse Yes that sounds very similar… I have the issue on low amps and (I have the feeling) when solar power is fluctuating because of clouds. (Like go to 10A drop to 5A go to 8A and so on)
At the Tesla SeC they told me that I should call the Tesla support via phone. Because they would be the best address to help me out with the firmware or TWC diagnostic… not sure if I should do this because they won’t find a “real issue” ;)

ngardiner, 2021-06-11:
> Well my point is that - at least in my case - I think the "grid" error is not correct.

You are getting the low quality grid error when the phases turn off? If so, I'd highly recommend updating the details in your issue to reflect this - I re-read your description just now and it specifically states that there's no voltage drop on the phases and that you don't believe that is the cause, yet this issue here is about an error that reflects excessive voltage drop and causes the vehicle to throttle back.

> my guess was that also in his case this is caused by 1 of the phases being switched off.

Not according to the sources I linked to earlier. The sources state that the car will draw less than the offered power when the Low Quality Grid error message is shown.

bikeymouse, 2021-06-11:
> > Well my point is that - at least in my case - I think the "grid" error is not correct.
> 
> You are getting the low quality grid error when the phases turn off? If so, I'd highly recommend updating the details in your issue to reflect this 

Well, I _did_ mention that in my [update](https://github.com/ngardiner/TWCManager/issues/298#issuecomment-857588485) I posted 2 days ago.

> I re-read your description just now and it specifically states that there's no voltage drop on the phases and that you don't believe that is the cause, yet this issue here is about an error that reflects excessive voltage drop and causes the vehicle to throttle back.

What I meant to say, is that in my opinion the car _thinks_ that there is a voltage drop, while the measurements show there is not. I have added some more info in [the issue](https://github.com/ngardiner/TWCManager/issues/298#issuecomment-859658279) why I think so.

> > my guess was that also in his case this is caused by 1 of the phases being switched off.
> 
> Not according to the sources I linked to earlier. The sources state that the car will draw less than the offered power when the Low Quality Grid error message is shown.

Yes, sure that's what I see: if the car _thinks_ there is something wrong it spikes to zero and finally completely shuts down a phase. 
The question is why does it concludes that there is a grid issue, because I really I don't believe I have. 
I'm wondering it that could have something to do with TWC-Manager telling the TWC to charge at low Amps and then some communication error between TWC-Manager and TWC or TWC and car?

ngardiner, 2021-06-12:
> I'm wondering it that could have something to do with TWC-Manager telling the TWC to charge at low Amps and then some communication error between TWC-Manager and TWC or TWC and car?

If that were the case, it would be affecting all the other users we have. The only reports we've had of this error are in this thread and yours.

danielsan1, 2021-06-12:
Also don't think that it has something to do with TWCManager as I have this issue since I’m charging with my solar roof in general. 
I just had the hope that there might be “a trick” how TWCM can avoid the issue or even bring the car back to charge on full power again. Because before TWCM I “fixed” the issue going to my car and increase the amps manually via the display to use the full provided kw again. (e.g. normal 8A ~6kw / low quality 12A ~6kw) You know what I mean?

ngardiner, 2021-06-12:
@danielsan1 yes, the simple answer is that it is technically feasible.

The concern however is whether it's rational and/or right for us to do. If the majority of vehicles showed this error then we might find ourselves in a position in which we need to compensate for the difference to charge effectively, but outside of the two current threads (that we may need to consolidate into one if this discussion is to go on for some time) this is not the experience of TWCManager users.

I understand the desire for a quick software fix and I've been there... it's not fun being in the position of hearing that your hardware install isn't performing the same as others and its always nice to think there's just a config setting to offset it, but....

This is a protective mode that the vehicle is entering. We're not sanctioned in any way to be using the TWC the way we are, to then program an override for a safety feature rather than just contacting Tesla / an electrician is a bridge too far in my eyes. You may be right, your install might just be that one in 100 that acts differently, but if that's the case please get Tesla to implement the override on the TWC so you're not putting your warranty at risk by overriding a safety setting.

I do recognize that if it just drops the draw to (for example) 2/3 of the specified offer, that's no different to us just increasing our offer by 2/3. I don't know why they implemented it that way. It doesn't seem right, but once again, if we just code up a hack to get you past a safety feature and it turns out your TWC install was like this:

[image]

https://teslamotorsclub.com/tmc/threads/low-quality-grid-detected-while-sitting-in-car-after-its-charged.147658/

It could just be a bigger problem waiting to happen. And if it's just load or a false positive, and Tesla are the vendor of the TWC equipment, they can provide a fix and you won't need a workaround.

danielsan1, 2021-06-13:
@ngardiner thanks for your detailed reply! I’m totally with you - we shouldn’t override a safety feature. Especially a safety feature with this specific purpose and the false positive behavior in these little cases. 
I still have to investigate the issue with the Tesla credentials and the immediate charging even if the solar surplus is way below 5A. (That’s more of an issue for me than the low quality grid error to be honest ;)

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.