ngardiner/TWCManager #294: Car not charging: kW vs. Ampere / 1 phase vs. 3 phases

Saftwerk, 2021-05-29:
The house is grid connected 3 phases each 50A.
The TeslaWallConnector is connected 32A and 3 phases, because visitor cars could charge with 3 phases.
The Photovoltaik is ~5,5 kWp only 1 phase up to 24A.
The car is a Tesla Roadster. It uses 1 phase only from 10A up to 70A @ 230V 50Hz.
The Grid meter reading is only a single value: "kW total" (pos.value = Consumption or neg.value = Solar surplus)

Today the TWCManager shows:
```
Solar Surplus = 3.51 kW / 4.86A
"Charges a vehicle if Solar production is 5 Amps or more than the current consumption"
```
Result: The Tesla Roadster **did not charge**. But it was possible: 3.51 kW / 230V = 15A single phase.
What setting is wrong?
Thank you for any help!

```
{# sudo nano /etc/twcmanager/config.json
 # File from Saftwerk 2021-05-22
"config": {
    "settingsPath": "/etc/twcmanager",
    "wiringMaxAmpsAllTWCs": 32,
    "wiringMaxAmpsPerTWC": 32,
    "minAmpsPerTWC": 10,
    "onlyChargeMultiCarsAtHome": true,
    "defaultVoltage": 230,
    "numberOfPhases": 1,
    "greenEnergyAmpsOffset": 0,
    "greenEnergyFlexAmps": 8,
    "useFlexAmpsToStartCharge": false,
    "subtractChargerLoad": false,
    "cloudUpdateInterval": 1800,
    "logLevel": 13,
    "displayMilliseconds": false,
    "fakeMaster": 1
},
"interface": {...},
"control": {
  "HTTP": {
    "enabled": true,
    "listenPort": 8181
  },
},
"policy":{
  "engine":{
    "policyCheckInterval": 30
  },
  "extend":{
    "emergency":[],
    "before":[],
    "after":[],
    "restrictions":{},
    "webhooks": {}
  },
  "override": []
},
"logging":{...},
"sources":{
    "Volkszahler": {
      "enabled": true,
      "serverIP": "pi3.fritz.box",
      "serverPort": "80",
      "uuid": "00000004-0001-1670-ffff-0123456789ab"
    }
},
"status": {...}
}
```

By the way a small question:
Is there a documentation of that State numbers?
Maybe it's a good idea to replace this "number" by a "word".
[image]

Saftwerk, 2021-05-29:
I think the problem is here:
Before charging would start, the TWCManager can never know, how many phases the car will use.

So the TWCManager can not calculate the correct Amps to send to the Wallconnector.
Solar surplus 5kW / 230V / 3 phases = 7,2A
Solar surplus 5kW / 230V / 2 phases = 14,5A
Solar surplus 5kW / 230V / 1 phase = 21,7A

A possible solution:
The Settings page get a new parameter "CarUseNumberOfPhases" =1/2/3.
So the TWCManager can calculate the correct Amps available to the Wallconnector and then decide to "_Charge a vehicle if Solar production is 5 Amps or more than the current consumption_"

Saftwerk, 2021-05-30:
Here a look into the log:
```
13:16:59 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
13:16:59 Policy     13 Policy conditions were not matched.
13:16:59 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
13:16:59 Policy     13 Policy conditions were not matched.
13:16:59 Policy     13 Evaluating Policy match (tm_hour [13]), condition (gte), value (6)
13:16:59 Policy     13 Evaluating Policy match (tm_hour [13]), condition (lt), value (20)
13:16:59 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [9]), condition (lte), value (13)
13:16:59 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
13:16:59 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
13:16:59 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1622373419.9515917 or empty carApiBearerToken ''
13:16:59 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
13:16:59 TWCManager 20 Green energy generates 4115W, Consumption 0W, Charger Load 0W
13:16:59 TWCManager 20 Limiting charging to 5.80A - 0.00A = 5.80A.
13:16:59 TWCManager 20 Charge when above 10A (minAmpsPerTWC).

```
Conclusion: There is a mismatch between
`13:16:59 TWCManager 20 Charge when above 10A (minAmpsPerTWC).`
and
`Charges a vehicle if Solar production is 5 Amps or more than the current consumption`

Until somebody finds a solution, I will multiply the "Generation" value by factor 3.

ngardiner, 2021-05-30:
We already have a setting in the config file for the number of phases, and you have it set to 1:

```
    "numberOfPhases": 1,
```

If your TWC has a new enough firmware that it reports voltage per phase to us, we detect the number of phases from that output, if not we use the config in the config file. You should have this set to 3, it seems your TWC is not reporting voltage per phase to us.

Saftwerk, 2021-05-30:
Yes, the TWC reports 3 Voltages: 
[image]

ngardiner, 2021-05-30:
Okay, that means it is detecting the number of phases from your TWC, so it would override any value you have in the config file.  This is necessary because when we talk about amperage, we mean amperage offered to the vehicle by the TWC, not the equivalent single-phase amperage generated by the solar.

So 5A minimum is 5A 3-phase, and you've got:

```
    "minAmpsPerTWC": 10,
```

Which means the minimum supported amps per TWC is 10A 3-phase. This should be your physical minimum, either 5A for North American systems or 6A for European. Setting this to 10A means you need the equivalent of 10A of 3-phase generation to start charging, regardless of what the vehicle is attached we can only control amperage offered, and only in the context of the connectivity (ie amperage in single phase context for a single phase connected TWC, or 3-phase context for 3-phase connected).

Saftwerk, 2021-05-30:
For me this is hard to understand.
You say i need "10A 3-phase" from solar to start charge a car which is using only "10A 1-phase"?
But then there would be a unused surplus of "10A 2-phase" ?!?

ngardiner, 2021-05-30:
> For me this is hard to understand.
> You say i need "10A 3-phase" from solar to start charge a car which is using only "10A 1-phase"?
> But then there would be a unused surplus of "10A 2-phase" ?!?

It's quite simple, we aren't setting the maximum draw for the car, we are talking to your TWC and we offer a maximum amperage that your TWC may offer to vehicles via the load-sharing protocol. We can't specify to charge using only a single phase, I wish we could. The closest I have seen to this is a user of TWCManager who has a relay to switch off the 2nd and 3rd phases, but outside of that we have no control over whether a 1-phase vehicle is connected to a 3-phase charger, we can still only specify amperages in relation to the 3-phase charger input.

It is feasible that we could detect if a vehicle that has this limitation is charging, but we would have to figure out how - if you are getting a VIN reading from your TWC then we could mark vehicles as being single phase only and multiply the offered amps by the number of phases as long as that vehicle is connected. But this is not a low-risk configuration, it could cause someone's installation to draw 3 times their rated circuit power if we incorrectly detected a single phase vehicle- it has nothing to do with TWCManager that we cannot instruct a TWC to charge using a single phase nor to specify the maximum amperage for a single phase only if all 3 are connected, that is not how your TWC works - and it's very rare to come across vehicles with only single phase charging capability, so I think your expectations are a little high - if you go and set your charger to 10A using the rotary switch it will act exactly the same way, ie it will charge at only 10A on one phase when that vehicle is connected, but 10A 3-phase for all other vehicles.

Also, please remember:

> You say i need "10A 3-phase" from solar to start charge a car which is using only "10A 1-phase"?

Because you have configured it that way. It is explained in the config file:

```
        # Most EU chargers are connected to 230V, single-phase power which means 12A is
        # about the same power as in US chargers. If you have three-phase power, you can
        # lower minAmpsPerTWC to 6 and still be charging with more power than 12A on
        # single-phase.  For example, 12A * 230V * 1 = 2760W for single-phase power, while
        # 6A * 230V * 3 = 4140W for three-phase power. Consult an electrician if this
        # doesn't make sense.
```

Set it to a lower value if you don't intend for 10A 3-phase to be the minimum amperage that you would like to charge *any* vehicle with (even if it doesn't support 10A 3-phase, that will be the minimum surplus power that we offer to ***any*** vehicle as we are instructing the charger as to what to offer, not the vehicle, and so as far as we are concerned everything offered is via a 3-phase charger and therefore the equivalent of 3-phase 10A.

MikeBishop, 2021-06-02:
The other problem you'll run into, even if you lower the minimum to start charging, is that the current offered will equate to a third of the power you're actually wanting to give to the car.  If the Roadster is your only vehicle, the simplest option might be to disconnect the other two phases from the TWC.  If you also have a three-phase-capable vehicle to support, that's... difficult to support.

Not impossible, as @ngardiner notes, but difficult and not currently supported.

Saftwerk, 2021-06-03:
I do not see a overload problem.
It cannot be occur since the parameters `wiringMaxAmpsAllTWCs` and `wiringMaxAmpsPerTWC` are correct.
If a 3-phase car would connect, then there is never a overload on any wire, because all 3 wires are able to transport Amps up to wiringMaxAmps.

For tryout i used a fixed value for number of phases for the current sent to TWC.
File:: TWCMaster.py Line 614:
Original: `return (total / (phases * len(slavesWithVoltage)), phases)`
Tryout: `return (total / (phases * len(slavesWithVoltage)), self.config["config"].get("numberOfPhases", 1))`

It worked well and correct with a 1-phase car.

Only the calculation of `Amps offered` would be 3 times too high during Solar surplus direct after start charging. Then the `de novo` calculator will reduce until power is balanced or the current drops below minimum and charging stops.
Conclusion: A 3-phase car would have a swing charge start, stop, start, stop ... until the TWCManager could detect the discrepancy in `Watt used` versus `Amps used * Phases`.
In that case the 1-phase setting would be wrong.
If only the same car is charging then this is not a problem as long the setting is correct.

----
Well, could TWCManager find out if the car is using 1 or 3-phase automatically?
There is this `de novo` (in Amps) formula:
newOffer = currentOffer + self.convertWattsToAmps(generationW - consumptionW)

**So let's think about this:**
There MUST always be the balance Total Generation = Total Consumption including Charging.

- `de novo` (in Watt): newOffer_watt = currentOffer_watt + (generationW - consumptionW)
- `ChargingPower_used_by_3_phase_car` = Amps in use * (VoltageL1 + VoltageL2 + VoltageL3) 
- `ChargingPower_used_by_1_phase_car` = Amps in use * VoltageL1

Now you can compare the `Offer` with `ChargingPower_used_by_1_phase_car` and `ChargingPower_used_by_3_phase_car`. Any of the two values should almost match and you know the phases of the car.

Oherwise the balance Total Generation = Total Consumption would be wrong.

ngardiner, 2021-06-03:
> I do not see a overload problem.
> It cannot be occur since the parameters `wiringMaxAmpsAllTWCs` and `wiringMaxAmpsPerTWC` are correct.

You have more faith in these values being universally correct than I do - and I'd avoid testing the theory if not entirely necessary.

> For tryout i used a fixed value for number of phases for the current sent to TWC.
> File:: TWCMaster.py Line 614:
> Original: `return (total / (phases * len(slavesWithVoltage)), phases)`
> Tryout: `return (total / (phases * len(slavesWithVoltage)), self.config["config"].get("numberOfPhases", 1))`

There's a lack of context here, the second line that you use (which is our fallback for if we can't detect voltage per phase in older firmware versions) fetches the current value in the config file **or** specifies 1 if there is no value in the config file. You showed a snippet above where this was set to 3 in the config file, is that still the case? If so, this should have been the same as the detected 3 phases in the original code.

> Only the calculation of `Amps offered` would be 3 times too high during Solar surplus direct after start charging. Then the `de novo` calculator will reduce until power is balanced or the current [...]

Where are we getting watts used from? Not every EMS setup provides us with total consumption and even if so, it's not aligned with amps used (which we read from the TWC) at all, it's just one big consumption reading, so we never know how many watts the charger draws, we just infer it by multiplying the amps used value provided by the TWC by the voltage provided by the TWC. There's also no amps used * phases, it's just amps used, ie a 3-phase TWC has a single value and so does a 1-phase TWC and they are not multiplied or divided in either case, they are either the number of amps used on 1 phase (for a physical 1 phase install) or number of amps used on each phase (for a physical 3 phase install).

> **So let's think about this:**
> There MUST always be the balance Total Generation = Total Consumption including Charging.
>
>     * `de novo` (in Watt): newOffer_watt = currentOffer_watt + (generationW [...]

If only life were that simple. Some people have a meter which sits between household meter, grid and inverter, with or without the charger metered. In those cases that's fine, you could balance the offering until consumption == generation (as long as we meet all the other criteria such as minimum supported amps per TWC, maximum per TWC, maximum per circuit etc). But many others have the meter between grid and meter box (which only captures export) or no meter at all, and not everybody uses TWCManager to just charge with all generated energy... we have a whole other discussion in the section where people want to only use energy that is above an export threshold, plus we have plenty of people who use batteries and this would not work at all for their use case. We can't just default to assuming that consumption == generation and breaking all other use cases, and we can't make that the default behaviour just to support some very edge case older Teslas if it's going to break things for everyone else.

There's also the problem that the average setup is just not going to be accurate enough to measure in the way you're presenting. Sure, it's a nice thought that we could just check if there's effectively (phase 2 * amps offered) + (phase 3 * amps offered) difference between consumption and generation but.. what if there's not? What if there's lag between the two readings or the refresh time causes the values to diverge?  Exactly how much fuzz do we need to add to that calculation and how do we stop it from triggering simply because a car fails to respond to our increased offer (which happens - hence why we already need to spike amps to get the car to match our offer) or when the vehicle is ramping up, or if it's ramping down. 

Thankfully we have a feature in TWCManager where you can override policy in your config file to have your policy do this if you think it would best suit your environment, and that means we don't have to find a way to make single-phase charging work in all cases, only for those who are affected (which would be a very small footprint given the roadster market). I'll be honest that I don't know exactly how I'd write such a policy, but it can't be too hard to have a policy definition which essentially says if consumption < generation and a vehicle is charging, increment amps offered, and if there isn't a way to do it in the current policy framework I think it would be pretty easy to add whatever is needed.

I don't see any need to check the consumption against the generation to detect that only 1 out of 3 phases is in use, because the logic above regarding charging until consumption =~ generation is simple enough that it would work in your case.

ngardiner, 2022-08-26:
Closing some stale issues, please feel free to re-open if still ongoing.

closed 2022-08-26 by ngardiner