ngardiner/TWCManager #296: Do not query Tesla API with Current VIN string is: ['', '', '']

Saftwerk, 2021-05-30:
Here is a log from a connected Tesla Roadster:
The command `python -m TWCManager` was executed during the Roadster was sleeping.
At `15:09:11` the Roadster was waked up manually. (I'm still searching, where I can insert my command to wake up by TWCManager)
Then the log is flooded with "Current VIN string is: ['', '', ''] at part..."
```
15:06:04 TWCMaster  14 Registered module ConsoleLogging
15:06:04 FileLoggin 40 Permission Denied error opening logfile for writing
15:06:04 TWCMaster  14 Registered module FileLogging
15:06:04 TWCMaster  14 Released module SentryLogging
15:06:04 TWCMaster  14 Released module CSVLogging
15:06:04 TWCMaster  14 Released module MySQLLogging
15:06:04 TWCMaster  14 Released module SQLiteLogging
15:06:04 TWCMaster  14 Registered module TWCProtocol
15:06:04 TWCMaster  14 Released module Dummy
15:06:04 TWCMaster  14 Registered module RS485
15:06:04 TWCMaster  14 Released module TCP
15:06:04 TWCMaster  14 Registered module Policy
15:06:04 TWCMaster  14 Registered module TeslaAPI
15:06:04 TWCMaster  14 Released module WebIPCControl
15:06:04 HTTPContro 20 Serving at port: 8181
15:06:04 TWCMaster  14 Registered module HTTPControl
15:06:04 TWCMaster  14 Released module MQTTControl
15:06:04 TWCManager 40 ImportError: No module named 'lib.TWCManager.Control.OCPPControl', when importing Control.OCPPControl, not using Control.OCPPControl
15:06:04 TWCMaster  14 Released module Efergy
15:06:04 TWCMaster  14 Released module Enphase
15:06:04 TWCMaster  14 Released module Fronius
15:06:04 TWCMaster  14 Released module HASS
15:06:04 TWCMaster  14 Released module Kostal
15:06:04 TWCMaster  14 Released module OpenHab
15:06:04 TWCMaster  14 Released module OpenWeatherMap
15:06:04 TWCMaster  14 Released module SmartMe
15:06:04 TWCMaster  14 Released module SmartPi
15:06:04 TWCMaster  14 Released module SolarEdge
15:06:04 TWCMaster  14 Released module SolarLog
15:06:04 TWCMaster  14 Released module TeslaPowerwall2
15:06:04 TWCMaster  14 Released module TED
15:06:04 TWCMaster  14 Registered module Volkszahler
15:06:04 TWCMaster  14 Released module HASSStatus
15:06:04 TWCMaster  14 Released module MQTTStatus
15:06:04 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
15:06:04 TWCMaster  13 Send master linkready1
15:06:04 TWCMaster  13 Send master linkready1
15:06:04 TWCMaster  13 Send master linkready1
15:06:04 TWCMaster  13 Send master linkready1
15:06:05 TWCMaster  13 Send master linkready1
15:06:05 TWCMaster  13 Send master linkready2
15:06:05 TWCMaster  13 Send master linkready2
15:06:05 TWCMaster  13 Send master linkready2
15:06:05 TWCManager 20 32.00 amp slave TWC 2388 is ready to link.  Sign: 63
15:06:05 TWCManager 20 Set slave TWC 2388 protocolVersion to 2, minAmpsTWCSupports to 6.
15:06:05 TWCMaster  13 Send master linkready2
15:06:05 TWCManager 20 32.00 amp slave TWC 2388 is ready to link.  Sign: 63
15:06:05 TWCMaster  13 Send master linkready2
15:06:05 Policy     13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
15:06:05 Policy     13 Policy conditions were not matched.
15:06:05 Policy     13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
15:06:05 Policy     13 Policy conditions were not matched.
15:06:05 Policy     13 Evaluating Policy match (tm_hour [15]), condition (gte), value (6)
15:06:05 Policy     13 Evaluating Policy match (tm_hour [15]), condition (lt), value (20)
15:06:05 Policy     13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [9]), condition (lte), value (15)
15:06:05 Policy     14 All policy conditions have matched. Policy chosen is Track Green Energy
15:06:05 Policy     20 New policy selected; changing to Track Green Energy
15:06:05 TWCSlave   20 SHB 2388: 08 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
15:06:05 TWCSlave   20 Slave power for TWCID 2388, status: 8
15:06:06 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:06:06 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1622379966.0146017 or empty carApiBearerToken ''
15:06:06 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
15:06:06 TWCManager 20 Green energy generates 0W, Consumption 0W (Other Load 0W, Charger Load 0W)
15:06:06 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
15:06:06 TWCManager 20 Charge when above 10A (minAmpsPerTWC).
[... 424 lines cut ...]
15:10:42 TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
15:10:42 TeslaAPI   13 car_api_available returning False because of recent carApiLasterrorTime 1622380242.8977132 or empty carApiBearerToken ''
15:10:42 TeslaAPI   13 applyChargeLimit return because car_api_available() == False
15:10:42 TWCManager 20 Green energy generates 9232W, Consumption 0W (Other Load 0W, Charger Load 0W)
15:10:42 TWCManager 20 Limiting charging to 13.09A - 0.00A = 13.09A.
15:10:42 TWCManager 20 Charge when above 10A (minAmpsPerTWC).
15:10:42 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:42 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:44 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:44 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:44 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:44 TWCManager 15 Current VIN string is: ['', '', ''] at part 1.
15:10:45 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:45 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:46 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:46 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:47 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:47 TWCManager 15 Current VIN string is: ['', '', ''] at part 1.
15:10:47 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:47 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:48 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:48 TWCManager 15 Current VIN string is: ['', '', ''] at part 2.
15:10:49 TWCManager 15 Slave TWC 2388 reported VIN data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00.
15:10:49 TWCManager 15 Current VIN string is: ['', '', ''] at part 1.
```

ngardiner, 2021-05-30:
That is because the vehicle is not reporting a VIN, and we retry several times to query a VIN for a connected vehicle.

Saftwerk, 2021-05-30:
Yes. Tesla Roadster has no communication.
But keep in mind all other nonTesla cars will have similar effect.
You decide: Close this topic if it is not an issue.

ngardiner, 2021-05-31:
That's not quite right - for a non-Tesla car the TWC will return all zeros for the VIN, whereas what's happening with the Roadster is that it returns all zeros for the first 2 segments of the VIN string, and does not return anything for the 3rd (making us think it missed the 3rd command and re-sending it)

So whilst we already treat non-Tesla vehicles a specific way, the Roadster has different behaviour, and it's harder to catch given it isn't responding to the 3rd VIN segment command. I am not sure why, perhaps the TWC can sense some sort of Tesla CAN communication and gets stuck waiting for the Vehicle, it is hard to say.

I have an idea about how I could dampen this, so I'll leave the issue open, but it's clearly very rare as we've never had a report of an original roadster being used with TWCManager before, so it may need some further consideration.

Saftwerk, 2021-05-31:
Just good to know: This Roadster has Firmware 4.6.5 and HW Pkg:40Von unterwegs gesendet-------- Originalnachricht --------Betreff: Re: [ngardiner/TWCManager] Do not query Tesla API with Current VIN string is: ['', '', ''] (#296)Von: Nathan Gardiner An: ngardiner/TWCManager Cc: Saftwerk ,Author 
That's not quite right - for a non-Tesla car the TWC will return all zeros for the VIN, whereas what's happening with the Roadster is that it returns all zeros for the first 2 segments of the VIN string, and does not return anything for the 3rd (making us think it missed the 3rd command and re-sending it)
So whilst we already treat non-Tesla vehicles a specific way, the Roadster has different behaviour, and it's harder to catch given it isn't responding to the 3rd VIN segment command. I am not sure why, perhaps the TWC can sense some sort of Tesla CAN communication and gets stuck waiting for the Vehicle, it is hard to say.
I have an idea about how I could dampen this, so I'll leave the issue open, but it's clearly very rare as we've never had a report of an original roadster being used with TWCManager before, so it may need some further consideration.

[
{
"target": "https://github.com/ngardiner/TWCManager/issues/296#issuecomment-851086989",
"url": "https://github.com/ngardiner/TWCManager/issues/296#issuecomment-851086989",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]