ngardiner/TWCManager #297: Volkszähler: "Production" AND "Consumption" are >0

Saftwerk, 2021-05-30:
Normal there is only value higher than zero: "Production" OR "Consumption", but I have seen moments with both higher than zero.

In SolarLog.py i can see `getConsumption(self)` and `getGeneration(self)` both are calling `self.update()`.
But in `Volkszahler.py` the function getConsumption(self) does not call `self.update()`.
Is `self.update()`. missing at line 49 of `Volkszahler.py`?

Volkszahler.py (Lines 44-70)
```
    def getConsumption(self):

        if not self.status:
            logger.debug("EMS Module Disabled. Skipping getConsumption")
            return 0

        # While we don't have separate generation or consumption values, if
        # the value is a positive value we report it as consumption
        if self.generatedW < 0:
            return self.generatedW * -1
        else:
            return 0

    def getGeneration(self):

        if not self.status:
            logger.debug("EMS Module Disabled. Skipping getGeneration")
            return 0

        # Perform updates if necessary
        self.update()

        # Return generation value
        if self.generatedW > 0:
            return self.generatedW
        else:
            return 0
```

SolarLog.py (Lines 46-68)
```
    def getConsumption(self):

        if not self.status:
            logger.debug("SolarLog EMS Module Disabled. Skipping getConsumption")
            return 0

        # Perform updates if necessary
        self.update()

        # Return consumption value
        return self.consumedW - self.excludeConsumedW

    def getGeneration(self):

        if not self.status:
            logger.debug("SolarLog EMS Module Disabled. Skipping getGeneration")
            return 0

        # Perform updates if necessary
        self.update()

        # Return generation value
        return self.generatedW
```

Saftwerk, 2021-05-31:
Tomorrow I try to fix "At night `Solar surplus` is not updating".

File: "Volkszahler.py"
Function: "getConsumption(self):"
Line 49 added: "self.update()"

Issue: "`Generation = 0` AND `Solar surplus > 0`"
After reloading the website the `Solar surplus` shows correct `0`.
[image]

ngardiner, 2021-06-03:
I'm not sure that it's going to fix all that you have observed (since the update function is the same for Generation and Consumption and has a timer which only allows it to run once every 10 seconds) but it is entirely plausible that we are reading consumption values prior to generation values and therefore serving up a value from ~10 seconds ago, so I have updated this in bd032a9.

I hope it does fix the issues you're seeing.

Saftwerk, 2021-06-04:
Since I have Volkszähler enhanced to get separate values for Generation and Consumption this issue is obsolete and can be closed.
see here: https://github.com/ngardiner/TWCManager/pull/300

closed 2021-06-04 by Saftwerk