ngardiner/TWCManager #299: Tesla API not working

juanjoqg, 2021-05-31:
Hi,

Anyone knows if there is any new change on the tesla API?, the token request is not working for me.

Regards.

ngardiner, 2021-05-31:
Unfortunately, Tesla has again modified the login flow to add a captcha challenge which is breaking the auth flow for 3rd party apps.

ngardiner, 2021-05-31:
Some discussion is here: https://github.com/timdorr/tesla-api/discussions/390

ngardiner, 2021-05-31:
For iOS users, the following app can assist with logging in and capturing your API keys, which can then be inserted into the input boxes on the Settings page: https://apps.apple.com/us/app/auth-app-for-tesla/id1552058613#?platform=iphone

juanjoqg, 2021-05-31:
Thanks, I'll keep an eye on this

ngardiner, 2021-06-02:
A brief reprieve - Tesla appear to have removed the captcha (for now)

ngardiner, 2021-07-20:
Captcha appears to be back. Some users may be unable to log in currently, if so, it's because Tesla have applied a captcha requirement to your account and we cannot yet handle the captcha in our login process.

ngardiner, 2021-07-20:
Commit 3b77656 adds detection of captcha prompt from Tesla Auth. It's backwards-compatible, if we don't see a captcha request we just continue on, if we do see a Captcha request we redirect to a form to type in the captcha code. 

Note that this is not a complete captcha flow - we still need to then take that code, resume the login flow and include the code in the login flow. We're closer to it working.

ngardiner, 2021-07-20:
13010ab is the other half of the logic. It takes the captcha code and returns back to the login flow. I've just successfully logged in with an MFA account using this flow. I'll do some more testing, but I think we're set for Captcha handling

ngardiner, 2021-07-20:
I've now tested this multiple times with both MFA and non-MFA login flows, and all attempts were successful.

KirkKirk, 2021-07-29:
@ngardiner today, TWC can't login to the Tesla account. I tried with and without MFA. Never had issues in the past. 
Maybe, is it possible because of today's Tesla software update?

ngardiner, 2021-07-29:
@KirkKirk which version are you on? The only version that will be able to log in successfully today is the latest revisions of the dev release, v1.2.3 because Tesla have introduced captcha codes at login.

If you're running v1.2.2, unfortunately you'll need to wait a couple of days for the release of v1.2.3. I don't know I could make a release today but I am gearing up to have it done this weekend at the latest.

If you're running v1.2.3, you'll need to do a ```git pull``` to get the latest updates, you should then be able to log in.

KirkKirk, 2021-07-29:
@ngardiner thanks for the response. Im on still on 1.2.2 but I would like 1.2.3 as the Sun is working well today :)

I will manually trigger the stop/start charging till the release because I know that I'm doing something wrong when I tried to get 1.2.3 with:

sudo apt-get update
sudo apt-get install -y docker.io
pip3 install docker-compose
curl https://raw.githubusercontent.com/ngardiner/TWCManager/main/contrib/docker/docker-compose.yml -o docker-compose.yml

but still is on 1.2.2 :(

contactmike1, 2021-08-19:
Does anyone have this working on 1.2.3? I went through the screens on 1.2.3, but I still get API errors. 

`09:24:36 TeslaAPI   20 ERROR: Can't access vehicle status for Fake Vehicle Name.  Will try again later.`

ngardiner, 2021-09-14:
So....

Tesla have changed the API login flow again. 

Now they're using Google Recaptcha, and we don't yet have support for it, but I am working on it.

wooter, 2021-09-14:
> So....
> 
> Tesla have changed the API login flow again.
> 
> Now they're using Google Recaptcha, and we don't yet have support for it, but I am working on it.

I presume the manual token workaround does not work either?

I'm getting this as errors, and a blank page when trying to load twcmanager webpage:

```

20:53:13 TWCManager 20 BackgroundError: Traceback (most recent call last):
  File "/home/user/TWCManager/lib/TWCManager/TWCManager.py", line 245, in background_tasks_thread
    carapi.applyChargeLimit(limit=task["limit"])
  File "/home/user/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 937, in applyChargeLimit
    if not self.car_api_available():
  File "/home/user/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 341, in car_api_available
    or self.getCarApiTokenExpireTime() - now < 30 * 24 * 60 * 60
TypeError: unsupported operand type(s) for -: 'str' and 'float'
, occurred when processing background task
```

MikeBishop, 2021-09-15:
Unfortunately, it's nigh impossible to handle ReCaptcha headlessly, and Tesla requiring the callback URL to be a particular target forecloses using true OAuth even though TWCManager users are browser-based.  It basically has to be a WebView now.

For MMM-Powerwall, I gave up and just take tokens directly; I included links to token-generating apps for iOS and Android, and am writing one for Windows.

However, @wooter, it's not clear that you're having the same problem.  I'd guess that the path to set a token manually is storing a (fake) expiry time and it's accidentally in the wrong type.

ngardiner, 2021-09-17:
So the good news is our login flow is not entirely headless, it's initiated from the UI and what we do is to redirect to a captcha prompt if we detect one. The thing is, the ReCaptcha flow is totally different to Tesla's own flow - they provided an image to validate against whereas ReCaptcha is javascript based.

It has been suggested that what is needed is to snarf the site id from the original page and then load the ReCaptcha Javascript with this site ID and ask the user to enter the code. I'm going to try it out this weekend and see.

What does worry me though, is this is about the 6th incarnation of Tesla's API login flow and they do seem to be trying to keep 3rd party apps out - at least that's how I interpret the constant cat and mouse games, and at some point we may just have to abandon the fight as it could go on like this for a very long time.

I'll try to figure out if there's an issue with manual token entry and at the same time try to update our flow. It's particularly frustrating as my development time is really limited at the moment with $dayjob pressures and my plan was to offer interfaces to avoid our reliance on Tesla API as much as possible by reading state from other projects such as TeslaMate, but the constant changing of the login flow by Tesla takes up all of that time and simply compounds the problem.

MikeBishop, 2021-09-17:
FWIW, I now have my Token generator app for Windows, so there should now be good options for getting tokens on any platform.  I've added app links above my token input form in MMM-Powerwall.

- https://www.microsoft.com/store/apps/9nhdtxbjppxn
- https://play.google.com/store/apps/details?id=net.leveugle.teslatokens
- https://apps.apple.com/us/app/authla/id1546597644

wooter, 2021-09-17:
> However, @wooter, it's not clear that you're having the same problem. I'd guess that the path to set a token manually is storing a (fake) expiry time and it's accidentally in the wrong type.

It's Friday evening and I did some checking. After manually submitting an Access Token and Refresh Token, the carApiTokenExpireTime in settings.json is "", an empty string. By manually replacing it with an epoch time value (expires_in + created_at from the API response) I've managed to avoid the error and get twcmanager working.

nean-and-i, 2021-09-17:
Is there a switch to disable TeslaAPI entirely?
apart the current issue and the fact that I'm very uncomfortable to handout any sensible car info or digital keys, I'd like to charge other cars as well.

MikeBishop, 2021-09-17:
@nean-and-i, you can change the stop mode in Settings to "Stop Responding" or "Stop command."  If TWCManager decides not to charge, though, the car will see it as a sudden loss of power and after a cycle or two won't accept offered power until it's unplugged and plugged back in.

nean-and-i, 2021-09-17:
@MikeBishop thanks for the hint, however, I put it again into "fakeMaster: 2" mode and switched the TWC to "5" position so anyone can charge right away.

ngardiner, 2021-09-29:
@nean-and-i  - Please open another issue for this. The vast majority of users do use the Tesla API and this is an ongoing issue related to login, your request will simply get lost in here.

ngardiner, 2021-09-29:
Okay, so after a long struggle with the new authentication flow I have made some progress. I'll share my findings here.

The challenge with the new recaptcha flow is that it is javascript based rather than image based, and as a result extra checks are performed by Google when you solve the captcha. One of those checks is an origin check, they check that the domain that the recaptcha check is being executed on is .tesla.com and fail if not.

This is not something we can handle natively, but I have a workaround that does work. In addition, I've made some improvements to the token handling on the settings page, if you enter a token now the refresh time will be automatically set to 45 days which is the default refresh token age, however we try to refresh it starting at half of that time.

Finally, I am working on an interface that will sync tokens from TeslaMate, and similar interfaces could be set up for other projects too. This will allow us to outsource token handling to other projects so we don't have to worry so much about the next time that Tesla changes the login flow.

In the latest commit, if you attempt to log in to the Tesla API and get challenged with a recaptcha, a set of options are presented:

  *  Use an app to authenticate to Tesla and obtain tokens. These can be entered under the Settings menu.
        Auth app for Tesla (iOS)
        Tesla Tokens (Android)
        Tesla Auth (macOS, Linux, Windows)
   * Configure Token Sync with an external application like TeslaMate
   * You could perform a DNS redirection trick to host TWCManager on a tesla.com domain.

Note that this should only impact those authenticating for the first time, or where existing tokens have been invalidating. If you logged in previously, you'll have the bearer and refresh tokens stored and we'll continue to refresh those tokens going forward. 

Development of the TeslaMate interface is still underway and it's not yet usable. I intend to have it ready soon and then release 1.2.4 for those suffering from auth issues as a result of the recent changes.

shanesturrock, 2021-12-29:
I’m running the docker development version (1.2.4) and I’ve tried every way to deal with authentication including setting up the DNS redirection to get the captcha working and while it offers, it never completes. Using the tokens gets the API working, but renewal with the refresh token seems to not work either so overnight charging fails. My most recent attempt is to get TeslaMate working and that’s fine with TeslaMate having no issues monitoring the car and I’ve got TWC talking happily to it but it won’t stop the car charging. It sends the stop signal but complains the car hasn’t been woken in the last two minutes and repeats constantly. In desperation, I tried the debug tab and sent a wake signal. The car immediately stopped charging so it seems if you send wake first and then the stop it works. I understand the TeslaMate interface is still in development and it’s really close.

blach, 2022-01-12:
After a long time without problems, my TWCManager instance was logged out of the Tesla account today. 

Now I cannot login again: after solving the recaptcha challenge it takes quite a while until I see this error in red:

`Error encountered during Phase 2 (POST) of the Tesla Authentication process.`

[image]

Things I did:
- added "twcmanager.tesla.com" to my hosts file to get the recaptcha
- updated to the latest commit of the main branch

Did Tesla change something again or is there a known workaround for this problem?

ngardiner, 2022-01-14:
> Did Tesla change something again or is there a known workaround for this problem?

Yes, what you are encountering is the issue that was introduced in September, where Tesla have effectively made it impossible for web apps to implement full end to end authentication flows. 

It is possible they've already worked out that the tesla.com trick was possible and further limited DNS domains to specific hosts - it worked back when I implemented it, but things change constantly with Tesla auth.

You will need to use one of the apps that allow you to obtain tokens I am afraid. It's a problem for all development using the Tesla API and I suspect it won't be going back to the way it was. The reason you need to reauthenticate is that someone was able to obtain the API credentials for 20 vehicles, leading Tesla to expire lots of API tokens, so I'd personally expect more heavy handedness towards API credentials, especially if it turns out it was a 3rd party app that exposed them.

https://www.techspot.com/news/92971-teen-hacker-gains-remote-control-over-20-teslas.html

ngardiner, 2022-01-14:
> The car immediately stopped charging so it seems if you send wake first and then the stop it works. I understand the TeslaMate interface is still in development and it’s really close.

That's interesting! I don't have the same issue in my environment and technically the vehicle should already be awake if it's charging, but I have no objection to waking the car before sending a stop charging command, especially if it makes it work with greater stability.

I found a separate issue which I'm about to commit a fix for (if the tokens expire after we do sync with TeslaMate, they don't update on the TWCManager side) so I will add this in at the same time.

Thanks for the testing.

blach, 2022-01-15:
> It is possible they've already worked out that the tesla.com trick was possible and further limited DNS domains to specific hosts - it worked back when I implemented it, but things change constantly with Tesla auth.

Good point. I just tried to add www.tesla.com (instead of twcmanager.tesla.com) to my /etc/hosts file and use that to connect to twcmanager and solve the recaptcha, but I get the same Phase 2 error during auth. Maybe they limit it to https now so it doesn't work over http?
 
> You will need to use one of the apps that allow you to obtain tokens I am afraid. 

I tried that in the past, but I couldn't get TWCManager to refresh the tokens. They alway expired after 7 hours. 

I have now tried it again using the access token and refresh token the "Auth app for Tesla" app from the App Store. This works so far, let's see if it is refreshed when the access token expires.

> The reason you need to reauthenticate is that someone was able to obtain the API credentials for 20 vehicles, leading Tesla to expire lots of API tokens, so I'd personally expect more heavy [...]

Thanks, I see. I didn't hear about that before.

Thanks a lot for your answer, I really appreciate it. 

Do you accept donations or something similar for this project?

blach, 2022-01-15:
> I tried that in the past, but I couldn't get TWCManager to refresh the tokens. They alway expired after 7 hours.
>
> I have now tried it again using the access token and refresh token the "Auth app for [...]

TWCManager cannot refresh tokens generated by "Auth app for Tesla". I forced a refresh and this is logged by TWCManager:

```
Jan 15 02:50:08 TWCManager python3[16231]: 02:50:08 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Jan 15 02:50:08 TWCManager python3[16231]: 02:50:08 🚗 TeslaAPI 13 Attempting token refresh
Jan 15 02:50:09 TWCManager python3[16231]: 02:50:09 🚗 TeslaAPI 19 Car API request<Response [401]>
Jan 15 02:50:09 TWCManager python3[16231]: 02:50:09 🚗 TeslaAPI 17 Car API auth response{'error': 'invalid_grant', 'error_description': 'The provided authorization grant is invalid, expired, revoked, does not match the redirection URI used in the authorization request, or was issued to another client.'}

```

I also added logging to `TeslaAPI.py` to see why the auth flow fails after the captcha. This is what Tesla's server returns when the request is sent in `apiLoginPhaseTwo`:

```
code: 403

headers: {'Server': 'AkamaiGHost', 'Mime-Version': '1.0', 'Content-Type': 'text/html', 'Content-Length': '296', 'Expires': 'Sat, 15 Jan 2022 02:22:54 GMT', 'X-Reference-Error': '18.1d8a1402.1642213374.8c79ce67', 'Date': 'Sat, 15 Jan 2022 02:22:54 GMT', 'Connection': 'close', 'Permissions-Policy': 'interest-cohort=()', 'Set-Cookie': '_abck=6EFE80B6232CB350321DAEC2D20E952F~-1~YAAQHYoUAmiavLV9AQAASnqLWweiTgbrVtZ7VpzsOkgeu0j/Py0BRYVthWbaCmLAvhuy+/9EhCih8yKMhLmPkzU843NuXgznvhlB/dt8JTLhffIXmGzvOqLMwzu55TF5qJdaqJSPFlqPxHY6b1dnwLaHBCBcro8vtACKTNBlBLoD9lylkj54eCNNihoNw8YOsHDR1/nC91+B7is231CE8YNSXupX9IleivTAaTu3WypM05Q9oUCg2LiAKQLTagXORhk2dI3o+vfab8C+B1Q73HvhJlkX8wT+RH/TJYQqOJ1AJ0VgL9+YgpoUb2ap2QoAryK43Ld5lYmnveENUu6nrv9jaEdd83DjoldAJqMvMiIgu7sYlKIOIkG8lVbhBovHIxY+pM3Z~-1~-1~-1; Domain=.tesla.com; Path=/; Expires=Sun, 15 Jan 2023 02:22:54 GMT; Max-Age=31536000; Secure'}

text: <HTML><HEAD>
<TITLE>Access Denied</TITLE>
</HEAD><BODY>
<H1>Access Denied</H1>
 
You don't have permission to access "http&#58;&#47;&#47;auth&#46;tesla&#46;com&#47;oauth2&#47;v3&#47;authorize&#63;" on this server.<P>
Reference&#32;&#35;18&#46;1d8a1402&#46;1642213374&#46;8c79ce67
</BODY>
</HTML>
```

So I'm not sure what to do now.

blach, 2022-01-15:
Success! I could make TWCManager refresh tokens created by "Auth app for Tesla" after I found this comment: https://github.com/timdorr/tesla-api/discussions/390#discussioncomment-1959992 

The following changes were needed in TeslaAPI.py:

Change `refreshURL` from 
`"https://owner-api.teslamotors.com/oauth/token"` to 
`"https://auth.tesla.com/oauth2/v3/token"`.

In `def apiRefresh(self):` change `data` to:

```
data = {
            "client_id": "ownerapi",
            "grant_type": "refresh_token",
            "refresh_token": self.getCarApiRefreshToken(),
            "scope": "openid email offline_access",
        }
```

Since refresh tokens expire after 8 hours, I also changed the expiry check from 30 days to 1 hour. Otherwise it would refresh tokens every time.

In line 371, changed from 
`or self.getCarApiTokenExpireTime() - now < 30 * 24 * 60 * 60` to 
`or self.getCarApiTokenExpireTime() - now < 60 * 60`

ngardiner, 2022-01-15:
> Success! I could make TWCManager refresh tokens created by "Auth app for Tesla" after I found this comment: [timdorr/tesla-api#390 (reply in thread)](https://github.com/timdorr/tesla-api/discussions/390#discussioncomment-1959992)

That's awesome, thank you for this! I'll get it fixed right now.

nean-and-i, 2022-01-15:
Hi,
after recent disclosures, I really vote for a bypass switch to use all features without tesla API.
There are serious reasons, not to handout "the keys of the car", I assume no one can guarantee security of the code or do security testings/audits, especially of libraries and dependencies (just thinking of logj).

What is the main purpose that requires the API, are there no other ways to get required information?

- location: as the car is connected and the VIN can be readout, we're sure the car is max. a cable length away of the TWC
- start/stop: is this really required, as the TWCManager can start/stop, right?

Can we do an evaluation and discuss ideas to workaround the need of tesla api and agree on set of principle eg.: "we don't track users, we don't need the keys of the car, we don't collect data other than required for charging"

thanks

wooter, 2022-01-15:
Reason is that if you don't tell the car top stop or start charging, but have it dependent on the TWC, the car thinks that if the TWC has stopped charging for 3 times, the TWC is malfunctioning and will stop further charging.

With using the API, TWCManager tells the car when to stop or start charging, and this fault does not occur.

ngardiner, 2022-01-15:
This is correct. There is no way to implement this without the Tesla API without buying additional hardware (a CAN bus adaptor) to tell the vehicle to stop charging via the CAN network. Outside of this, any attempt to stop charging via the TWC itself results in the vehicle sensing an issue with the TWC and stopping any further charging.

nean-and-i, 2022-01-15:
> Reason is that if you don't tell the car top stop or start charging, but have it dependent on the TWC, the car thinks that if the TWC has stopped charging for 3 times, the TWC is malfunctioning and [...]

this behaviour is new to me and somehow strange.

but this is known by the TWCManager, I could think of a sort of counter and respecting charging curve could do the trick, to not run into this issue?
maybe a deep dive to find the root cause on protocol level, maybe there is another way that signals the tesla a valid/intentional and non-faulty stop.

I can do tests with Keba charging stations, that I'm reverse engineering right and playing with occp.
from Charging network point of view there are valid reasons and use cases where sending stop is normal even multiple times.

wooter, 2022-01-15:
Don't forget TWCManager can only influence the functions the TWC exposes. So we won't be able to talk OCCP to it, let alone influence what it is saying to the car.

If you want that functionality, you're better off building your own EVSE.

nean-and-i, 2022-01-15:
> Perhaps we need to tweet Elon? The greatest irony of all of this is that we'd never have used the API if the vehicle were able to be controlled via the TWC sharing protocol, but then again perhaps Tesla hate us hacking the hardware more than they do the API? Who knows, Tesla has never communicated with 3rd party developers ever - we don't know what they want of us.

I think Tesla will completely lockout everyone from the API soon, as they can and the'd need to because of press and pressure (secure their crown jewels). 
Theoretically they could change access to some sort of key auth only, like their service wifi, then they control and exactly know who and when the API is accessed.

ngardiner, 2022-01-15:
> from Charging network point of view there are valid reasons and use cases where sending stop is normal even multiple times.

There's some confusion here about what we're leveraging within this project. We are using the proprietary Tesla TWC specific sharing protocol which allows multiple TWCs to share a single circuit, and we use that protocol to adjust the charge rate.

Stopping a vehicle when tracking green energy is necessary because there's a minimum charge rate or we may go negative in terms of generation vs consumption. This doesn't happen with the TWC circuit sharing which the protocol was developed for. This is why it doesn't exist. What you're working on is vastly different to what we are doing - we're focused on a Tesla-specific implementation. 

If we had the ability to go with dedicated hardware we'd have a much more tailored solution, but we're working specifically with Tesla TWC in this project. With what you're doing - as interesting as it is, I'm not sure why you'd be using TWCManager in any capacity as you'd have the luxury of full control over the EVSE behaviour, which would be lovely, but we don't have that.

ngardiner, 2022-01-15:
> I think Tesla will completely lockout everyone from the API soon, as they can and the'd need to because of press and pressure (secure their crown jewels).

This would be a topic for a Discussion rather than this issue IMO. Currently we have API access and while getting and maintaining the keys is harder than just entering a username and password as we did in the past, it's only moderately more complex at the moment - and once you have the token, and once refresh is fixed, we should have no issue maintaining it unless it's invalidated by password change, or by action on Tesla's side.

We should keep this issue clear for tracking any confirmed changes in the Tesla API - I think that's a good compromise, that way I can ensure that the discussion doesn't hide any important info that requires us to make changes to maintain current functionality.

nean-and-i, 2022-01-15:
So why not do an evaluation, like put together in a list all the functions/code snippets that requires the API, and start to discuss/think of ways to use functionality from TWCmanager protocol only or different methods?

ngardiner, 2022-01-15:
> So why not do an evaluation, like put together in a list all the functions/code snippets that requires the API, and start to discuss/think of ways to use functionality from TWCmanager protocol only or different methods?

This has been done long ago - there's several hundred posts discussion about it on the Tesla forums:

https://teslamotorsclub.com/tmc/threads/new-wall-connector-load-sharing-protocol.72830

There is absolutely no reason that we'd use the API if the functions required were available in hardware.

MikeBishop, 2022-01-15:
It's also worth noting that, because this protocol is reverse-engineered, it's even possible that the functionality we need *does* exist in hardware, and we just don't know about it.  The ability to have a TWC hard stop -- open its contactor immediately -- is a relatively recent addition, IIRC because no one knew that command was present before.

However, that command doesn't tell the car anything -- from the car's point of view, it's a power loss just like when we stop responding.  And the car gets touchy about EVSEs that repeatedly lose power without communication.

ngardiner, 2022-01-15:
> It's also worth noting that, because this protocol is reverse-engineered, it's even possible that the functionality we need _does_ exist in hardware, and we just don't know about it. The ability to [...]

That's not quite historically accurate. This command was introduced in firmware 4.5.3, and was found after this version was extracted from a newer TWC:

https://teslamotorsclub.com/tmc/threads/tesla-wall-connector-load-sharing-protocol.72830/page-24#post-4388511

Because of the low regularity of updates to TWCs because they don't do OTA updates, it's particularly hard to get your hands on any new firmware versions to even test with, so it makes things all the more difficult. We know the stop command does not work reliably in 4.5.3,

The thread that I linked to previously had two different approaches taken - the firmware was extracted and reverse engineered and every command possible was sent to at least one TWC. That, incidentally, caused the TWC to be bricked, but it did expose which commands garnered a response from the TWC. This was the first approach and enumerated all known TWC commands.

The firmware decompilation however was much more thorough and it was identified that there is **nothing** within the firmware which references the CAN bus command ID (0x30C)  which would result in the vehicle stopping charging:

https://teslamotorsclub.com/tmc/threads/tesla-wall-connector-load-sharing-protocol.72830/page-28#post-4711377

There may have been firmware updates since, we'd have to go to a lot of trouble to extract the firmware to find out since it isn't updated OTA and needs to be extracted from the TWC itself, but let's be honest - Tesla are probably not working on the TWCv2 anymore. 

I would love to not rely on the API, I'd take that path in a minute. Sadly nobody has a solution to this, and we're really just at the same point we were years ago looking for a solution. If we did have a command or mechanism to cleanly stop charging, we wouldn't need to discuss it any further, it would be the default approach out of the box.

Unfortunately the original author of TWCManager seems to have deleted all of their posts, so there's a lot of good discussion around this which is gone, but thankfully we still have replies which quote the original text.

nean-and-i, 2022-01-15:
> this behaviour is new to me and somehow strange.
>
> but this is known by the TWCManager, I could think of a sort of counter and respecting charging curve could do the trick, to not run into this [...]

I just tested this case with my Keba testsetup ( TM3 <= Keba KeContact P30 <-ocpp1.6Json-> Rasbpi_SteVe )

I could remote start and stop charging via OCPP msgs to Keba mutiple times (up to 10 times), without the Tesla M3 confusing and without app / tesla API.
I "assume" there is no special comms going on between Keba and M3 (without knowing whats happening via CP - Control Pilot) and Keba simply switch off power, which means to me, switching off power seems not to cause an issue, or do I overlook something?

Flitzer42, 2022-01-19:
Hello ngardiner,
not sure whether this helps or not. I am also struggling with the Tesla API token issue. The AuthTesla app shows, that the token is valid only for 8 hours..... I updated yesterday and after entering the tokens it works for some time, but get stuck after a while...  Using journalctl it shows "ERROR: Can't access vehicle status for Model 3.  Will try again later." only one time. From this log (level) I do not see any frther attempt to access the Tesla API.

I was looking into the OpenWB project. They use the Tesla API for getting the SoC. They had to update the procedure as well.
Here is (from the outside) what they do:
They open a dialog, showing: 
```
"Since the switch to reCaptcha for Tesla registration, registration can no longer be carried out automatically. Please carry out the following steps one after the other: 
    Please click here to register directly with Tesla. A new window will open for this purpose. Please allow popups in the browser!
    Log in in the new window with your own access data.
    If the login is successful, a "Page Not Found" message will appear on the Tesla website. Then copy the complete URL of this website (e.g. https://auth.tesla.com/void/callback?code=.....&state=...&issuer=....)
    Paste the copied URL into the field here and click the Login button."
```
The "click here" link is: "https://auth.tesla.com/oauth2/v3/authorize?audience=&client_id=ownerapi&code_challenge=OTQ2NjNmMzY3MTM3ZDIzMGM4MmJjOTUyNmNlMTA4YzM0ZDQ4ZjAyYmZlODYyNTI3OTFhMzQwYjU0YWMzOTYwM2I1NGE0NjA0NDJiOTQ1N2IyNDg3ZmY&code_challenge_method=S256&locale=en-US&prompt=login&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&response_type=code&scope=openid+email+offline_access&state=N2IyNDlmNWVkMmZh"
Entering the credentials, a "page not found" message appears, but the URL of this page is sufficient for the OpenWB to get the token. No idea how this works...
I have no Idea reagarding the roken refresh policy they use.
I have a bit more confidence when using the Tesla homepage for creating the tokens...

Only good news here (off topic): OpenWB now offers the option to select "HTTPControl / Ngardiner Fork " to control TWCmanager. Which makes the legacy IPC interface obsolete (at least for the OpenWB use case) :-)

Flitzer42, 2022-02-07:
Hello @ngardiner, on my installation the start/stop charging via the APi is still not working (or only for a few hours).
Is it supposed to work, or is there still a general / known issue with the (always changing) API of Tesla?
Is there a solution round the corner?
Thanks for your short feedback.
BTW: here is the OpenWB solution for the API access: https://github.com/snaptec/openWB/tree/master/modules/soc_tesla
Unfortunately they only retreive the State of Charge, but do not offer to start or stop charging.

markdakerhanson, 2022-02-13:
Hello All

I have been experiencing similar symptoms to @Flitzer42 since first installing TWCManager in September – I have tried to ignore the issue, putting it down to my inexperience with Pi.  For me, everything functions as expected except for the updating of the Tesla API bearer token, preventing start/stop charge control.  I have pulled the updated files following the recent update to refreshURL and expiry check in TeslaAPI.py (thank you  @ngardiner & @blach!) which I was hopeful would fix my issue but unfortunately I still get the following error log after a few hours when the access/bearer token has expired.

Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 10 Car API vehicle status{'error': 'invalid bearer token'}

```
Feb 13 11:49:57 TWCManager python3[462]: 11:49:57 🚗 TeslaAPI 15 Car API disabled for 1 more seconds due to recent error.
Feb 13 11:49:57 TWCManager python3[462]: 11:49:57 🚗 TeslaAPI 13 car_api_charge return because car_api_available() == False
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.3.154)
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.3.154)
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Master   10 Total amps all slaves are using: 6.19
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Master   10 Number of cars charging now: 1
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.3.154)
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🎮 HTTP     10 Ending API GET (Url: /api/getStatus / IP: 192.168.3.154)
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 RS485      12 Tx@: C0 FB E0 77 77 32 17 09 00 00 00 00 00 00 00 00 20 C0
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🚗 TeslaAPI 13 car_api_available returning False because of recent carApiLasterrorTime 599.6800711154938 or empty carApiBearerToken 'eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVC….XXXXXXXXX ………….  07bNeaR1y5bwvCMtTg'
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 🚗 TeslaAPI 13 car_api_charge return because car_api_available() == False
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Manager  12 Rx@: (FC) FD E0 32 17 77 77 09 02 58 02 6F 00 00 00 00 EB
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Master   10 Number of cars charging now: 1
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Slave    10 desiredAmpsOffered TWC: 32 17 reduced from 0 to 0 with 1 cars charging and flex Amps of 0.0.
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Slave    10 desiredAmpsOffered: 0 < minAmpsToOffer: 6 (flexAmps: 0.0)
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Slave    10 desiredAmpsOffered TWC: 32 17 reduced to 0 from 0 because maxAmpsToDivideAmongSlaves 0 / numCarsCharging 1 < minAmpsToOffer 6
Feb 13 11:49:58 TWCManager python3[462]: 11:49:58 ⛽ Slave    10 set_last_amps_offered(TWCID=32 17, desiredAmpsOffered=0)
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Manager  12 Rx@: (FC) FD E0 32 17 77 77 09 02 58 02 6F 00 00 00 00 EB
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Master   10 Number of cars charging now: 1
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Slave    10 desiredAmpsOffered TWC: 32 17 reduced from 0 to 0 with 1 cars charging and flex Amps of 0.0.
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Slave    10 desiredAmpsOffered: 0 < minAmpsToOffer: 6 (flexAmps: 0.0)
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Slave    10 desiredAmpsOffered TWC: 32 17 reduced to 0 from 0 because maxAmpsToDivideAmongSlaves 0 / numCarsCharging 1 < minAmpsToOffer 6
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 ⛽ Slave    10 set_last_amps_offered(TWCID=32 17, desiredAmpsOffered=0)
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 RS485      12 Tx@: C0 FB E0 77 77 32 17 09 00 00 00 00 00 00 00 00 20 C0
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 🚗 TeslaAPI 13 car_api_available returning True
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 🚗 TeslaAPI 13 startOrStop is set to stop
Feb 13 11:49:59 TWCManager python3[462]: 11:49:59 urllib3.co 10 Starting new HTTPS connection (1): owner-api.teslamotors.com:443
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 urllib3.co 10 [https://owner-api.teslamotors.com:443](https://owner-api.teslamotors.com/) "GET /api/1/vehicles/XXXXXXXXXXXXXXX HTTP/1.1" 401 32
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/XXXXXXXXXXX <Response [401]>
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 10 Car API vehicle status{'error': 'invalid bearer token'}
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 20 ERROR: Can't access vehicle status for 0dB.  Will try again later.
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 13 updateCarApiLastErrorTime() called due to Tesla API Error. Updating timestamp from 1644712799.0171013 to 1644713400.6335592
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 13 0dB not ready because it wasn't woken in the last 2 minutes.
[... cut ...]
Feb 13 11:50:00 TWCManager python3[462]: 11:50:00 🚗 TeslaAPI 13 0dB not ready because of recent lastErrorTime 1644713400.6335592
```

If I now use Auth for Tesla app to generate a new refresh and access token and copy them into the Settings page, the tokens will write to the settings.json file successfully and the system will correct itself again.  For the next few hours the API will be able to stop/start charging until the access/bearer token once again expires.

I'm afraid I don't have the knowledge to troubleshoot the API further, but hope this helps...

ngardiner, 2022-02-25:
Just an update on this - we have a new way of getting tokens if you pull the latest dev version of TWCManager (or once v1.2.6 is released shortly), this should fix the token issues people have been having.

I also believe there's a bug with the refresh timing for manually entered tokens and I'm hoping to have it identified and fixed very soon, but in the meantime the new approach should be easier than having to use an external app for getting API tokens into TWCManager.

Flitzer42, 2022-02-26:
Hi Nathan @ngardiner ,
thanks for working on this issue (was watiting day by day for an update :-)).
Unfortunately I did not manage to get it work.
The homescreen shows: Error encountered during Phase 2 (POST) of the Tesla Authentication process.

Remark: I have my pi on my desk, connected to a USB-RS485 adapter, but no TWC connected.
Updated with the following commands to the lasted SW today
```
cd /home/pi/TWCManager
git pull
make install
```

Set loglevel to 20, started TWC manager as service, used jounalctl for logging:
```
--> When entering credentials (token replaced by XXX)
Feb 26 11:23:35 raspberrypi python3[501]: 11:23:35 urllib3.co 10 Starting new HTTPS connection (1): auth.tesla.com:443
Feb 26 11:23:35 raspberrypi python3[501]: 11:23:35 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:23:35 raspberrypi python3[501]: 11:23:35 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:23:36 raspberrypi python3[501]: 11:23:36 urllib3.co 10 https://auth.tesla.com:443 "GET /oauth2/v3/authorize?client_id=ownerapi&code_challenge=XXX&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&response_type=code&scope=openid+email+offline_access&state=4Rm0PiLwul7JzaQpYfIz2w HTTP/1.1" 200 10232
Feb 26 11:23:36 raspberrypi python3[501]: 11:23:36 🚗 TeslaAPI 15 Tesla Auth form fetch success, attempt: 0
Feb 26 11:23:36 raspberrypi python3[501]: 11:23:36 urllib3.co 10 https://auth.tesla.com:443 "POST /oauth2/v3/authorize?client_id=ownerapi&code_challenge=XXX&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&response_type=code&scope=openid+email+offline_access&state=4Rm0PiLwul7JzaQpYfIz2w HTTP/1.1" 403 296
--> Next section repeats multiple times:
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 ⛽ Master   10 Total amps all slaves are using: 0
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 ⛽ Master   10 Number of cars charging now: 0
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 🎮 HTTP     10 Ending API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:23:38 raspberrypi python3[501]: 11:23:38 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:23:39 raspberrypi python3[501]: 11:23:39 urllib3.co 10 Resetting dropped connection: auth.tesla.com
Feb 26 11:23:40 raspberrypi python3[501]: 11:23:40 urllib3.co 10 https://auth.tesla.com:443 "POST /oauth2/v3/authorize?client_id=ownerapi&code_challenge=XXX&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&response_type=code&scope=openid+email+offline_access&state=4Rm0PiLwul7JzaQpYfIz2w HTTP/1.1" 403 296

Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 🚗 TeslaAPI 19 Wasn't able to post authentication form after 9 attempts
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 ⛽ Master   10 Total amps all slaves are using: 0
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 ⛽ Master   10 Number of cars charging now: 0
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 🎮 HTTP     10 Ending API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:24:09 raspberrypi python3[501]: 11:24:09 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:24:12 raspberrypi python3[501]: 11:24:12 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)

Feb 26 11:24:31 raspberrypi python3[501]: 11:24:31 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Feb 26 11:24:31 raspberrypi python3[501]: 11:24:31 🚗 TeslaAPI 13 car_api_available returning False because of recent carApiLasterrorTime 1645871071.976424 or empty carApiBearerToken ''
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 ⛽ Master   10 Total amps all slaves are using: 0
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 ⛽ Master   10 Number of cars charging now: 0
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 🎮 HTTP     10 Ending API GET (Url: /api/getStatus / IP: 192.168.178.89)
Feb 26 11:24:32 raspberrypi python3[501]: 11:24:32 🎮 HTTP     10 Ending API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:24:35 raspberrypi python3[501]: 11:24:35 🎮 HTTP     10 Starting API GET (Url: /api/getSlaveTWCs / IP: 192.168.178.89)
Feb 26 11:24:35 raspberrypi python3[501]: 11:24:35 🎮 HTTP     10 Starting API GET (Url: /api/getStatus / IP: 192.168.178.89)

```

I curious, wha TWCManager tries to access a IP address in my local network.

Any helpis welcome.

Thanks again for your great efforts on this projects. There are many people out in the work beeing happy to connect their solar to TWC.

ngardiner, 2022-02-26:
Hi @Flitzer42,

This would mean you are running a stable/older version of TWCManager and not the current development version, as the error you are getting does not exist anymore, I removed it as part of the latest login updates. You should not be seeing username/password prompts at all if the update was successful, you should instead see a 3 step login process.

Please try a ```git checkout main``` and then run ```make install``` again

The IP on your local network shown in the logs is the Source IP of the web client who is requesting that page from TWCManager, there's no outbound connection to that IP (just a reply to the request recieved).

Flitzer42, 2022-02-26:
Hi @ngardiner , 
Thanks a lot for your fast response. I still get stuck.
I tried to update an existing installation running into the issue described above.
`git checkout main` shows me that I am already on main. `make install` runs without error.
Which file contains the fix? Perhaps I can manually check whether I have the latest version running.

In parallel I tried to setup on a blank new system. Installation successfull, but two issues:

1. Yaml
pi@raspberrypi:~/TWCManager $ sudo -u twcmanager python -m TWCManager
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 39, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'

2. twcmanager.service
twcmanager.service is not installed in etc/systemd/system

BTW: Installation is very fast, takes only a few seconds, something missing?

Thanks a lot for your help and support

Flitzer42, 2022-02-26:
Hi @ngardiner ,
Thanks for your support. I had an issue with the installation. PIP3 was missing, so the installation did not run correct.

Now I see the new UI method to get the token.
Two observations:
1. No request for credential on tesla page
When clicing on "Tesla login" it shows a "page not found", but does not request any login. So perhaps this link is wrong.
It does not mather whether I am already signed in to Tesla.com in or not.
2. Crash
When pasting the URL anyways into the field and pressing return or clicking on "Save API key", the page disapears and TWCManager crashes.:
```
pi@raspberrypi:~/TWCManager $ sudo -u twcmanager python -m TWCManager
14:45:36 🎮 HTTP     20 Serving at port: 8080
14:45:36 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
----------------------------------------
Exception happened during processing of request from ('192.168.178.89', 50384)
Traceback (most recent call last):
  File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 151, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 933, in do_POST
    res = master.getModuleByName("TeslaAPI").saveApiToken(url)
  File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1035, in saveApiToken
    req = requests.post(self.__authURL, headers=headers, json=data)
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 515, in request
    prep = self.prepare_request(req)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 453, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 321, in prepare
    self.prepare_body(data, files, json)
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 473, in prepare_body
    body = complexjson.dumps(json, allow_nan=False)
  File "/usr/lib/python3.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.7/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.7/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type bytes is not JSON serializable
----------------------------------------
^C14:48:02 ⛽ Manager  20 Exiting after background tasks complete...
```
3. OpenWB Method
OpenWB utilizes the same method. When obtaining an URL (also page not found...) via OpenWB and paste it into TWCManager, it also crashes.
The URL OpenWB utilizes is different:
TWCManager: 
`https://auth.tesla.com/void/callback?code=XXX&issuer=https%3A%2F%2Fauth.tesla.com%2Foauth2%2Fv3`
OpenWB:
`https://auth.tesla.com/oauth2/v3/authorize?audience=&client_id=ownerapi&code_challenge=XXX&code_challenge_method=S256&locale=en-US&prompt=login&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&response_type=code&scope=openid+email+offline_access&state=MTQ3MDM3NmViZTEx`

OpenWB code: https://github.com/snaptec/openWB/tree/master/modules/soc_tesla

So at my installation, there are issues with both ends (link to correct tesla page, parsing the result)

Hope this helps to come closer to the issue.

Thanks a lot for your support

ngardiner, 2022-02-27:
@Flitzer42

Number 1 is normal, if you're already authorized to tesla.com you will not get the login prompt. I can see you're convinced we're using the wrong URL but we are not, it is working fine for me. Try it in an incognito browser and you'll see a login prompt.

For Number 2, I'll investigate further.

Number 3 is a misunderstanding of the flow. The URL you see for TWCManager is as a result of the user-side login process. The URL you see for openWB is the request prior to this response. See https://tesla-api.timdorr.com/api-basics/authentication Step 2, you are seeing the Request URL for openWB and the Response URL for TWCManager.

Number 4 - Something hasn't gone right with ```make install```. Did you see any errors?

ngardiner, 2022-02-27:
@Flitzer42 - not sure why our environments differ, but I looked at the code and the only possible parameter that could have been a bytes object to cause the error above is definitely a string on my system, but I've added a check in the latest commit 9553b89 to see if it is a bytes object and if so convert it back to string - could you please give it a try

dehsgr, 2022-02-27:
@ngardiner fetching token with the new process fails currently when /teslaAccount/saveToken is called (empty respone). Token is not saved yet. I would provide you with this info before you release v1.2.5...

ngardiner, 2022-02-27:
Hi, as mentioned before it's working for me so I need more information before I can do anything about it.

I'm releasing v1.2.5 regardless as I've tested it under multiple conditions myself without any issue and don't have any technical detail as to why it would not work for others - if I can get some useful detail quickly I can update the release - otherwise those affected will need to manually enter token details until a subsequent release fixes it.

dehsgr, 2022-02-27:
@ngardiner which information would you bring forward? I would try to provide you that information. Might location/country from which authentication is performed be relevant? I‘m authenticating from Germany.

ngardiner, 2022-02-27:
Hi @dehsgr, most useful would be the log output at the time that you get the blank page - it should show an error which would give some background on why it's not working for you

dehsgr, 2022-02-27:
@ngardiner here you are:
```
Feb 27 15:38:14 twc python3[18938]: ----------------------------------------
Feb 27 15:38:14 twc python3[18938]: Exception happened during processing of request from ('192.168.2.101', 51057)
Feb 27 15:38:14 twc python3[18938]: Traceback (most recent call last):
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
Feb 27 15:38:14 twc python3[18938]:     self.finish_request(request, client_address)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
Feb 27 15:38:14 twc python3[18938]:     self.RequestHandlerClass(request, client_address, self)
Feb 27 15:38:14 twc python3[18938]:   File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 151, in __init__
Feb 27 15:38:14 twc python3[18938]:     BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
Feb 27 15:38:14 twc python3[18938]:     self.handle()
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/http/server.py", line 426, in handle
Feb 27 15:38:14 twc python3[18938]:     self.handle_one_request()
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
Feb 27 15:38:14 twc python3[18938]:     method()
Feb 27 15:38:14 twc python3[18938]:   File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 933, in do_POST
Feb 27 15:38:14 twc python3[18938]:     res = master.getModuleByName("TeslaAPI").saveApiToken(url)
Feb 27 15:38:14 twc python3[18938]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1037, in saveApiToken
Feb 27 15:38:14 twc python3[18938]:     req = requests.post(self.__authURL, headers=headers, json=data)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/api.py", line 119, in post
Feb 27 15:38:14 twc python3[18938]:     return request('post', url, data=data, json=json, **kwargs)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/api.py", line 61, in request
Feb 27 15:38:14 twc python3[18938]:     return session.request(method=method, url=url, **kwargs)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/sessions.py", line 528, in request
Feb 27 15:38:14 twc python3[18938]:     prep = self.prepare_request(req)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/sessions.py", line 466, in prepare_request
Feb 27 15:38:14 twc python3[18938]:     hooks=merge_hooks(request.hooks, self.hooks),
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/models.py", line 319, in prepare
Feb 27 15:38:14 twc python3[18938]:     self.prepare_body(data, files, json)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/local/lib/python3.7/dist-packages/requests-2.25.1-py3.7.egg/requests/models.py", line 469, in prepare_body
Feb 27 15:38:14 twc python3[18938]:     body = complexjson.dumps(json)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/__init__.py", line 231, in dumps
Feb 27 15:38:14 twc python3[18938]:     return _default_encoder.encode(obj)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/encoder.py", line 199, in encode
Feb 27 15:38:14 twc python3[18938]:     chunks = self.iterencode(o, _one_shot=True)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
Feb 27 15:38:14 twc python3[18938]:     return _iterencode(o, 0)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/encoder.py", line 179, in default
Feb 27 15:38:14 twc python3[18938]:     raise TypeError(f'Object of type {o.__class__.__name__} '
Feb 27 15:38:14 twc python3[18938]: TypeError: Object of type bytes is not JSON serializable
Feb 27 15:38:14 twc python3[18938]: ----------------------------------------
Feb 27 15:38:14 twc python3[18938]: ----------------------------------------
Feb 27 15:38:14 twc python3[18938]: Exception happened during processing of request from ('192.168.2.101', 51058)
Feb 27 15:38:14 twc python3[18938]: Traceback (most recent call last):
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
Feb 27 15:38:14 twc python3[18938]:     self.finish_request(request, client_address)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
Feb 27 15:38:14 twc python3[18938]:     self.RequestHandlerClass(request, client_address, self)
Feb 27 15:38:14 twc python3[18938]:   File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 151, in __init__
Feb 27 15:38:14 twc python3[18938]:     BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
Feb 27 15:38:14 twc python3[18938]:     self.handle()
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/http/server.py", line 426, in handle
Feb 27 15:38:14 twc python3[18938]:     self.handle_one_request()
[... cut ...]
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
Feb 27 15:38:14 twc python3[18938]:     return _iterencode(o, 0)
Feb 27 15:38:14 twc python3[18938]:   File "/usr/lib/python3.7/json/encoder.py", line 179, in default
Feb 27 15:38:14 twc python3[18938]:     raise TypeError(f'Object of type {o.__class__.__name__} '
Feb 27 15:38:14 twc python3[18938]: TypeError: Object of type bytes is not JSON serializable
Feb 27 15:38:14 twc python3[18938]: ----------------------------------------
```

[10 more replies]

closed 2026-10-26 by MikeBishop