ngardiner/TWCManager #303: Feature request: prevent cars from charging

dehsgr, 2021-06-06:
It would be nice if there would be a way to prevent cars from charging at all. Currently there is no option to set Amps to Zero. So some option could be to stop responding to all TWC. 

Controllable via HTTP API would be nice too.

Thanks in advance!

Saftwerk, 2021-06-06:
Switch OFF the Fuse?!

dehsgr, 2021-06-06:
Via HTTP API! ;-)

Maybe an option could be to use „allowed VINs for charging“ feature?

magic7music, 2021-06-07:
What about using an NFC reader with some NFC tags (one for every car) to access TWCM and set a state for charging allowed or not?

ngardiner, 2021-06-07:
> Maybe an option could be to use „allowed VINs for charging“ feature?

Yes - this is what the feature is intended to do. You can either set it to deny by default (and you then need to authorize vehicles one by one) or you can set it to charge by default and block vehicles that you don't want to charge.

The key limitations are:

- Needs a TWC with new enough firmware that it reads VINs. If you see VINs in the web interface, you're good to go
- If it's not a Tesla vehicle, we only see a string of zeros for VIN. You can still block or allow this zero VIN in the web interface, but it is universal for all non Tesla vehicles. Sadly the CAN communication that provides VINs is Tesla specific.

dehsgr, 2021-06-07:
Thanks for information. I just try to clarify a few things, what I'm trying to archive:
- I've a Tesla Model 3 I would like to charge
- I've a non-Tesla EV (Renault ZOE) I would like to charge
- I would like to block all other vehicles from charging
- The blocking mechanism should be enabled/disabled via HTTP API

So my intention is to get TWC into some kind of "blocking state" via HTTP API when all persons left the home. I might archive that if there is any possibility to switch over "allow/block vehicle by VIN" setting. Another way could be to be able to tell TWCManager not to respond to TWC or other ways to tell TWCs not to start charging.

ngardiner, 2021-06-07:
Got it. What I would do is use the just added ```addConsumptionOffset``` command to add a huge consumption offset which would make it look like consumption was way higher than generation.... Then just ```deleteConsumptionOffset``` when you are back... I am assuming there is some other platform that determines whether you are home or away, when away you can just do something like:

```
POST /api/addConsumptionOffset
{
   "offsetName": "No charging while I'm away",
   "offsetValue": 1000,
   "offset unit": "A"
}
```

Then when you are home again, you can either edit that offset by setting the offsetValue to 0 or delete it with:

```
POST /api/deleteConsumptionOffset
{
   offsetName": "No charging while I'm away"
}
```

BTW, which stop charging method do you use today - Tesla API? The problem being that it won't cause Tesla vehicles not on your account or non-Tesla vehicles like the Zoe to stop charging, so I think you'll need to set it to "Stop Responding to Slaves" for it to work universally

magic7music, 2021-06-07:
Hi Nathan, thanks for that information.
We just got our second EV today. Sadly no M3 but a Nissan Leaf.
I´ll try your workaround tomorrow morning. Now you probably understand the idea with the RC522.

dehsgr, 2021-06-07:
@ngardiner thank you. I'll try your suggestion. You are right when assuming other platform to determine being home or away.

Another thing I found out within this context is the possibility of set car blocking mode via
> curl -d "chargeAuthorizationMode=1" -X POST http://twcaddress:port/settings/save
> curl -d "chargeAuthorizationMode=2" -X POST http://twcaddress:port/settings/save

It could be a good idea to let us change settings via JSON/POST/HTTP API. But this would be another feature request. ;-)
And yes, I am currently experimenting with stop charging notification too, especially using "Stop responding to slaves". I had some issues with it last days, as settings.json got corrupted using that method. I'm evaluating for the reasons yet and will open another issue, if I'm able to clarify what is causing this corruption.

dehsgr, 2021-06-07:
@ngardiner using
`
POST /api/addConsumptionOffset
{
   "offsetName": "No charging while I'm away",
   "offsetValue": 1000,
   "offset unit": "A"
}
`
currently gives me HTTP 400 with latest pull. Is there sth. other wrong on my site?

dehsgr, 2021-06-07:
So far I got it working. But there seems to be an issue with using "Stop responding" method. If I activate this setting TWC crashes and settings.json is defective.

ngardiner, 2021-06-09:
> So far I got it working. But there seems to be an issue with using "Stop responding" method. If I activate this setting TWC crashes and settings.json is defective.

This has been fixed, it was reported by @magic7music earlier, a ``git pull``` should resolve it for you

For the issue with ```settings.json```, what are you seeing? Could you stop TWCManager and run the contents of ```settings.json``` through a JSON linter to see if there's a parsing error somewhere - all that the change you make to the stop method should do is change:

```"chargeStopMode": 1```

to 

```"chargeStopMode": 2```

in ```settings.json```.

>It could be a good idea to let us change settings via JSON/POST/HTTP API.

That is possible now, but (and it's a big but) it's very easy to break the settings file doing this, because ```settings.json``` represents multi-dimensional dictionaries that we use to store fairly complex settings, and yet the API call takes parameters in key value pairs, so it's not something that is ready to be documented for general use (and may never be, given this complexity). It also does not support saving the settings to disk yet, just changing them at runtime, but in your case it shouldn't be a problem.

```
POST /api/setSetting
{
   "setting": "chargeStopMode",
   "value": 2
}
```

Note that I haven't tested the above and if it does fail that's not entirely unexpected - it's not properly tested or documented yet.

ngardiner, 2021-06-09:
Curiosity did get the better of me :)

This does appear to work although YMMV, it's on the list to put the setSetting API call through some proper test cases, one thing that needs addressing is that it will always return HTTP 204 even if it doesn't get the correct set of parameters and whatnot, but if the query is correct it is usable currently:

```
curl -X POST -d '{ "setting": "chargeStopMode", "value": 2 }' http://127.0.0.1:8080/api/setSetting
```

magic7music, 2021-06-09:
HI Nathan,
   
  
  
   I'll backup my SD card first later during the day.
   
  
  
   I'll try your workaround later during the day (3PM Berlin time)
   
  
  
   
  
  
   Cheers
   
  
  
   Koen

dehsgr, 2021-06-09:
@ngardiner great! That all together solved my needs. Thank you very much and I really appreciate your work!

closed 2021-06-09 by dehsgr