ngardiner/TWCManager #304: TWCM stops running with Errno 98 error

magic7music, 2021-06-08:
Hi guys.

the script stopped running when I tried to charge a non Tesla car.
I did a reboot, git pull of the main branch but still it´s not running.

Here is the output:

pi@TWC-Manager:~/TWCManager $ python3 -m TWCManager
13:30:15 TWCMaster  14 Registered module ConsoleLogging
13:30:15 TWCMaster  14 Released module FileLogging
13:30:15 TWCMaster  14 Released module SentryLogging
13:30:15 TWCMaster  14 Released module CSVLogging
13:30:15 TWCMaster  14 Released module MySQLLogging
13:30:15 TWCMaster  14 Released module SQLiteLogging
13:30:15 TWCMaster  14 Registered module TWCProtocol
13:30:15 TWCMaster  14 Released module Dummy
13:30:15 TWCMaster  14 Registered module RS485
13:30:15 TWCMaster  14 Released module TCP
13:30:15 TWCMaster  14 Registered module Policy
13:30:15 TWCMaster  14 Registered module TeslaAPI
13:30:15 TWCMaster  14 Released module WebIPCControl
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 486, in <module>
    modinstance = modclassref(master)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 51, in __init__
    httpd = ThreadingSimpleServer(("", self.httpPort), HTTPHandler)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/http/server.py", line 137, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use

what could be wrong?

cheers
Koenraad

ngardiner, 2021-06-08:
Hi Koen,

This means something is already listening on port 8080 and the webserver was unable to bind to it. Usually it means that there is already an instance of TWCManager running, perhaps with systemctl if you have installed the systemctl service.

I will update TWCManager to give a nicer error than that when it encounters this issue though :)

magic7music, 2021-06-08:
Hi Nathan,
I found out that on my phone, the website was still in the browser.
After a reboot I got this output.
The strange thing is, that in the web interface, I see that the Charge POint State is o an the Amps offert has the value -1A.
This stays also after a reboot.
This happend after the Nissan leaf was on. 
I tried all 3 posibilities of sending the stop command, Tesla API(I know this won't work but you never know) , stop communicating and sending stop command.

pi@TWC-Manager:~/TWCManager $ python3 -m TWCManager
14:42:08 TWCMaster  14 Registered module ConsoleLogging
14:42:08 TWCMaster  14 Released module FileLogging
14:42:08 TWCMaster  14 Released module SentryLogging
14:42:08 TWCMaster  14 Released module CSVLogging
14:42:08 TWCMaster  14 Released module MySQLLogging
14:42:08 TWCMaster  14 Released module SQLiteLogging
14:42:08 TWCMaster  14 Registered module TWCProtocol
14:42:08 TWCMaster  14 Released module Dummy
14:42:08 TWCMaster  14 Registered module RS485
14:42:08 TWCMaster  14 Released module TCP
14:42:08 TWCMaster  14 Registered module Policy
14:42:08 TWCMaster  14 Registered module TeslaAPI
14:42:08 TWCMaster  14 Released module WebIPCControl
14:42:08 HTTPContro 20 Serving at port: 8080
14:42:08 TWCMaster  14 Registered module HTTPControl
14:42:08 TWCMaster  14 Released module MQTTControl
14:42:08 TWCManager 40 ImportError: No module named 'lib.TWCManager.Control.OCPPControl', when importing Control.OCPPControl, not using Control.OCPPControl
14:42:08 TWCMaster  14 Released module Efergy
14:42:08 TWCMaster  14 Released module Enphase
14:42:08 TWCMaster  14 Released module Fronius
14:42:09 TWCMaster  14 Released module HASS
14:42:09 TWCMaster  14 Released module Kostal
14:42:09 TWCMaster  14 Released module OpenHab
14:42:09 TWCMaster  14 Released module OpenWeatherMap
14:42:09 TWCMaster  14 Released module SmartMe
14:42:09 TWCMaster  14 Registered module SmartPi
14:42:09 TWCMaster  14 Released module SolarEdge
14:42:09 TWCMaster  14 Released module SolarLog
14:42:09 TWCMaster  14 Released module TeslaPowerwall2
14:42:09 TWCMaster  14 Released module TED
14:42:09 TWCMaster  14 Released module Volkszahler
14:42:09 TWCMaster  14 Released module HASSStatus
14:42:09 TWCMaster  14 Released module MQTTStatus
14:42:09 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
14:42:09 TWCMaster  13 Send master linkready1
14:42:09 TWCMaster  13 Send master linkready1
14:42:09 TWCMaster  13 Send master linkready1
14:42:09 TWCMaster  13 Send master linkready1
14:42:09 TWCMaster  13 Send master linkready1
14:42:09 TWCMaster  13 Send master linkready2
14:42:09 TWCMaster  13 Send master linkready2
14:42:10 TWCMaster  13 Send master linkready2
14:42:10 TWCManager 20 32.00 amp slave TWC 6160 is ready to link.  Sign: 3F
14:42:10 TWCManager 20 Set slave TWC 6160 protocolVersion to 2, minAmpsTWCSupports to 6.
Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 927, in <module>
    slaveTWC.send_master_heartbeat()
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 334, in send_master_heartbeat
    if self.master.settings.get("respondToSlavesExpiry", time.time()) > time.time():
NameError: name 'time' is not defined
14:42:10 TWCManager 20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 927, in <module>
    slaveTWC.send_master_heartbeat()
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 334, in send_master_heartbeat
    if self.master.settings.get("respondToSlavesExpiry", time.time()) > time.time():
NameError: name 'time' is not defined

^CTraceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 927, in <module>
    slaveTWC.send_master_heartbeat()
  File "/home/pi/TWCManager/lib/TWCManager/TWCSlave.py", line 334, in send_master_heartbeat
    if self.master.settings.get("respondToSlavesExpiry", time.time()) > time.time():
NameError: name 'time' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 1496, in <module>
    time.sleep(5)
KeyboardInterrupt

ngardiner, 2021-06-08:
That looks like an issue, let me take a quick look. This is on the latest version (v1.2.2)?

magic7music, 2021-06-08:
yes, i´m running the latest main branch

ngardiner, 2021-06-08:
Great, could you please try a ```git pull``` and test again?

magic7music, 2021-06-08:
ok, just one minute

magic7music, 2021-06-08:
the pull is asking to overwrite the etc/twcmanager/config.json file. Should I merge? And what´s the code for that?

ngardiner, 2021-06-09:
> the pull is asking to overwrite the etc/twcmanager/config.json file. Should I merge, commit or stash ? And what´s the code for that?
> Could you please help me on this?? Thanks

Hi Koen,

You should make your changes to ```/etc/twcmanager/config.json``` rather than the one under the git reponsitory (etc/). This is because we update the config template from time to time. The following commands will allow you to pull the latest update:

```
git stash
git pull
git stash pop
```

Once it's updated, if you could run ```make install```, that will copy the config file to ```/etc/twcmanager/config.json``` and you can edit it there going forward

magic7music, 2021-06-09:
Hi Nathan, did it and it's running again including the new features in the web interface. Lets see how stable it runs during the day.
If you want, I can do some logging. I'll send them over later during the day.
Btw: thanks mate and I hate the time difference between our countries.

magic7music, 2021-06-09:
trying to save the settings in the web interface, showed me the following output: 
"We experienced an error trying to save your settings. This mea ns that while your current settings will be retained for the current session, th ey will not be saved if TWCManager exits. Please check the permissions on /etc/t wcmanager/settings.json and try saving your settings again."

This happend in the "Schedule" and "Settings"part of the interface.

ngardiner, 2021-06-09:
@magic7music could you do the following:

```
chown -R pi:pi /etc/twcmanager
```

No need to stop/start TWCManager, just hit save again after running that, and it will save your settings. I'll try and work out why the permissions weren't set right.

magic7music, 2021-06-09:
all 4 operations has been changed but not allowed.
chown: der Eigentümer von '/etc/twcmanager/settings.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/etc/twcmanager/old-config.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/etc/twcmanager/config.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/etc/twcmanager' wird geändert: Die Operation ist nicht erlaubt

ngardiner, 2021-06-09:
How about:

```
sudo chown -R pi:pi /etc/twcmanager
````

magic7music, 2021-06-09:
sudo, yes of course. Stupid me and the lack on python basic knowledge.
It runs without problems.

Saftwerk, 2021-06-11:
I see this user rights problem too.

What about adding this to make install?
`sudo chown -R pi:pi /etc/twcmanager`

MikeBishop, 2021-06-11:
I don't think we can safely assume everyone is running as the user pi, even if it's very common.  Perhaps something like....

    ME=`whoami`
    sudo chown -R $ME:$ME /etc/twcmanager

ngardiner, 2021-06-12:
It is slightly difficult - we've always had a hardcoded chown to root:pi and chmod of 775 in the Makefile, which should work, but perhaps there's a change to group membership that's been introduced into raspi, I'm not entirely sure, and as @MikeBishop says, moving away from relying on pi is the best option.

The problem is that if we use ```whoami```, and someone has run make install with sudo, it will chown it to root. What I am leaning towards is creating our own user/group for the daemon so we can be explicit in which user runs twcmanager, but... I also have to consider that if we want to make packaging easier in the future it should align with python packaging standards so we can just have people pull it from pypi using pip instead of the current way.

I will spend some time researching this and come back with a suggestion on what I think might work most universally

MikeBishop, 2021-06-21:
You could also have `make install` check what user it's running as and refuse to run as root, I suppose.

ngardiner, 2021-06-27:
I tested that out, but it still has a problem, for example if you sudo su - and then run it, it will still see the logged in user as root - and it's just far too hard to have to get people to change back the permissions if they ever inadvertently run it as root. Ultimately after searching around and looking at other projects, the middle ground I've settled on is to create a dedicated ```twcmanager``` user which gives us absolute deterministic permissions.

What this has allowed me to do is:

   * Replace the ```pi``` user hardcoded in Makefile with a ```twcmanager``` user created within the Makefile
   * Replace the ```pi``` user in the systemctl service definition with twcmanager, removing the last of the hardcoded details in the file
   * Ensure we're running with both the uid and gid that we own everything with

The downside is it will only work properly if people run ```make install``` on release of v1.2.3, but that's what the upgrade instructions should say to do anyway as we may introduce new dependencies for a new release. I've already added it to the changelog and it wasn't a terrible time to change things around because I also removed all of the legacy web ui dependencies and steps from the default Makefile install target going forward, so with some testing hopefully we can validate it works for everyone and then, fingers crossed, never see a permissions issue again!

closed 2026-04-14 by ngardiner