ngardiner/TWCManager #307: TWCManager.py "index out of range" after fresh install (13. Jun)

Saftwerk, 2021-06-13:
Today I did a fresh installation with this commands:
```
rm -r TWCManager/TWCManager/
git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout main
make install
```

Then installed my own 3 files 
	`config.json` (see pull request #302 )
	`TWCMaster.py` (I use always '1' phases at line 640)
	`Volkszahler.py` (see pull request #302)

Then tryout
	`python -m TWCManager`

Result is `IndexError: index out of range` at line 171 of `TWCManager.py`
	if msgLen == 0 and data[0] != 0xC0:

Here is "logLevel": 10 all info:
12:19:51 TWCMaster  14 Registered module ConsoleLogging
12:19:51 TWCMaster  14 Released module FileLogging
12:19:51 TWCMaster  14 Released module SentryLogging
12:19:51 TWCMaster  14 Registered module TWCProtocol
12:19:51 TWCMaster  14 Registered module RS485
12:19:51 TWCMaster  14 Registered module Policy
12:19:51 TWCMaster  14 Registered module TeslaAPI
12:19:51 TWCMaster  14 Released module WebIPCControl
12:19:51 HTTPContro 20 Serving at port: 8181
12:19:51 TWCMaster  14 Registered module HTTPControl
12:19:51 TWCMaster  14 Released module Kostal
12:19:51 TWCMaster  14 Released module OpenHab
12:19:51 TWCMaster  14 Released module TeslaPowerwall2
12:19:51 TWCMaster  14 Registered module Volkszahler
12:19:51 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
12:19:51 TWCMaster  13 Send master linkready1
12:19:51 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
12:19:51 TWCMaster  13 Send master linkready1
12:19:51 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
12:19:52 TWCMaster  13 Send master linkready1
12:19:52 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
12:19:52 TWCMaster  13 Send master linkready1
12:19:52 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
12:19:52 TWCMaster  13 Send master linkready1
12:19:52 RS485      12 Tx@: C0 FC E1 77 77 77 00 00 00 00 00 00 00 00 46 C0
12:19:52 TWCMaster  13 Send master linkready2
12:19:52 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
12:19:52 TWCMaster  13 Send master linkready2
12:19:52 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
12:19:52 TWCMaster  13 Send master linkready2
12:19:52 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
12:19:52 TWCMaster  13 Send master linkready2
12:19:52 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
12:19:52 TWCMaster  13 Send master linkready2
12:19:52 RS485      12 Tx@: C0 FB E2 77 77 77 00 00 00 00 00 00 00 00 47 C0
Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 717, in <module>
    if msgLen == 0 and data[0] != 0xC0:
IndexError: index out of range
12:20:05 TWCManager 20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 717, in <module>
    if msgLen == 0 and data[0] != 0xC0:
IndexError: index out of range

Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 717, in <module>
    if msgLen == 0 and data[0] != 0xC0:
IndexError: index out of range
12:20:33 TWCManager 20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 717, in <module>
    if msgLen == 0 and data[0] != 0xC0:
IndexError: index out of range

^CTraceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 717, in <module>
    if msgLen == 0 and data[0] != 0xC0:
IndexError: index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 1561, in <module>
    time.sleep(5)
KeyboardInterrupt

Saftwerk, 2021-06-13:
further analysis prior line 717 showed this:
msgLen = 0
len(data) = 0

ngardiner, 2021-06-13:
We can prevent that exception but it's not causing your issue... your issue is we're not receiving anything over RS485 - so any fix will just stop that error from appearing, it won't make it work any better for you I'm afraid.

Saftwerk, 2021-06-13:
So you think the RS485 might be broken? That would be bad news.

ngardiner, 2021-06-13:
I'm not sure, to be honest. We've never had anything take this code path before, hence the exception you are seeing. We read the buffer length from the RS485 interface, it gives us a result of > 0 yet when we request the data we get an empty response. 

I have added a check which will match this condition to the code, but it's just going to log the above as an error. I am not sure why we'd be told that the RS485 buffer contains data but then receive no data when we request it, the reason no code path exists for that behavior is that it doesn't make sense.

Still, it would be interesting to capture the output on your machine just to see what buffer length it is reporting,. I've pushed the update in commit f7a5e87 and a git pull will update your installation. The message you'll be looking for is:

```We recieved a buffer length of (n) from the RS485 module, but data buffer length is (n). This should not occur.```

Saftwerk, 2021-06-13:
Yes, this should not occur, but it is here.

I will cut the wires now and see where it is broken.
Thank You for support.

13:42:22 TWCMaster  13 Send master linkready2
13:42:25 TWCManager 40 We recieved a buffer length of 1 from the RS485 module, but data buffer lengt                h is 0. This should not occur.
13:42:36 TWCManager 40 We recieved a buffer length of 1 from the RS485 module, but data buffer lengt                h is 0. This should not occur.
Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 710, in <module>
    data = master.getInterfaceModule().read(dataLen)
  File "/home/pi/TWCManager/lib/TWCManager/Interface/RS485.py", line 65, in read
    return self.ser.read(len)
  File "/usr/local/lib/python3.7/dist-packages/pyserial-3.5-py3.7.egg/serial/serialposix.py", line 5                96, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device dis                connected or multiple access on port?)
13:42:48 TWCManager 20 Unhandled Exception:Traceback (most recent call last):
  File "/home/pi/TWCManager/TWCManager.py", line 710, in <module>
    data = master.getInterfaceModule().read(dataLen)
  File "/home/pi/TWCManager/lib/TWCManager/Interface/RS485.py", line 65, in read
    return self.ser.read(len)
  File "/usr/local/lib/python3.7/dist-packages/pyserial-3.5-py3.7.egg/serial/serialposix.py", line 5                96, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device dis                connected or multiple access on port?)

13:42:57 TWCManager 40 We recieved a buffer length of 1 from the RS485 module, but data buffer lengt                h is 0. This should not occur.
13:43:07 TWCManager 40 We recieved a buffer length of 1 from the RS485 module, but data buffer lengt                h is 0. This should not occur.

Saftwerk, 2021-06-16:
Hello ngardiner,

Now , after long search with oscilloscope, Shield, Earth and 120 ohm resistors I found the reason for this:
`TWCManager 40 We recieved a buffer length of 1 from the RS485 module, but data buffer length is 0. This should not occur.`
The raspberry has changed the USB port enumeration. This happened without unplug or swap my 3 USB connectors. After changing the `config.json`  from `"port": "/dev/ttyUSB0"` to `"port": "/dev/ttyUSB2"` the TWCManager worked again.

Here is the fix:
(maybe you want to add this somewhere to your documentation or `config.json`)

List all USB devices:
`ls -l /dev/*USB*`
```
crw-rw---- 1 root dialout 188, 0 Jun 16 17:34 /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 1 Jun 16 17:34 /dev/ttyUSB1
crw-rw---- 1 root dialout 188, 2 Jun 16 17:34 /dev/ttyUSB2
```
If you have only one entry, then you don't need this fix, but you can be prepared for future.

List infos about a device:
`/sbin/udevadm info --query=all --name=/dev/ttyUSB2`
```
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/ttyUSB2/tty/ttyUSB2
N: ttyUSB2
L: 0
S: serial/by-path/platform-3f980000.usb-usb-0:1.4:1.0-port0
S: serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/ttyUSB2/tty/ttyUSB2
E: DEVNAME=/dev/ttyUSB2
E: MAJOR=188
E: MINOR=2
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=5776874
E: ID_VENDOR=1a86
E: ID_VENDOR_ENC=1a86
E: ID_VENDOR_ID=1a86
E: ID_MODEL=USB2.0-Ser_
E: ID_MODEL_ENC=USB2.0-Ser\x21
E: ID_MODEL_ID=7523
E: ID_REVISION=0254
E: ID_SERIAL=1a86_USB2.0-Ser_
E: ID_TYPE=generic
E: ID_BUS=usb
E: ID_USB_INTERFACES=:ff0102:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=ch341
E: ID_USB_CLASS_FROM_DATABASE=Vendor Specific Class
E: ID_VENDOR_FROM_DATABASE=QinHeng Electronics
E: ID_MODEL_FROM_DATABASE=HL-340 USB-Serial adapter
E: ID_PATH=platform-3f980000.usb-usb-0:1.4:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_4_1_0
E: DEVLINKS=/dev/serial/by-path/platform-3f980000.usb-usb-0:1.4:1.0-port0 /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0
E: TAGS=:systemd:
```
From the list choose a name like `ID_SERIAL` , wich would have a unique value from other USB devices.

Edit a rules file:
`sudo nano /etc/udev/rules.d/99-com.rules`
and append a new SUBSYSTEM line with your ID_SERIAL like my template which has now 3 lines:
```
....<truncared>....
	elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi \
'", SYMLINK+="serial%c"

SUBSYSTEM=="tty", ENV{ID_SERIAL_SHORT}=="00F31F50", SYMLINK+="lesekopf0" 
SUBSYSTEM=="tty", ENV{ID_SERIAL_SHORT}=="00F32228", SYMLINK+="lesekopf1"
SUBSYSTEM=="tty", ENV{ID_SERIAL}=="1a86_USB2.0-Ser_", SYMLINK+="USB_TWC"

```
Notes:
Keep an empty line at the very end.
If you replace the RS485 converter, then you have to edit this file again.

Now apply the changed rule:
`sudo udevadm trigger`

and verify the new entry:
`ls -l /dev/*USB*`
```
crw-rw---- 1 root dialout 188, 0 Jun 16 17:34 /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 1 Jun 16 17:34 /dev/ttyUSB1
crw-rw---- 1 root dialout 188, 2 Jun 16 17:34 /dev/ttyUSB2
lrwxrwxrwx 1 root root         7 Jun 16 17:34 /dev/USBTWC -> ttyUSB2
```
Now you can use in `config.json` the new name `"port": "/dev/USBTWC"`

ngardiner, 2021-06-17:
Glad you found the issue @Saftwerk!

closed 2021-06-17 by ngardiner

ngardiner, 2021-06-17:
I have added your instructions to: https://github.com/ngardiner/TWCManager/blob/main/docs/config_examples.md

Thanks for the clear steps, this will be helpful to others with multiple USB to Serial interfaces

tjikkun, 2021-06-17:
Just to add some info:
You could alternatively use `/dev/serial/by-path/platform-3f980000.usb-usb-0:1.4:1.0-port0` or `/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0` in de TCWManager `config.json`. Use `by-path` if you always plug in at the same port, this means it will still work if you replace the converter. If you use `by-id` it doesn't matter which port, as long as you don't replace.