ngardiner/TWCManager #310: v1 TWC display issues in v1.2.2 - fields missing/blank

Squazel, 2021-06-18:
Working correctly in v1.2.1, but now older "v1" HPWCs/TWCs (that came with AU cars in 2015/2016?) are not displaying key data:

[image]

The second row is a v1 charger - differences from v1.2.1 are:
* Truncated leading zero at start of ID
* Missing/blank State, Version, Max Amps, Amps Offered, Amps In Use, Last Heartbeat (previously poplated correctly in v1.2.1)

Similarly in the modern theme:
[image]
* ID is okay, but other key data (version / amps offered / in use) are blank

Note that the unit appears to respond to charging commands and works as expected, so this appears to just be a display issue.

ngardiner, 2021-06-24:
Sorry about that, @Squazel. Going to take a look as a high priority and get that fixed.

Squazel, 2021-08-23:
Hi @ngardiner - although this fix appears to have resolved the missing leading zero in the legacy/default web interface, it does not appear to have addressed any of the other missing data fields. Could you please review/reopen this one?

ngardiner, 2021-08-24:
Thanks for letting me know, @Squazel 

Could I please get you to capture something for me. It might be called something different in your browser, but if you could please go to the TWCManager web interface in Firefox or Chrome and open the Developer Toolbar:

[image]

If you could refresh, and let it run for a while, and then save the HAR file (it's under the cog button in Firefox or menu in Chrome) and email it to me that would help a lot!

ngardiner, 2021-08-31:
Will be fixed in v1.2.4 - thanks again @Squazel for your help troubleshooting this one

commit 7667fe65 (2021-08-01): Backport fix in 97d89bd to legacy interface to fix #310
closed 2021-08-31 by ngardiner