ngardiner/TWCManager #314: Charging time counter: intended display?

VIDGuide, 2021-06-24:
Looking at some of the missing pieces in the modern UI, and I see Charging time counter isn’t wired up yet at all. 

Curious what kind of value you intended here, was it to be a “counter, upwards” of time charging in the current session, or more of a “time remaining” for when a “charge now” has been used, or something completely different? :)

i don’t want to go making assumptions without understanding the plan first :)

ngardiner, 2021-06-25:
The modern web theme was introduced by @hopfi2k (who we're very much indebted to as my frontend design skills are zero) so I might also be speaking out of turn here but my belief is that it would be the time since charging started for that TWC. Hopefully we already have a timestamp in TWCSlave (I haven't had a chance to take a look yet) but if not, we'd need to add it and then it would be accessible as a property of the individual TWC.

VIDGuide, 2021-06-25:
Doesn't look like there is. There is a timpstamp for tracking heartbeats, but not actually charging. 
I can see in TWCMaster where we set isCharging = 1 (or = 0)

So I can add a new property for startTime and set/clear it at these same points, then add it to the AJAX returned data.

VIDGuide, 2021-06-26:
Closing this issue. The PR for this was accepted. I may add some tweaks to its function/layout, but it's operational. I'm thinking about adding a 2nd timer into that column/space for "time left" when a Charge Now is used, count down until it's expired.

closed 2021-06-26 by VIDGuide