ngardiner/TWCManager #318: "We experienced an error trying to save your settings."

Saftwerk, 2021-06-26:
After a complete empty fresh install I started a debug session:
pi@pi3:~/TWCManager $ python -m TWCManager

Then in modern theme `Settings` change any parameter and click `Save Settings`
Then a red warning is displayed: "**We experienced an error trying to save your settings.**"

Then I checked folder `/etc/twcmanager/....`
There was no file `settings.json`

Then I tried the same with sudo:
pi@pi3:~/TWCManager $ `sudo python -m TWCManager`
After changing a setting the file was here `settings.json`.
Note: The file did not have WRITE permissions.

Solution:
Enable WRITE permissions to `settings.json` and then it will work without sudo.

VIDGuide, 2021-06-27:
Did you create the /etc/twcmanager/ folder, or did you let the script/installer create it?
Also, was the main app running under the user 'twcmanager', or just locally in your current user at the time of the initial setup?

Saftwerk, 2021-06-27:
1st I deleted all twcmanager folders:
/etc/twcmanager => /etc/
/home/pi/TWCManager > /home/pi/

Then I did a manual installation as [here](https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md):
```
git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout main
make install
```

ngardiner, 2021-08-29:
Going to close this - this shouldn't occur anymore as we explicitly create a twcmanager user from v1.2.3 onwards and use that user to ensure consistency

closed 2021-08-29 by ngardiner