ngardiner/TWCManager #322: High Amps Offered - Is this just boost to start?

VIDGuide, 2021-07-05:
[image]
I could probably dig into this and find out, but I typically don't get too much time to watch this closely during the day, since I'm usually working. 

Car isn't connected right now, but see the screenshot, Solar is generating 11A, consumption ~2A, meaning a surplus of 8A. Policy is to charge when above 6A Surplus, so I understand that it would be "good to charge" at this point, but of course offering 8A would be the expectation. Is the 14A offered here just a boost/high level to get the car started (I've read about the way this works, sometimes the car needs a higher level offered and then bring it down), or is my policy perhaps not how I expect it to be, allowing it to offer more than I'm generating surplus?

MikeBishop, 2021-07-06:
The TWC doesn't always know whether a car is connected if it's not pulling any power.  If we're offering power and it's not being accepted, TWCManager can get into a loop of changing the amount of power being offered in an effort to "catch the car's attention" as it were.  On my (single-phase, US) system, it oscillates between offering 19A and 21A, IIRC.

If it doesn't settle down to an expected amount of power within a few seconds of being plugged into a car, I'd be more concerned.

VIDGuide, 2021-07-07:
Awesome. Thanks for clarifying that, exactly what I needed. I'll close this issue :)

closed 2021-07-07 by VIDGuide