ngardiner/TWCManager #328: Feature request: Interface to disable/enable scheduled charging

colwilliamsnz, 2021-07-09:
Love this project, it’s a dream come true for this solar energy enthusiast. One thing I’d love to be able to do is easily disable scheduled charging independently of the TWCM UI. Via say MQTT (or any of the interfaces so I can easily expose via Home Assistant).

Effectively a simple enable/disable function for the entire scheduler via a single click.

My reasoning is:

We actively track our solar output including forecast generation and work our day to day lives around it. Some nights I don’t want the scheduled charging to kick in because I know the next day(s) are going to be sunny and I’d much prefer to delay charging and utilise the excess solar instead.

Being able to click a single button in the UI (or ideally Home Assistant) means I can easily give this access to my family (and/or do it remotely if I’m not at home during the day but the car is).

Such a feature would open up the possibility of using automation via Node-Red or the likes to do something like: use Forecast.Solar to determine next days likely generation and so automatically disable the overnight schedule for the next 24 hours.

Thoughts?

MikeBishop, 2021-07-10:
I effectively do that on mine -- you can use the restrictions node to add additional constraints to when Scheduled Charging runs, but the thing you set has to be something accessible by a policy.  In my case, I have a Tesla Powerwall and change the reserve percentage based on the next day's forecast; I then can gate Scheduled Charging on whether the reserve percentage is high or low.

Admittedly that's indirect; it would be cleaner to have a way for, say, the MQTT module to be able to query a particular value within a policy rule and something else could set it.

colwilliamsnz, 2021-07-10:
I found I can use the HTTP control, setScheduledChargingSettings, to disable the schedule as follows (and the opposite JSON payload to turn it back on). Seems to work fine but the TWCM UI settings no longer reflect what is actually happening...I guess a small price to pay for the convenience of a single button inside Home Assistant.

{
    "enabled": true,
    "startingMinute": 1380,
    "endingMinute": 420,
    "monday": true,
    "tuesday": true,
    "wednesday": true,
    "thursday": true,
    "friday": true,
    "saturday": true,
    "sunday": true,
    "amps": 20,
    "flexBatterySize": 50,
    "flexStartEnabled": false
}