ngardiner/TWCManager #330: HTTPControl.py separate for legacy template?

hopfi2k, 2021-07-11:
I finally found some time to work on improving the Modern template (e.g. Bootstrap 5.x, better responsive support etc.) and also creating templates files for some of the menu links (vehicles/schedule...).

The current `HTTPControl.py` file is a historically grown file that still includes a lot of GUI stuff, even if we are using templates. Some parts for the templates are still rendered by `HTTPControl.py` (navigation bar) which makes it hard to create proper template...

Could we improve `HTTPControl.py` by moving most of it (e.g. as legacy version) to the Default theme folder and create a more streamlined version that uses the Modern Theme (and other future themes)? I'd love to hear @ngardiner thoughts on such a move as it would require to modify loading `HTTPControl.py` from the theme folder (or creating a wrapper that loads it from the theme folder).

ngardiner, 2021-07-11:
Hi @hopfi2k,

Great to hear from you and thanks for all of your work on the UI.

To give a bit of context to why there is presentation logic in HTTPControl.py, this is because prior to the Modern UI that you developed, we didn't have a templating layer nor themes. At the time, everything had been interspersed within the functions.

To allow the development of themes like yours, jinja2 was placed inbetween the processing and rendering, but because this would have required complete redevelopment of the legacy web UI prior to you being able to add the Modern UI interface, a shortcut was taken where the existing code was retained within those functions for the legacy UI, and are called by jinja2 to render the content.

Those functions could be simply removed if one of two things happens:

   * The legacy UI is completely removed from TWCManager

or 

  * The Legacy UI is rewritten in jinja2 completely (or partially, for any of the parts that will remain such as the nav bar, but are still rendered within the code, although 

For option 1 to be possible, we need to ensure that the Modern UI covers all use cases. Given we only just made it the default in v1.2.3 and that's the development version rather than stable, it's a bit too soon for that. That said, it is by and large the most popular UI now and I haven't had any feedback that it isn't suitable in any install so far, so that time may not be far off

For option 2 to happen, we'd need to put in development work into the Legacy UI - but if we were going for Option 1, we may as well just focus on the deprecation of the Legacy UI, at which point we just remove all of the functions that are orphaned because they were only used by the Legacy UI

So ultimately those functions are accelerating towards an impending deprecation anyway, but we need a bit more time to have people move onto the new UI and validate that there's no need for the Legacy UI going forward, as once we remove it that's final, there's much less pain phasing it out than trying to quickly code any delta into the new interface.

So the short answer is - there's presentation logic in HTTPControl for the Legacy UI because it was written before Jinja2 and templates existed as a construct. Anything written from now on should not use that approach (and wouldn't get merged) because we now have fully functional templating that removes the need for this. Once Legacy UI is pronounced dead, all of those functions will be removed. 

In the meantime, if you want to override any of those functions you can do so simply by replacing the jinja2 template within your theme - eg replace ```lib/TWCManager/Control/themes/Modern/navbar.html.j2``` with whatever content you want to override for the nav bar when your theme is active.

hopfi2k, 2021-07-12:
Hi @ngardiner ,

Thanks for the brief replay and the valuable remarks you made!

While I totally agree with your suggestions, I'm honestly not too happy with the choices (its like choice between plague and cholera...). The legacy theme is hardwired at so many places into HTTPControl.py that it is really, really hard to break this up into a true template based approach. Nevertheless I'll try my best to get around this!

hopfi2k, 2021-07-14:
Full theme rewrite using Bootstrap 5 coming along nicely. Using flex box now, so theme is going to be fully responsive. Removed most of the previous custom CSS. Next: getting rid of jQuery.

[image]

closed 2022-02-25 by ngardiner