ngardiner/TWCManager #348: TWC wont start P1 error

Darosnl, 2021-08-05:
twcmanager_1  | 17:34:59 HTTPContro 20 Serving at port: 8080
twcmanager_1  | --- Logging error ---
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/P1Monitor.py", line 26, in __init__
twcmanager_1  |     self.configP1Mon = master.config["sources"]["P1Monitor"]
twcmanager_1  | KeyError: 'P1Monitor'
twcmanager_1  | 
twcmanager_1  | During handling of the above exception, another exception occurred:
twcmanager_1  | 
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 1034, in emit
twcmanager_1  |     msg = self.format(record)
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 880, in format
twcmanager_1  |     return fmt.format(record)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/Logging/ConsoleLogging.py", line 20, in format
twcmanager_1  |     s = super(ColorFormatter, self).format(record)
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 619, in format
twcmanager_1  |     record.message = record.getMessage()
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 380, in getMessage
twcmanager_1  |     msg = msg % self.args
twcmanager_1  | TypeError: not all arguments converted during string formatting
twcmanager_1  | Call stack:
twcmanager_1  |   File "TWCManager.py", line 20, in <module>
twcmanager_1  |     import TWCManager.TWCManager
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
twcmanager_1  |   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 552, in <module>
twcmanager_1  |     modinstance = modclassref(master)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/P1Monitor.py", line 33, in __init__
twcmanager_1  |     logger.error("Cannot get configuration for P1Monitor in config.json", e)
twcmanager_1  | Message: 'Cannot get configuration for P1Monitor in config.json'
twcmanager_1  | Arguments: (KeyError('P1Monitor'),)
twcmanager_1  | --- Logging error ---
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/P1Monitor.py", line 26, in __init__
twcmanager_1  |     self.configP1Mon = master.config["sources"]["P1Monitor"]
twcmanager_1  | KeyError: 'P1Monitor'
twcmanager_1  | 
twcmanager_1  | During handling of the above exception, another exception occurred:
twcmanager_1  | 
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 1034, in emit
twcmanager_1  |     msg = self.format(record)
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 880, in format
twcmanager_1  |     return fmt.format(record)
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 619, in format
twcmanager_1  |     record.message = record.getMessage()
twcmanager_1  |   File "/usr/lib/python3.7/logging/__init__.py", line 380, in getMessage
twcmanager_1  |     msg = msg % self.args
twcmanager_1  | TypeError: not all arguments converted during string formatting
twcmanager_1  | Call stack:
twcmanager_1  |   File "TWCManager.py", line 20, in <module>
twcmanager_1  |     import TWCManager.TWCManager
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
twcmanager_1  |   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
twcmanager_1  |   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 552, in <module>
twcmanager_1  |     modinstance = modclassref(master)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/P1Monitor.py", line 33, in __init__
twcmanager_1  |     logger.error("Cannot get configuration for P1Monitor in config.json", e)
twcmanager_1  | Message: 'Cannot get configuration for P1Monitor in config.json'
twcmanager_1  | Arguments: (KeyError('P1Monitor'),)
twcmanager_1  | Traceback (most recent call last):
twcmanager_1  |   File "TWCManager.py", line 20, in <module>
twcmanager_1  |     import TWCManager.TWCManager
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 552, in <module>
twcmanager_1  |     modinstance = modclassref(master)
twcmanager_1  |   File "/usr/src/TWCManager/lib/TWCManager/EMS/P1Monitor.py", line 36, in __init__
twcmanager_1  |     if not self.serverIP:
twcmanager_1  | AttributeError: 'P1Monitor' object has no attribute 'serverIP'

Darosnl, 2021-08-05:
When i try to start the docker-compose i got the following error as above.

mvaneijken, 2021-08-05:
Did you configure P1Monitor as described in the docs?
https://github.com/ngardiner/TWCManager/blob/main/docs/modules/EMS_P1Monitor.md

mvaneijken, 2021-08-05:
Not sure, but when you are not using the P1Monitor EMS integration this issue happens because the EMS was enabled by default in the config provided. Just set `"enabled": false` at the P1Monitor config entry in `/etc/twcmanager/config.json`

Darosnl, 2021-08-06:
I got an working config and TWC but dit een docker pull and now i got the error.
I dont have an P1 in the config.json so i added it but on the wrong place.

I added it to CONTROL: now i added it to STATUS: and it working, i added:
"P1Monitor": {
          # Support for P1 Monitor API (https://www.ztatz.nl/)
          # Reads /api/v1/phase
          "enabled": false,
          "serverIP":"192.168.1.2",
          # Amount of samples to take average from (min 1, max 10, default 1)
          "samples": 1
        },

closed 2021-08-06 by Darosnl