ngardiner/TWCManager #351: ERROR: no matching manifest for linux/arm/v7 in the manifest list entries

KirkKirk, 2021-08-06:
@ngardiner, I tried to deal with this error but I got stuck with it.
I made a fresh install on my Pi3 from scratch, but still, It works only when  image: twcmanager/twcmanager:v1.2.2
Is there any guidance on it? 

pi@TWCManager:~/TWCManager $ docker-compose -f contrib/docker/docker-compose.yml up
Creating network "docker_default" with the default driver
Pulling twcmanager (twcmanager/twcmanager:latest)...
latest: Pulling from twcmanager/twcmanager
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries

pi@TWCManager:~/TWCManager $ docker-compose -f contrib/docker/docker-compose.yml up
Pulling twcmanager (twcmanager/twcmanager:main)...
ERROR: manifest for twcmanager/twcmanager:main not found: manifest unknown: manifest unknown

pi@TWCManager:~/TWCManager $ docker-compose -f contrib/docker/docker-compose.yml up
Pulling twcmanager (twcmanager/twcmanager:v1.2.3)...
ERROR: manifest for twcmanager/twcmanager:v1.2.3 not found: manifest unknown: manifest unknown

pi@TWCManager:~/TWCManager $ docker-compose -f contrib/docker/docker-compose.yml up
Creating docker_twcmanager_1 ... done
Attaching to docker_twcmanager_1
twcmanager_1  | 22:03:21 HTTPContro 20 Serving at port: 8080
twcmanager_1  | 22:03:21 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
twcmanager_1  | 22:03:22 TWCManager 20 32.00 amp slave TWC 1406 is ready to link.  Sign: 85

ngardiner, 2021-08-08:
I'll take a look at this now, apologies for the latest image being
unavailable.

The issue is that the websockets dependency was broken for armv7 and was
causing all docker builds to break so I took it out of the build
architecture list as I didn't have a chance to work out why.

I'll revisit it now and advise when it is available again.

ngardiner, 2021-08-10:
Thanks for your patience,

Took a bit of work, but docker builds are working again. Hopefully it should pull down properly for you, and there's an imminent release of v1.2.3 that is going to get pushed out shortly too. Let me know if you have any difficulties with the docker images

KirkKirk, 2021-08-11:
Works perfectly!
On behalf of everyone who uses it, THANK YOU for all your hard work.

Cheers

KirkKirk, 2021-08-14:
Closed as it is ok.

closed 2021-08-14 by KirkKirk