ngardiner/TWCManager #352: Can't enable the service.

soso1999, 2021-08-08:
Hi there,
i can't enable the service. I get this message:

pi@twcmanager:~ $ journalctl -f
-- Logs begin at Sun 2021-08-08 10:46:20 BST. --
Aug 08 10:48:59 twcmanager systemd[1215]: twcmanager.service: Failed to determine user credentials: No such process
Aug 08 10:48:59 twcmanager systemd[1215]: twcmanager.service: Failed at step USER spawning /usr/bin/python3: No such process
Aug 08 10:48:59 twcmanager systemd[1]: twcmanager.service: Main process exited, code=exited, status=217/USER
Aug 08 10:48:59 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Aug 08 10:49:00 twcmanager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Aug 08 10:49:00 twcmanager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Aug 08 10:49:00 twcmanager systemd[1]: Stopped Tesla Wall Charger Manager.
Aug 08 10:49:00 twcmanager systemd[1]: twcmanager.service: Start request repeated too quickly.
Aug 08 10:49:00 twcmanager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Aug 08 10:49:00 twcmanager systemd[1]: Failed to start Tesla Wall Charger Manager.

With python -m TWCManager service starts without issues.

pi@twcmanager:~/TWCManager $ python -m TWCManager
10:50:03 HTTPContro 20 Serving at port: 8080
10:50:03 TWCManager 40 ImportError: No module named 'growattServer', when importing EMS.Growatt, not using EMS.Growatt
10:50:04 TWCManager 40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
10:50:04 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77

ngardiner, 2021-08-08:
Did you run make install? If not, you need to for the service to work.

soso1999, 2021-08-08:
Yes i did.

soso1999, 2021-08-08:
> Did you run make install? If not, you need to for the service to work.
> […](#)

With python -m TWCManager service starts without issues.

ngardiner, 2021-08-08:
> With python -m TWCManager service starts without issues.

This is the wrong way to run it, though. It has to run as the twcmanager user or your permissions will be all wrong.

The twcmanager user is created by the makefile.

What documentation are you following?

ngardiner, 2021-08-08:
Also, if you moved from an earlier version, you need to run make install again. Here's the detail from the changelog:

v1.2.3 - Current development version

    NOTE: v1.2.3 contains a potentially breaking change for users of OpenWB or the Legacy Web Interface
        As of v1.2.3, the Legacy Web Interface and its dependencies such as lighttpd and php are no longer installed by default.
        Please see the documentation for instructions on how to install it, if required. Most users will not need this.
    NOTE: Starting from v1.2.3, Modern theme is the default theme for TWCManager, unless configured otherwise.
    NOTE: v1.2.3 introduces a dedicated user account for TWCManager. For Manual install, you'll need to run make install after upgrading.

soso1999, 2021-08-08:
> > With python -m TWCManager service starts without issues.
> 
> This is the wrong way to run it, though. It has to run as the twcmanager user or your permissions will be all wrong.
> 
> The twcmanager user is created by the makefile.
> 
> What documentation are you following?

I followed your documentation. 

I did now this:
sudo nano /etc/rc.local
sudo nohup python /home/pi/TWCManager/TWCManager.py &
and service starts on reboot.
But it is only a workaround, i know.

ngardiner, 2021-08-10:
Okay, you are absolutely free to run it that way if you'd like, but all you need to do is a make install per the v1.2.3 changelog above

I know that hasn't been done because the logs you pasted show dependencies added recently that aren't available:

```
10:50:03 TWCManager 40 ImportError: No module named 'growattServer', when importing EMS.Growatt, not using EMS.Growatt
10:50:04 TWCManager 40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
```

These only get updated when you run ```make install``` again. Since it's in the changelog that you'll need to do this for the service to run (and I'm about to release v1.2.3 so everyone will need to do this when they upgrade) I have to close this issue off because this is a known breaking change and that's why it's documented in the changelog.

closed 2021-08-10 by ngardiner