ngardiner/TWCManager #353: docker - no such file or directory

drexcia, 2021-08-08:
Hi All,

Complete noob here. Have spent most of the weekend trying to get 1.2.3 going on a new pi zero but having no luck. I have followed the instructions here:

https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Docker.md

The first issue I come across is that after I install docker-compose I get command not found. Then I installed docker compose using apt which works fine. I then run this: 

socker-compose -f contrib/docker/docker-compose.yml up

And get this:

pi@twcmanager:~ $ docker-compose -f contrib/docker/docker-compose.yml up
ERROR: .FileNotFoundError: [Errno 2] No such file or directory: './contrib/docker/docker-compose.yml'
pi@twcmanager:~ $ docker-compose -f contrib/docker/docker-compose.yml up

Any ideas? I receive the same error regardless if I go with stable or dev.

ngardiner, 2021-08-08:
Sorry, that is an issue with the documentation that I will correct. The
correct command is:

docker-compose -f docker-compose.yml up

drexcia, 2021-08-08:
Thank you. I receive permission denied when I ran this (last line of the error below) but then I ran this via sudo and I receive the manifest error which has already been reported. 

docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', PermissionError(13, 'Permission denied'))
pi@twcmanager://home/pi/TWC $ sudo docker-compose -f docker-compose.yml up
Creating network "twc_default" with the default driver
Pulling twcmanager (twcmanager/twcmanager:latest)...
latest: Pulling from twcmanager/twcmanager
ERROR: no matching manifest for unknown in the manifest list entries

ngardiner, 2021-08-08:
Cheers. Will reply to #351 shortly.

drexcia, 2021-08-08:
Thanks Nathan!

closed 2021-08-08 by drexcia