ngardiner/TWCManager #354: docker - exiting with error code 139

drexcia, 2021-08-08:
Hi, 

Fresh install on Raspberry Pi Zero using the docker instructions, except I have downloaded docker-composer using apt-get. Receiving an error code 139 on startup. 

pi@twcmanager:~ $ sudo docker-compose -f docker-compose.yml up
Creating network "pi_default" with the default driver
Pulling twcmanager (twcmanager/twcmanager:v1.2.2)...
v1.2.2: Pulling from twcmanager/twcmanager
89475607b1df: Pull complete
da38a388f362: Pull complete
e79e893cf28e: Pull complete
4a94bd5baf34: Pull complete
a5cdd1fa90e4: Pull complete
Digest: sha256:510fd50521450abe103cf99ecaf559733fcdddc5322eaae3de5bad72376fb82e
Status: Downloaded newer image for twcmanager/twcmanager:v1.2.2
Creating pi_twcmanager_1 ... done
Attaching to pi_twcmanager_1
pi_twcmanager_1 exited with code 139
pi@twcmanager:~ $

ngardiner, 2021-08-10:
That doesn't look good... code 139 is a segfault (SIGSEGV) which might mean the pi zero is out of memory, however the pi zero is a different arm architecture (arm6l) which also might be the reason. I've just started pushing arm6l docker images as well, would you please try editing your docker-compose.yml file and change twcmanager:v1.2.2 to twcmanager:latest and see if it works for you?

If so, I'm about to upload twcmanager:v1.2.3 and it will have arm6l support as well

drexcia, 2021-08-10:
Same result unfortunately, I should have mentioned that I had already tried that.

pi@twcmanager:~ $ sudo docker-compose -f docker-compose.yml up
Starting pi_twcmanager_1 ... done
Attaching to pi_twcmanager_1
pi_twcmanager_1 exited with code 139
pi@twcmanager:~ $

drexcia, 2021-08-10:
Please disregard, I was running an old version of docker-compose (v1.21). I have upgraded this to 1.29 and I'm no longer getting this error.

closed 2021-08-10 by drexcia