ngardiner/TWCManager #355: Python exception with 1.2.3 only when running as a service

drexcia, 2021-08-10:
Finally got 1.2.3 installed manually. When I try and browse to twcmanager.local:8080 using "sudo -u twcmanager python -m TWCManager" I am able to see the page fine but when I enable the service following these instructions:

sudo cp contrib/twcmanager.service /etc/systemd/system/twcmanager.service
sudo systemctl enable twcmanager.service --now

I receive a blank page and the below exception in the journalctl -f logs:

ug 10 13:37:06 twcmanager python3[1304]: Exception happened during processing of request from ('192.168.1.5', 64229)
Aug 10 13:37:06 twcmanager python3[1304]: Traceback (most recent call last):
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
Aug 10 13:37:06 twcmanager python3[1304]:     self.finish_request(request, client_address)
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
Aug 10 13:37:06 twcmanager python3[1304]:     self.RequestHandlerClass(request, client_address, self)
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager-1.2.3-py3.7.egg/TWCManager/Control/HTTPControl.py", line 146, in __init__
Aug 10 13:37:06 twcmanager python3[1304]:     BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
Aug 10 13:37:06 twcmanager python3[1304]:     self.handle()
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/lib/python3.7/http/server.py", line 426, in handle
Aug 10 13:37:06 twcmanager python3[1304]:     self.handle_one_request()
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
Aug 10 13:37:06 twcmanager python3[1304]:     method()
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager-1.2.3-py3.7.egg/TWCManager/Control/HTTPControl.py", line 721, in do_GET
Aug 10 13:37:06 twcmanager python3[1304]:     self.template = self.templateEnv.get_template("main.html.j2")
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/Jinja2-2.11.2-py3.7.egg/jinja2/environment.py", line 883, in get_template
Aug 10 13:37:06 twcmanager python3[1304]:     return self._load_template(name, self.make_globals(globals))
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/Jinja2-2.11.2-py3.7.egg/jinja2/environment.py", line 857, in _load_template
Aug 10 13:37:06 twcmanager python3[1304]:     template = self.loader.load(self, name, globals)
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/Jinja2-2.11.2-py3.7.egg/jinja2/loaders.py", line 115, in load
Aug 10 13:37:06 twcmanager python3[1304]:     source, filename, uptodate = self.get_source(environment, name)
Aug 10 13:37:06 twcmanager python3[1304]:   File "/usr/local/lib/python3.7/dist-packages/Jinja2-2.11.2-py3.7.egg/jinja2/loaders.py", line 197, in get_source
Aug 10 13:37:06 twcmanager python3[1304]:     raise TemplateNotFound(template)
Aug 10 13:37:06 twcmanager python3[1304]: jinja2.exceptions.TemplateNotFound: main.html.j2
Aug 10 13:37:06 twcmanager python3[1304]: ----------------------------------------

ngardiner, 2021-08-10:
Hi,

This should be fixed now. You can fix it with a:

```
git pull
sudo python3 setup.py install
```

And then restart the twcmanager service:

```
sudo service twcmanager restart
```

drexcia, 2021-08-10:
Thank you! Working now.

closed 2021-08-10 by drexcia