ngardiner/TWCManager #356: Unable to manually install v1.2.3 (numpy and scipy issue in make install)

mikey4321, 2021-08-11:
performed a fresh gitpull of v1.2.3 and can't get past the following errors when I follow the install steps

Pi OS Buster
Python version 3.7.3
Python packages (from "pip list"):

`Package            Version
------------------ ---------
asn1crypto         0.24.0
attrs              20.3.0
certifi            2018.8.24
cffi               1.14.5
chardet            3.0.4
commentjson        0.9.0
cryptography       3.4.7
entrypoints        0.3
formatizer         0.1.1
future             0.18.2
growattServer      1.0.1
idna               2.6
importlib-metadata 4.0.0
Jinja2             2.11.2
jsonschema         3.2.0
keyring            17.1.1
keyrings.alt       3.1.1
lark-parser        0.7.8
MarkupSafe         1.1.1
numpy              1.21.1
ocpp               0.8.1
paho-mqtt          1.5.1
pip                21.2.2
pycparser          2.20
pycrypto           2.6.1
PyGObject          3.30.4
pyModbusTCP        0.1.8
PyMySQL            1.0.2
pyrsistent         0.17.3
pyserial           3.5
python-apt         1.8.4.3
pyxdg              0.25
requests           2.25.1
scipy              1.7.1
SecretStorage      2.3.1
sentry-sdk         1.0.0
setuptools         57.4.0
six                1.12.0
ssh-import-id      5.7
sysv-ipc           1.0.1
termcolor          1.1.0
TWCManager         1.2.3
typing-extensions  3.7.4.3
urllib3            1.24.1
websockets         9.1
wheel              0.32.3
ww                 0.2.1
zipp               3.4.1`

When I run sudo make install, it works through and tries to run the setup.py subfile.  Even though I have numpy 1.21.1 and the dependency scipy 1.7.1 already installed, it tries to re-download scipy and then throughs the following error as it can't find numpy

`Installed /usr/local/lib/python3.7/dist-packages/TWCManager-1.2.3-py3.7.egg
Processing dependencies for TWCManager==1.2.3
Searching for scipy>=1.7.0
Reading https://pypi.org/simple/scipy/
Downloading https://files.pythonhosted.org/packages/47/33/a24aec22b7be7fdb10ec117a95e1e4099890d8bbc6646902f443fc7719d1/scipy-1.7.1.tar.gz#sha256=6b47d5fa7ea651054362561a28b1ccc8da9368a39514c1bbf6c0977a1c376764
Best match: scipy 1.7.1
Processing scipy-1.7.1.tar.gz
Writing /tmp/easy_install-4h0hjzyl/scipy-1.7.1/setup.cfg
Running scipy-1.7.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-4h0hjzyl/scipy-1.7.1/egg-dist-tmp-pmoyxy2b
/tmp/easy_install-4h0hjzyl/scipy-1.7.1/setup.py:492: UserWarning: Unrecognized setuptools command ('-q bdist_egg --dist-dir /tmp/easy_install-4h0hjzyl/scipy-1.7.1/egg-dist-tmp-pmoyxy2b'), proceeding with generating Cython sources and expanding templates
  ' '.join(sys.argv[1:])))
Error: 'numpy' must be installed before running the build.

error: Setup script exited with 1`

Any suggestions on what might be causing the issue would be greatly appreciated

Mike

ngardiner, 2021-08-11:
Strange - it's trying to reinstall the same version of scipy that you're already running, and complaining that numpy isn't installed when it is.

I'm not sure what's causing this but I will look around and see if there's any similar cases.

wooter, 2021-08-11:
I had the numpy issue before, but after updating pip I was good to go.

Now I have 
`Running scipy-1.7.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-b2y_h7ls/scipy-1.7.1/egg-dist-tmp-xishojyb
/tmp/easy_install-b2y_h7ls/scipy-1.7.1/setup.py:492: UserWarning: Unrecognized setuptools command ('-q bdist_egg --dist-dir /tmp/easy_install-b2y_h7ls/scipy-1.7.1/egg-dist-tmp-xishojyb'), proceeding with generating Cython sources and expanding templates
  ' '.join(sys.argv[1:])))
Error: 'None' must be installed before running the build.`

wooter, 2021-08-11:
> I had the numpy issue before, but after updating pip I was good to go.
>
> Now I have
>
> ```
> Running scipy-1.7.1/setup.py -q bdist_egg --dist-dir [...]

I fixed it by first checking Makefile, where I noticed all commands are run as sudo. By manually doing `sudo pip3 install scipy`the root account also installed scipy for itself. Re-running `make install` did not give any errors anymore.

It appears this part of Makefile does not recognise /usr/bin/pip3 and tries to install dependencies with setup.py```

ifeq ($(CI), 1)
	$(SUDO) /home/docker/.pyenv/shims/python3 setup.py install
else
ifneq (,$(wildcard /usr/bin/pip3))
	$(SUDO) pip3 install --upgrade setuptools
else
ifneq (,$(wildcard /usr/bin/pip))
	$(SUDO) pip install --upgrade setuptools
endif
endif
	$(SUDO) ./setup.py install
endif
```

soso1999, 2021-08-11:
I have a same issue! Maybe it is because i use raspberry w zero?

contactmike1, 2021-08-12:
> I fixed it by first checking Makefile, where I noticed all commands are run as sudo. By manually doing `sudo pip3 install scipy`the root account also installed scipy for itself. Re-running `make [...]

I have the same issue.  First I had to install pip3, then I installed scipy and nimpy. I found it interesting I got this error after doing the manual install for scipy and nimpy. 

[image]

It doesn't fix the issue that occurs with "make install".  Is there a problem using raspbian lite on a pi zero?
[image]

Last question: Is there somewhere to manually stick my tesla key and refresh token inside a file or config with an older version of TWCManager? I am trying to do a first time install, and TWCManager doesn't appear to work correctly without the token. The issues log says I must use 1.2.3 for the get token part to work, but this main branch doesn't appear to be able to install currently. Is there a way to get a token to work with an older version (if I generate my own), or are brand new installs just out of luck for the time being?

ngardiner, 2021-08-12:
> It appears this part of Makefile does not recognise /usr/bin/pip3 and tries to install dependencies with setup.py
>
> ```
> ifeq ($(CI), 1)
> 	$(SUDO) /home/docker/.pyenv/shims/python3 setup.py install [...]

We do install dependencies with setuptools - the reason for us looking for the correct pip executable in the Makefile is to upgrade the version of setuptools before we run setup.py, since earlier versions of setuptools didn't understand the directory structure that we use and would throw an error.

We don't install dependencies with pip, however.

ngardiner, 2021-08-12:
Hi folks,

The issue is related to the ```numpy``` and ```scipy``` depdencies introduced by the P1Monitor EMS module. For these to be compiled, cython needs to be installed. I'm working to try and tune the dependencies to take care of this and avoid a long installation due to compilation by leveraging piwheels to download the dependencies. If you want a quick fix, you can delete references to ```scipy``` and ```numpy``` in setup.py, but be aware you'll need to do a ```git stash``` before your next ```git pull``` to avoid a conflict due to your manual local changes. I hope to push out a fixed version shortly - and we'll add some controls to avoid this happening again.

ngardiner, 2021-08-12:
Okay, I've just pushed out a new v1.2.3 tag so this should bypass the issue for now. Just to explain a little further:

- We added the numpy and scipy depdencies as we would any other for modules
- They're slightly different though as they are very heavy packages with a lot of C and if you don't have a precompiled wheel for your given architecture (eg arm6, arm7) it has to compile them from source
- I was originally going to point the setuptools script to precompiled piwheels wheels but ran into a separate issue (other packages install a local package, python3-numpy which is an older version incompatible with scipy) and I realised at this point I can't test this on different raspberry pi machines because we don't have any to test on as part of our CI tests, we test on normal intel machines

I try to avoid anything like this that is likely to fail or behave differently due to different hardware or software versions, we have a set of test environments to pick that sort of thing up rather than worrying about it manually.

So I've just removed them for now and added some instructions in the docs for anyone who does use the P1Monitor module to manually install them. Meanwhile, I'll look for a way to test our builds on actual raspberry pis to pick this up if it ever happened again in the future, since it's specific to arm6 and arm7 machines which some RPis are.

mikey4321, 2021-08-12:
Thanks @ngardiner.

Unfortunately it is failing in a new way now - still trying to install cryptography, despite the fact I have version 3.4.7 and the dependency is 3.4.    Not sure if this is a general issue, or if in the process of trying to fix it I've borked some permissions.

`Installed /usr/local/lib/python3.7/dist-packages/websockets-9.1-py3.7-linux-armv6l.egg
Searching for cryptography>=3.4
Reading https://pypi.org/simple/cryptography/
Downloading https://files.pythonhosted.org/packages/9b/77/461087a514d2e8ece1c975d8216bc03f7048e6090c5166bc34115afdaa53/cryptography-3.4.7.tar.gz#sha256=3d10de8116d25649631977cb37da6cbdd2d6fa0e0281d014a5b7d337255ca713
Best match: cryptography 3.4.7
Processing cryptography-3.4.7.tar.gz
Writing /tmp/easy_install-8sq215ya/cryptography-3.4.7/setup.cfg
Running cryptography-3.4.7/setup.py -q bdist_egg --dist-dir /tmp/easy_install-8sq215ya/cryptography-3.4.7/egg-dist-tmp-87zrboo3

        =============================DEBUG ASSISTANCE==========================
        If you are seeing an error here please try the following to
        successfully install cryptography:

        Upgrade to the latest pip and try again. This will fix errors for most
        users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
        =============================DEBUG ASSISTANCE==========================
        
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 152, in save_modules
    yield saved
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 193, in setup_context
    yield
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 254, in run_setup
    _execfile(setup_script, ns)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 43, in _execfile
    exec(code, globals, locals)
  File "/tmp/easy_install-8sq215ya/cryptography-3.4.7/setup.py", line 14, in <module>
    "commentjson < 0.9.0; python_version < '3.6'",
ModuleNotFoundError: No module named 'setuptools_rust'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./setup.py", line 43, in <module>
    url="https://github.com/ngardiner/twcmanager",
  File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/install.py", line 117, in do_egg_install
    cmd.run(show_deprecation=False)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 408, in run
    self.easy_install(spec, not self.no_deps)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 650, in easy_install
    return self.install_item(None, spec, tmpdir, deps, True)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 697, in install_item
    self.process_distribution(spec, dist, deps)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 745, in process_distribution
    [requirement], self.local_index, self.easy_install
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 768, in resolve
    replace_conflicting=replace_conflicting
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 1051, in best_match
    return self.obtain(req, installer)
  File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 1063, in obtain
    return installer(requirement)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 669, in easy_install
    return self.install_item(spec, dist.location, tmpdir, deps)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 695, in install_item
    dists = self.install_eggs(spec, download, tmpdir)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py", line 890, in install_eggs
[... 4 lines cut ...]
    run_setup(setup_script, args)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 257, in run_setup
    raise
  File "/usr/lib/python3.7/contextlib.py", line 130, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 193, in setup_context
    yield
  File "/usr/lib/python3.7/contextlib.py", line 130, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 164, in save_modules
    saved_exc.resume()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 139, in resume
    raise exc.with_traceback(self._tb)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 152, in save_modules
    yield saved
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 193, in setup_context
    yield
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 254, in run_setup
    _execfile(setup_script, ns)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/sandbox.py", line 43, in _execfile
    exec(code, globals, locals)
  File "/tmp/easy_install-8sq215ya/cryptography-3.4.7/setup.py", line 14, in <module>
    "commentjson < 0.9.0; python_version < '3.6'",
ModuleNotFoundError: No module named 'setuptools_rust'
make: *** [Makefile:66: setup] Error 1`

drexcia, 2021-08-12:
I thought it was just me who had issues with the manual install, I managed to get it going by installing Scipy and a bunch of other modules before getting it working through docker (eventually)

Not sure if this helps but for the manual install my history file shows that I had to run: 

sudo pip3 install setuptools_rust
pip install cryptography (which appears you already have)
sudo apt-get install build-essential libssl-dev libffi-dev python3-dev cargo
pip install -U pip
make install

mikey4321, 2021-08-13:
`sudo pip3 install setuptools_rust` 
seemed to do the trick (I also ran apt-get upgrade to upgrade all the other packages in the background.

Compiling all the files needed when I ran
`sudo make file`. 
took the best part of two hours on the pi zero, but got there eventually.  It still spits some warnings about deprecated bits inside cryptography, but doesn't seem to stop TWCManager running now its installed.

KirkKirk, 2021-08-13:
that's the same in Docker as well. After install, I use  **sudo chown -R pi:pi /etc/twcmanager**

All is working as expected but today I noticed that the minimum charge is locked to 5A and doesn't change to the value specified in **"minAmpsPerTWC": 10,**
Is it a bug or do I have to do something else? 

[image]

drexcia, 2021-08-14:
Getting some strange charge behaviour here too, I've hit the minimum 12amp charge threshold earlier today and it was charging but it appears to keep charging after that drops. Mine also says 5 amps but I suspect that's a bug?

[image]

[image]

soso1999, 2021-08-14:
Same behavior here. Offering 6 amp all the time!

MikeBishop, 2021-08-16:
That looks like it's failing to connect to the Tesla API.  Have you verified that it has working tokens?

jherby2k, 2021-08-16:
guys please move the 6amps thing to a different thread.

Back to installation issues, I had the same rust error and had to do these:

sudo pip3 install setuptools_rust
sudo apt-get install build-essential libssl-dev libffi-dev python3-dev cargo

just doing the former complained about an out-of-date rust. Not sure what was specifically required but the second command brought down whatever it was.

I'll try docker next time, but it doesn't look like its updated for 1.2.3 yet.

drexcia, 2021-08-16:
The Docker installation isn't completely without its issues but definitely an easier install process. I'm running 1.2.3 using the latest image steps under the doco.

contactmike1, 2021-08-18:
> That looks like it's failing to connect to the Tesla API. Have you verified that it has working tokens?

This appears to be the case. After entering the credentials, TWCManager doesn't ask for api information again and acts like it has the token, but the screen output says

`21:49:32 TeslaAPI   20 Could not parse JSON result from https://owner-api.teslamotors.com/api/1/vehicles`

I know it's not my credentials because I deleted the entire TWCManager folder, and started over from git clone. I got the same result after entering my tesla credentials again on another new installation. Just for fun I I grabbed the token and refresh token from my MMM-Powerwall files, and imported that one, but that also didn't work.

closed 2022-02-15 by ngardiner