ngardiner/TWCManager #359: No consumption value from Enphase EMS Interface

jclendon, 2021-08-11:
Hi,

I'm using the Enphase EMS Interface with the cloud based API and I'm getting the generation data but the consumption data is just showing zero. Should the Enphase EMS Interface provide consumption data too or is that not provided?

Thanks, James

drexcia, 2021-08-12:
Do you have the hardware and CT's installed? For my Fronius installation I had to get a Fronius Smart Meter installed otherwise I would have only been able to see generation. 

https://www4.enphase.com/en-au/support/what-consumption-monitoring

jclendon, 2021-08-12:
Yes I have the hardware installed and can see the consumption in the local web interface and on the Enlighten web interface. I had a play with sending a GET request to the cloud API and it looks like it is responding with data. Just wondering if the Enphase EMS fetches that data or just gets the production data.

alexeiw123, 2021-09-20:
> Hi,
> 
> I'm using the Enphase EMS Interface with the cloud based API and I'm getting the generation data but the consumption data is just showing zero. Should the Enphase EMS Interface provide consumption data too or is that not provided?
> 
> Thanks, James

Hi James,

Not really helpful for the exact issue but I'll just say that I'm using the local enphase API and it is working perfectly. I chose local because that way it will work even if the internet is down (and I expect it will also respond quickly). Have you looked in to using that instead until your cloud API issues are resolved?

jclendon, 2021-09-20:
> Hi James,
>
> Not really helpful for the exact issue but I'll just say that I'm using the local enphase API and it is working perfectly. I chose local because that way it will work even if the internet [...]

Thanks for the advice @alexeiw123. I was using the local API and had it working until Enphase pushed out a firmware update blocking it. There is a thread here... https://community.enphase.com/s/question/0D53m00006ySLuR/unimpressed-with-loss-of-local-api-connectivity-to-envoys

After about 3 months Enphase finally rolled back my firmware so it is working again now using the local API. My advice is to watch out for firmware updates. Version 7 blocked the local API. They rolled me back to version 5.

alexeiw123, 2021-09-20:
> Thanks for the advice @alexeiw123. I was using the local API and had it working until Enphase pushed out a firmware update blocking it. There is a thread here... [...]

Oh wow - that is some good info. I'm on firmware R5.0.55 - I might head in to installer toolkit and disable firmware updates as per the link you've provided.

closed 2022-02-15 by ngardiner