ngardiner/TWCManager #360: TWC doesn't stop charging.

soso1999, 2021-08-14:
After Update to v1.2.3 TWC doesn't stop charging any more, offering 6 amps all the time. 
I can stop the charging only over the Tesla-App setting chargelimit under the actual charge-value.

drexcia, 2021-08-16:
Definitely some strange behaviour. I just stopped my charge session manually because it was still going even after my solar had stopped producing anything decent. I only see this when a charge session is in progress

[image]

ngardiner, 2021-08-16:
This is after logging in to your Tesla account? It can only stop the vehicle charging if it has API access, unless you're using a different stop mechanism than "Tesla API"

drexcia, 2021-08-16:
Yeah I have set it up to stop from the Tesla API but for some reason it didn't stop today. (It stops most of the time)

jherby2k, 2021-08-19:
I'm having the same problem today after it worked for 2 days straight. Problem persists after restarting it. If the Tesla token had expired, I believe from past experience it would say "0 amps offered, 6 in use". Right now i'm getting 6 offered and 6 in use:

Aug 19 11:03:57 twcmanager python3[539]: 11:03:57 TWCManager 20 Green energy Generates 596W, Consumption 1876W (Charger Load 1478W, Other Load 397W)
Aug 19 11:03:57 twcmanager python3[539]: 11:03:57 TWCManager 20 Limiting charging to 2.48A - 1.81A = 0.67A.
Aug 19 11:03:57 twcmanager python3[539]: 11:03:57 TWCManager 20 Charge when above 6A (minAmpsPerTWC).

I disabled "greenEnergyFlexAmps" and "useFlexAmpsToStartCharge" to troubleshoot - same issue. Seems to be a bug in 1.2.3. Let me know how i can help troubleshoot.

jherby2k, 2021-08-19:
In the log i'm seeing this:

`Number of cars charging now: 1
Aug 19 11:42:13 twcmanager python3[1492]: 11:42:13 TWCSlave   10 desiredAmpsOffered TWC: 54 06 reduced from 0.94 to 0 with 1 cars charging and flex Amps of 0.0.
Aug 19 11:42:13 twcmanager python3[1492]: 11:42:13 TWCSlave   10 desiredAmpsOffered: 0 < minAmpsToOffer: 6 (flexAmps: 0.0)
Aug 19 11:42:13 twcmanager python3[1492]: 11:42:13 TWCSlave   10 desiredAmpsOffered TWC: 54 06 reduced to 0 from 0 because maxAmpsToDivideAmongSlaves 0.94 / numCarsCharging 1 < minAmpsToOffer 6
Aug 19 11:42:13 twcmanager python3[1492]: 11:42:13 TWCSlave   10 Don't stop TWC 54 06 yet.
Aug 19 11:42:13 twcmanager python3[1492]: 11:42:13 TWCSlave   10 set_last_amps_offered(TWCID=54 06, desiredAmpsOffered=6)
`

jherby2k, 2021-08-19:
Issue appears to be at line 946 in TWCSlave.py. I'm trying to understand the logic enough to fix it... seems like dampenChanges doesn't work quite right.

```python
if self.reportedAmpsActual > 0 and desiredAmpsOffered == 0 and dampenChanges:
            logger.debug("Don't stop TWC " + self.master.hex_str(self.TWCID) + " yet.")
            desiredAmpsOffered = self.minAmpsTWCSupports
```

MikeBishop, 2021-08-19:
Yes, that's a recent change.  So if you go up to line 702, the question is what's causing dampenChanges to be set when it shouldn't be....

        dampenChanges = (
            True
            if now
            - max(self.timeLastAmpsDesiredFlipped, self.timeLastAmpsOfferedChanged)
            < self.startStopDelay
            else False
        )

jherby2k, 2021-08-19:
My instance stopped charging finally just over 2 hours later. Solar surplus has never been high enough to charge today, but was non-zero (and still is).

MikeBishop, 2021-08-19:
The intent is that if the system is flip-flopping about whether to charge, it should just continue doing whatever it's doing (charge, don't charge) until the system stabilizes enough to pick one.  That avoids cycling the connector for a passing cloud, or the heating element in an oven.  I did reduce how much flex I had configured in parallel with that change.  However, if you never have enough surplus to charge, it should sit consistently at not offering anything.

It currently looks at whether it's zero or not, assuming that the existing logic will already have dropped anything too small to offer down to zero.  But maybe the right dividing line would instead be minAmpsToOffer.  Can you try the `divide_by_zero` branch and see if it behaves better in this situation?  I've changed it to look at >= minAmpsToOffer rather than >= zero.

jherby2k, 2021-08-19:
Sure, i'll try that. Setting startStopDelay to 0 should also fix it (but break damping), right?

MikeBishop, 2021-08-19:
Yes, since the time since those events should never be negative.

jherby2k, 2021-08-19:
I'm running divide_by_zero now. Will let you know over the next few days - weather should be more of the same.

drexcia, 2021-08-19:
Giving this branch a go as well. Thank you

jherby2k, 2021-08-20:
Hmm seems to still be an issue. After leaving for a bit and then returning, after plugging in I'm back to 6 amps offered. Some solar generation during that time but never had a surplus even for a second. I'll log the values for timeLastAmpsDesiredFlipped and timeLastAmpsOfferedChanged and see what I can figure out.

KirkKirk, 2021-08-20:
Worked fine and today I had the same experience.
Deleted /etc/twcmanager/settings.json, change settings as suitable for me, enter the Tesla credentials for access and all come back to normal.
Maybe there are some changes in the Tesla API causing the issue.

MikeBishop, 2021-08-20:
@jherby2k, it's normal to have 6A offered, because the TWC can't *not* offer current unless TWCManager stops responding to it (red light), and if the car wants to charge it will take that minimal amount.  The expected behavior is for TWCManager to use the API to stop the car shortly thereafter, though.

jherby2k, 2021-08-20:
its not normal for the UI to say that 6 amps is offered though. it normally says 0 when the API is requesting a stop.

I figured out how to reproduce the issue! Hit charge now, then cancel it after it starts charging. Now you'll get 6 amps offered until its next disconnected and then reconnected.

Behavior is identical on both 1.2.3 and your divide_by_zero branch.

MikeBishop, 2021-08-21:
I think I found it....  Can you try the `no_oscillate` branch?  Basically, I can't consider when the lastAmpsOffered changed when I'm also changing the amps offered.  Instead, I should be looking at when the amount of power actually being drawn changed.

jherby2k, 2021-08-21:
Sure, it'll be a few hours tho.

jherby2k, 2021-08-24:
I believe it's fixed by no_oscillate. At least I haven't been able to reproduce the problem since upgrading. Thanks @MikeBishop! 🙏

soso1999, 2021-08-24:
> I believe it's fixed by no_oscillate. At least I haven't been able to reproduce the problem since upgrading. Thanks @MikeBishop! 🙏

What do we have to do? Wehre do we have to change this Parameter?

MikeBishop, 2021-08-24:
If the fix is good, I'll make a PR and merge it.  Then you'll just need to pull.

PR #364 (2021-08-24): ReportedAmpsChanged, not LastAmpsOfferedChanged
Fixes #360.
closed 2021-08-24 by MikeBishop