ngardiner/TWCManager #361: TWCManager Data Stops at 6am

drexcia, 2021-08-14:
Hi All, 

I've noticed that my TWCManager data stops populating around 6am over the last couple of mornings so the only I data I see from TWCSlave from this time. (v.1.2.3 latest)

===
05:58:35 TWCManager 20 Slave TWC 6816: Delivered 6131 kWh, voltage per phase: (243, 0, 0).
05:58:51 TWCManager 20 Green energy Generates 0W, Consumption 974W (Charger Load 0W, Other Load 974W)
05:58:51 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
05:58:51 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
05:59:30 TWCManager 20 Green energy Generates 0W, Consumption 571W (Charger Load 0W, Other Load 571W)
05:59:30 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
05:59:30 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
05:59:36 TWCManager 20 Slave TWC 6816: Delivered 6131 kWh, voltage per phase: (243, 0, 0).
05:59:48 TWCManager 20 Green energy Generates 0W, Consumption 575W (Charger Load 0W, Other Load 575W)
05:59:48 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
05:59:48 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
06:00:23 TWCManager 20 Green energy Generates 0W, Consumption 658W (Charger Load 0W, Other Load 658W)
06:00:23 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
06:00:23 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
06:00:31 TWCSlave   20 SHB 6816: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:00:31 TWCSlave   20 Slave power for TWCID 6816, status: 9
06:00:37 TWCManager 20 Slave TWC 6816: Delivered 6131 kWh, voltage per phase: (244, 0, 0).
06:00:49 TWCManager 20 Green energy Generates 0W, Consumption 581W (Charger Load 0W, Other Load 581W)
06:00:49 TWCManager 20 Limiting charging to 0.00A - 0.00A = 0.00A.
06:00:49 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
06:10:31 TWCSlave   20 SHB 6816: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:10:31 TWCSlave   20 Slave power for TWCID 6816, status: 9
06:20:31 TWCSlave   20 SHB 6816: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:20:31 TWCSlave   20 Slave power for TWCID 6816, status: 9
06:30:32 TWCSlave   20 SHB 6816: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:30:32 TWCSlave   20 Slave power for TWCID 6816, status: 9
===

A down/up on the docker service fixes it up (unsure if that sysv_ipc error is a concern or not?)

===
06:37:03 TWCManager 40 ImportError: No module named 'sysv_ipc', when importing Control.WebIPCControl, not using Control.WebIPCControl
06:37:04 HTTPContro 20 Serving at port: 8080
06:37:04 TWCManager 20 TWC Manager starting as fake Master with id 7777 and sign 77
06:37:06 TWCManager 20 32.00 amp slave TWC 6816 is ready to link.  Sign: 91
06:37:06 TWCManager 20 Set slave TWC 6816 protocolVersion to 2, minAmpsTWCSupports to 6.
06:37:06 TWCManager 20 32.00 amp slave TWC 6816 is ready to link.  Sign: 91
06:37:07 TWCManager 20 32.00 amp slave TWC 6816 is ready to link.  Sign: 91
06:37:07 TWCManager 20 32.00 amp slave TWC 6816 is ready to link.  Sign: 91
06:37:14 TWCManager 20 32.00 amp slave TWC 6816 is ready to link.  Sign: 91
06:37:16 Policy     20 New policy selected; changing to Track Green Energy
06:37:16 TWCSlave   20 SHB 6816: 08 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:37:16 TWCSlave   20 Slave power for TWCID 6816, status: 8
06:37:16 TWCManager 20 Slave TWC 6816: Delivered 6131 kWh, voltage per phase: (247, 0, 0).
06:37:20 TWCSlave   20 SHB 6816: 09 00.00/06.00A 0000 0000  M: 09 00.00/00.00A 0000 0000
06:37:20 TWCSlave   20 Slave power for TWCID 6816, status: 9
06:37:34 TWCManager 20 Green energy Generates 50W, Consumption 600W (Charger Load 0W, Other Load 600W)
06:37:34 TWCManager 20 Limiting charging to 0.20A - 0.20A = 0.00A.
06:37:34 TWCManager 20 Charge when above 12A (minAmpsPerTWC).
===

drexcia, 2021-08-17:
Had this happen again this morning but later so have renamed it slightly. 

My car is still charging at 6A despite not being offered anything

[image]

The logs show that my Consumption/Generation data seems to drop off at 0735. The charge start/stop/start is when I have returned home and plugged the car in. It seems to start up a 6A charge each time I plug it in even though the conditions don't meet. 

[image]

MikeBishop, 2021-08-18:
That part is expected -- the TWC is always offering at least 6A (US version) and if the car wants to charge, it will accept that 6A.  The reaction from TWCManager should be to stop the car charging after a moment or two.  If that's not happening, that's the thing to diagnose.

Is TWCManager successfully sending start/stop commands to your car?  If you go back to the older web UI, does the Stop All Charging button work?

drexcia, 2021-08-19:
Yeah it does appear to be. You can see at 7:55 where I have plugged the car in and it starts/stops/starts and just keeps going at 6A. The cancel charge now is greyed out on my screen but I did send an additional Charge Now command with 32A (which you can see) and I get the option to stop this in the GUI but you can see in the logs that the policy changes but no stop command actually shows and I had to this via the Tesla app.

There's also no consumption information coming through and no messages from the TWCManager service. 

[image]

When I do a down and up on the service you see that things start to change, I start getting TWCManager information messages again including my generation and consumption. 

[image]

I do manage to get the car to start charging but not to stop (even after a restart of the service). 

Is there a way to re-enter my Tesla creds? I've only done this the once when I first set it up.

MikeBishop, 2021-08-19:
It's almost like something's hanging the background thread, but you'd need to see where it stops to get a better clue for that.  Surest way to forcibly remove the token is to delete `carApiBearerToken` and `carApiRefreshToken` from `/etc/twcmanager/settings.json`.  There might be a way through settings, possibly by supplying invalid token(s) there, but I haven't tested what that does.

ngardiner, 2022-02-15:
Is this still an issue? Please reopen if so.

closed 2022-02-15 by ngardiner