ngardiner/TWCManager #362: error code 139

PJMikols, 2021-08-17:
Raspberry pi Zero W
Fresh install
Docker
Docker-compose - 1.29.2
when i first launched the container , it said it didn't have USB0, so I Plugged that in and then it exited with code 139

i have had this pi work before , but decided to move over to docker on this pi. 

`pi@twcmanager:~/twcmanager $ sudo docker-compose -f docker-compose.yml up
Recreating twcmanager ... done
Attaching to twcmanager
pi@twcmanager:~/twcmanager $`

`CONTAINER ID        IMAGE                           COMMAND                  CREATED              STATUS                           PORTS                              NAMES

6e8f9adedfff        twcmanager/twcmanager:latest    "./entrypoint.sh /us…"   About a minute ago   Restarting (139) 5 seconds ago                                      twcmanager

`

this feels very much the same as: #354 , but it appears they got theirs going. 

My only known change to the instructions is adding "container_name" to the docker-compose.yml

contactmike1, 2021-08-17:
I had a similar issue with a pi zero fresh install.   I had to manually install pip3, which is not included in the instructions.  After progressing through the steps, there are several warnings about certain scripts not being on PATH. I added directory it was complaining about to $PATH.  (/home/pi/.local/bin) However, _sudo docker-compose -f docker-compose.yml up_ would fail as command not found.  If I run _docker-compose -f docker-compose.yml up_ instead, it would attempt to start, but everything would fail as not having permission. 

To make more progress, i ran _sudo apt install docker-compose_.   This solved the sudo issues and at least let me run the command to start TWCManager, but it fails right away with code 139. 

```
pi@TWCManager:~ $ sudo docker-compose -f docker-compose.yml up
pi_twcmanager_1 is up-to-date
Attaching to pi_twcmanager_1
pi_twcmanager_1 exited with code 139
```
I am trying the manual one now.

drexcia, 2021-08-17:
Are you sure you aren't running an old version of docker-compose? If you run docker-compose --version it will show you. I was running 1.21 when I used apt-get. I did the following to get rid of the old version and install the new one. 

sudo rm /usr/local/bin/docker-compose
sudo apt-remove docker-compose
sudo pip3 install docker-compose
docker-compose --version (should show 1.29)

contactmike1, 2021-08-17:
> Are you sure you aren't running an old version of docker-compose? If you run docker-compose --version it will show you. I was running 1.21 when I used apt-get. I did the following to get rid of the [...]

It looks right to me.  The install was on a brand new image of rasbian lite.  Should I remove docker-compose and reinstall it anyway?

[image]

drexcia, 2021-08-17:
Not sure if that will help to be honest, that version is identical to what I am running. Not sure sorry :|

PJMikols, 2021-08-21:
after reviewing the comments, i went back and did a fresh install. I thought i recalled a few "errors" like PATH during the initial install, so i tried again. 

Rasp Pi OS Light 05-21-2021
TWCmanager 1.2.3 (from Docker compose)

ran a few of the extra installs for Rust, numpy/scipy, install tools etc. and confirmed i had no errors on install. 
installed pip3
upgraded the kernel, 
did confirm Docker-compose is 1.29.2
and still failing and restarting

I did load Portainer on it from docker compose and it is working just fine. so i'm pretty sure dock itself is fine. I'm wondering if its the USB interface to the dongle. 

i did see a folder "/etc/twcmanager" created, but it was empty. for luck i threw the config file in there and "chown -R pi:pi /etc/twcmanager "

what is interesting to me, it doesn't even start a log in Docker. 

I'm going to go try the manual install to get this going. (i still prefer docker)

PJMikols, 2021-08-21:
other interesting thing of note: 

i was having some full RaspiPi Zero W kernel penics early on, so i adjusted the config file per this youtube video and it was much stabler. FYI. 

https://www.youtube.com/watch?v=zp1Eq_pY0Rw

might mean it's a Card issue for me.

closed 2022-02-15 by ngardiner