ngardiner/TWCManager #369: Communication Issues

drexcia, 2021-09-12:
Hi All. I have an issue when I plug the charger in that it doesn't seem to handshake correctly (not sure if that's the right term). I generally receive one of the following errors:

Unable to AC Charge - Unplug and retry 
Charging equipment communication lost 

Generally by the 3rd or 4th time of pulling out the charger and plugging it back in it works and I hear the click and the charge of the initial 6A. 

I've only experienced this issue since using TWCManager and the car chargers ok at Superchargers. 

The RS485 wiring looks good (was done by an electrician). Any ideas?

VIDGuide, 2021-09-12:
Interestingly, I've seen this too, but kind of just gotten used to the unplug and re-plug. It only happens when the charger is in a controlled "stop" time. If the charger has sufficient solar, or is scheduled "on", then it doesn't behave like this. I've also disabled the slave mode to confirm as you mention that it doesn't do it when not controlled. It's something to do with being in that "stop" state when the car connects. 

If it connects and goes to "Not charging" right away, it will never start, even if the schedule becomes enabled, and you try to start a charge in the app. (The app will show as connected, not charging. Pressing Start Charging returns a "Charging already in progress" error, but the screen remains on "not charging". The only solution from here I have found is to disconnect and reconnect. 

When connecting in an "off" state, the only way to know it's working is exactly as you describe, the car needs to go to Starting Charging (6A) and then let TWCManager stop the charge. Then it will stay connected/idle ready to charge automatically when an "on" state resumes.

ngardiner, 2022-02-15:
TWCManager can't tell the TWC to start or stop charging unfortunately (well, it can, but in such an abrupt way that we don't use it because it causes the car not to start charging again), all we do is get a signal once charging begins that a vehicle is charging, at which point if we don't want it to we then have to use the API to tell it not to.

So there isn't really a stop charging message sent to a TWC, unless you specifically set the stop mode to use that command or you're using the access control function which allows VINs to be blocked, normally we only tell the TWC the rate we want to charge at. The stop command is sent to the vehicle, but only after the TWC tells us that there's a vehicle charging.

It's interesting that you find the issue goes away after disconnecting the TWCManager, but that could only really mean it is the existence of a master that causes it, not that it's to do with us having the TWC in any given state, as we can't control the state of TWCs. 

It's possible that there's a communication issue there between your TWCManager and the TWC over the Rs485 bus - are you seeing any red flashing LED on the TWC at any point?

ngardiner, 2022-08-26:
Closing some stale issues, please feel free to re-open if still ongoing.

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop