ngardiner/TWCManager #370: Not charging when Unscheduled Charging is active

bikeymouse, 2021-09-13:
I'm using the "latest" Docker version of TWC (version "1.24" build from 31/8/2021). 

In settings I have configured the "Non Scheduled Power Action" to "Charge At Specified Non Scheduled Charge Rate" and set the rate at "16Amps".

However, although the main screen shows that the policy is indeed "Fixed Rate/Amps" the car does not charge and is not offered any amps ("0" offered/used).

Is that a bug or am I missing something?

[image]

[image]

FYI: the  "Charge Now" button works as expected, the car does get offered the proper 16Amps and starts charging. Also solar tracking works.

VIDGuide, 2021-09-13:
Can you show us the "Schedule" tab? Given this is technically "non-scheduled" behaviour, it may be affected by the schedule itself. (not that I believe you can schedule "0"..)

bikeymouse, 2021-09-14:
Sure, here you have it. No strange settings here I think:

[image]

While I'm at it, also the policy tab:

[image]

bikeymouse, 2021-09-14:
Oh, wait, while I was checking again, suddenly 8 Amps is being offered... It looks like it is now starting to track-solar (despite the 16A Fixed/Rate Amps setting):
[image]

hopfi2k, 2021-09-14:
I have made several major improvements to address these issues some weeks ago in a pull request...waiting for @ngardiner to merge ;-)

bikeymouse, 2021-09-15:
Thanks! And good, because it looks like the logic is a bit messed up. Today, when I plugged my car in again after a drive it suddenly started charging, as configured, at 16Amps. So hope these improvement are merged soon :-)

ngardiner, 2021-09-19:
I see your PR for the UI changes @hopfi2k and while they're much appreciated and I intend to merge them into the next dev release, in what way would they possibly be related to the core functionality?

bikeymouse, 2021-10-30:
I have just updated to the "latest" Docker version (of 29/10/2021), which I guess is the latest dev-release, and the strange behavior is still here.

After more experimenting 'm actually now even more confused how things (should) work.

I'm wondering: what do I have to do if I basically don't want solar tracking or scheduled charging (because I wan't the car to decide when to start/stop charging, during the night)?

I thought I could do this by setting the "Non-Scheduled power action:" to "Do not charge".  The Home Screen then indeed shows that the current policy is "Don't charge" but there is 0 Amps offered...? 
The subtitle in the Homesceen then indicates "Doesn't charge a vehicle when plugged in unless manually started (e.g. in the Tesla App)". But that won't happen of course if 0 Amps are offered.

So how to specify the Amps to offer when the "Non-Scheduled power action:" is set "Do not charge"?
("Non-scheduled power charge rate"  only works in combination with "Charge at specified no scheduled charge rate")

Even more strange, when the homescreen suggest that the "Don't charge". policy is active, the log says that it is tracking green energy??

```
17:31:06 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
17:31:06 ⛽ Policy   13 Policy conditions were not matched.
17:31:06 ⛽ Policy   13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
17:31:06 ⛽ Policy   13 Policy conditions were not matched.
17:31:06 ⛽ Policy   13 Evaluating Policy match (tm_hour [17]), condition (gte), value (6)
17:31:06 ⛽ Policy   13 Evaluating Policy match (tm_hour [17]), condition (lt), value (20)
17:31:06 ⛽ Policy   13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [9]), condition (lte), value (17)
17:31:06 ⛽ Policy   14 All policy conditions have matched. Policy chosen is Track Green Energy

```
P.s. I have deselected all days in the Scheduled Charging tab.

bikeymouse, 2021-11-14:
Does anybody know an elegant way to have TWC configured in such a way so that the car can decide when to charge at a rate I can configure?

Currently this is not possible as both Scheduled Charging and "Charge Now" use the Tesla API to start the car with charging and keep doing that, even if it is stopped. So I ad to removed the API-key from the configuration to prevent that happening...

And linked to this: how could I then use the API to change te maxAmps-setting dynamically to be able to use TWC for load balancing (or actually, protecting  the main fuse of my house)

bikeymouse, 2022-02-22:
Bump:
Any suggestions how I can configure TWCManager to basically offer a fixed charge rate that can be adjusted using the API?

MikeBishop, 2022-02-22:
Wouldn't setting the Non-Scheduled Charging rate to the max value / a value you select be what you're asking for?  Then just don't schedule anything and don't use solar tracking.

bikeymouse, 2022-02-22:
Yes, that would be the idea, but unfortunately the charge rate for Non-Scheduled Charging cannot be controlled via the API as far as I can see.

MikeBishop, 2022-02-22:
Looks like something that `/api/setSetting` could handle -- POST something like '{"setting": "nonScheduledAmpsMax", "value": 48}`.

bikeymouse, 2022-02-23:
Thanks for the tip. Didn't know I could do this, that is not documented.

I have tried it now, and it seems to work. Sort of. 

I can see that if I send  {"setting": "nonScheduledAmpsMax", "value": 13} 
the GUI indeed shows that it is changed to 13Amps:

[image]

But the GUI and the /api/status still reports that 16Amps is offered:

[image]

Result of the /api/status also shows:

- currentPolicy: Non Scheduled Charging" 
- but still maxAmpsToDivideAmongSlaves 16A

`
{
   "carsCharging":0,
   "chargerLoadWatts":"0.00",
   "chargerLoadAmps":[
      "0.00"
   ],
   "currentPolicy":"Non Scheduled Charging",
   "maxAmpsToDivideAmongSlaves":"16.00",
   "currentServerTime":"2022-02-23, 12:32&nbsp;|&nbsp;",
   "consumptionAmps":"0.00",
   "consumptionWatts":"0.00",
   "generationAmps":"0.00",
   "generationWatts":"0.00",
   "isGreenPolicy":"No",
   "scheduledChargingStartHour":0,
   "scheduledChargingFlexStart":0,
   "scheduledChargingEndHour":1,
   "ScheduledCharging":{
      "enabled":true,
      "amps":16,
      "startingMinute":0,
      "endingMinute":60,
      "monday":true,
      "tuesday":true,
      "wednesday":true,
      "thursday":true,
      "friday":true,
      "saturday":true,
      "sunday":true,
      "flexStartEnabled":0,
      "flexStartingMinute":0,
      "flexEndingMinute":60,
      "flexMonday":true,
      "flexTuesday":true,
      "flexWednesday":true,
      "flexThursday":true,
      "flexFriday":true,
      "flexSaturday":true,
      "flexSunday":true,
      "flexBatterySize":100
   }
}
`
Sending a /api/saveSetting does not make a difference.

So what is wrong here?

MikeBishop, 2022-02-24:
A good question.  From a cursory read of the code, I don't see why it would get stuck on the previous value; there's not anything obvious that should cache the value.  TWCManager is pretty squarely built around the idea that the selected policy could change at any time, though this is unusual in that a non-green policy is changing in-memory.

Does it update properly if you change it from the UI?

bikeymouse, 2022-02-24:
Thanks for checking. No it doesn’t work as well when changed from the UI.

closed 2026-04-25 by ngardiner