ngardiner/TWCManager #371: Tesla auth issue

duane23, 2021-09-15:
Just getting established with the current main branch of TWCManager. Trying to authenticate with Tesla as follows;
 - Tesla account setup with MFA
 - Tried this link (https://registration.teslatasks.com/generateTokens) but it eventually times out with no error or status
 - Tried the Refresh and Access Tokens produced by the AuthAppForTesla but I get an error response in the logs of 'unsupported SSO app' - extract as follows:
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:51,705 - TeslaAPI   13 Entering car_api_available - next step is to query Tesla API
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:51,719 - urllib3.co 10 Starting new HTTPS connection (1): owner-api.teslamotors.com:443
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,345 - urllib3.co 10 https://owner-api.teslamotors.com:443 "GET /api/1/vehicles HTTP/1.1" 401 31
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,357 - TeslaAPI   13 Car API cmd vehicles <Response [401]>
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,360 - TeslaAPI   10 Car API vehicle list{'error': 'unsupported SSO app'}
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,361 - TeslaAPI   19 ERROR: Can't get list of vehicles via Tesla car API.  Will try again in 1 minutes.
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,362 - TeslaAPI   13 updateCarApiLastErrorTime() called due to Tesla API Error. Updating timestamp from 1631499600.143021 to 1631499712.3620083
/etc/twcmanager/log/logfile.2021-09-13_12:2021-09-13 12:21:52,363 - TeslaAPI   13 applyChargeLimit return because car_api_available() == False

Any suggestions for where to go from here?

Cheers,
Duane.

shanesturrock, 2021-11-16:
I've been having similar issues but finally managed to get it going. First off, I'm using the latest Docker container v1.2.4 and made sure I chowned the /etc/twcmanager directory to pi:pi as that's the account the container is running under and without that it can't write to the configs. Once I started it with sudo docker-compose -f docker-compose.yml up I stopped again with CTRL-C and edited the config.json to bump the logging so I could see the debug info (set it to 10, you'll get lots but that helped me figure out what was happening) and then start the server again with the docker command.

Once going, I went into the settings and pasted in both the Access Token and Refresh Token from Auth for Tesla (previously I had only provided the Refresh Token and that clearly didn't work, it seems to need both) and I could see the TeslaAPI recognise my car and start returning true. The car was plugged in so I manually started charging even though my solar isn't generating enough and shortly after I saw the TelsaAPI Car API cmd charge_stop get sent and it stopped.

ngardiner, 2022-02-25:
Closing this off as we have a new auth mechanism now which I've been testing without issues - if you were having any issues and are on the development version you can do a git pull to get the latest update, otherwise v1.2.6 will be out soon with this and potentially another fix around API auth and tokens

closed 2022-02-25 by ngardiner