ngardiner/TWCManager #372: use TWCManager for logging to HASS, MQTT when in fakeMaster=2

nean-and-i, 2021-09-17:
Hi,

Have a quite different use case and wondering if I could use this fork without solar?
I noticed that logging modules HASS,MQTT, etc don't work in the "passive" scenario. ( -> "fakeMaster": 2 and TWC in "5" - not slave) 
The data is flying by, lifetime kWh is being captured, and it would be great to have the data available in HA or MQTT broker for further use.
No need to actively control anything, I'd be interested if there is a chance to use it as data collector/aggregator or if it is designed/dedicated for the solar use case only?

thanks!

nean-and-i, 2022-09-04:
Hi,

I guess i found the correct/working place for pushing status update to MQTT/HASS in passiv scenario.

please check PR #469 "Status updates for MQTT/HASS in fakeMaster=2 mode ", it works now, but maybe some cleanup is required

thanks

MikeBishop, 2026-10-26:
Fixed by your PR #469 (v1.3.3): status updates are published to MQTT/HASS in `fakeMaster=2` mode. Closing.

closed 2026-10-26 by MikeBishop