ngardiner/TWCManager #374: API Support for Charge Rate

MikeBishop, 2021-09-25:
According to a recent leak, car firmware 2021.36+ supports remotely adjusting the charge rate via the API. This means TWCManager could support cars connected by UMC, Gen3 TWC, etc. once the specific API endpoint is known.

mikey4321, 2021-09-26:
Finally :)

[image]

MikeBishop, 2021-09-27:
So I was thinking about architecture.  I think the way to manifest this is probably to support a collection of EVSEs which can be of different types:

- API vehicles (filter by location; only control their charging activity if they're at home)
- RS-232 TWCs (almost certainly becomes a legacy option RSN)
- (Perhaps) Network-connected TWCs (if we ever figure out how to support Gen3 and want to bother)
  - This might open the door to supporting other Wi-Fi connected EVSEs in the future, also if we want to bother.

There also needs to be a dedupe step where a vehicle found in the API isn't treated as a separate charging instance if it's connected to a TWC we already control.  If the TWC reports VINs, that's easy.  If not, there's a path that offloads it to the user (they should disable vehicles-as-EVSEs in config) or a path that attempts to be smart (if we change the charge current, does the car see that change?).  We'll need to pick one.

We probably need to support a maximum charge rate for the home overall and perhaps for the connected TWCs specifically.  I'd wait to see if there's demand for load-sharing within API vehicles.

Of course, we still need someone to extract the API endpoint from the APK.  That's not something I know how to do offhand, but I bet it won't take long for someone to identify it.

MikeBishop, 2021-09-27:
Oh, and the big bugaboo here -- the fakeMaster / process message loop will probably now live in a module.  There is *so much* logic that triggers on TWC status messages coming in which now still needs to run in the absence of an RS-232 interface.  That'll be... *fun.*

ngardiner, 2021-10-01:
So the endpoint is identified already, it is:

```api/1/vehicles/{vehicle_id}/command/set_charging_amps```

I don't know any more about it at this point, but am feeling brave enough to do some testing over the weekend.

MikeBishop, 2021-10-05:
I haven't seen any vehicles get 2021.36 yet, though more power to you if you have.

ngardiner, 2021-10-05:
No, however this was about testing the API endpoint more than the vehicle functionality, and the outcome was a 404, so either Tesla returns 404 for vehicle firmwares pre 2021.36 or perhaps it hasn't been enabled yet on account of the lack of vehicles on this release.

We'll find out soon enough.

surfarn, 2021-11-05:
Can the I use the Tesla API to controll the Charge rate without an RS-485 connected? I would like to use load balancing so that fuses do not blow when the spa is heating when the car is charging.

RichieB2B, 2021-11-05:
In general, yes you can. I believe TWCManager now also supports this.

surfarn, 2021-11-06:
I think something is missing to run it without the RS-485.  Keep getting this error even when after removing the USB-port in the compose file. 

`Traceback (most recent call last):
  File "TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/usr/src/TWCManager/lib/TWCManager/TWCManager.py", line 553, in <module>
    modinstance = modclassref(master)
  File "/usr/src/TWCManager/lib/TWCManager/Interface/RS485.py", line 52, in __init__
    self.ser = self.serial.serial_for_url(self.port, self.baud, timeout=0)
  File "/usr/local/lib/python3.7/dist-packages/pyserial-3.5-py3.7.egg/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.7/dist-packages/pyserial-3.5-py3.7.egg/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
`

RichieB2B, 2021-11-06:
You can try to remove the line with "Interface.RS485" from lib/TWCManager/TWCManager.py [line 82](https://github.com/ngardiner/TWCManager/blob/main/lib/TWCManager/TWCManager.py#L82). The should prevent the error of trying to initialise the RS485 port.

surfarn, 2021-11-08:
Thanks managed to start without the error message now

rickhuizinga, 2022-01-05:
If using the API to change the charge rate, is the full functionality supported without plugging into the RS485 port?  Specifically:will it stop charging if the car is plugged in at night when tracking green energy reports 0W produced?

jherby2k, 2022-01-25:
Anyone else seeing crazy behavior with this API? 

Not TWCManager related, just the API itself. Even using the mobile app - if i set the current from like 32 to 7, it'll flip back to 32 a few seconds later. If i set it to 16, then 7, it'll flip back to 16, then back to 32. Like it rolls back the history or something. So strange.

RichieB2B, 2022-01-25:
No, I am using the API daily for loadbalancing (not using TWCManager) and it is very stable for me.

jherby2k, 2022-01-25:
Thanks - i'm gonna go pull the TWCManager out of there and see if that's causing issues. You using a V2 charger as well?

RichieB2B, 2022-01-25:
Yes, I have a TWC V3 using it with 3 phases at 230V.

jherby2k, 2022-01-25:
Removed the TWCManager and its stopped doing it. Got a moderately complex Home Assistant automation now handling it all via API. It was a good hard hack while it lasted *sniff*.

bikeymouse, 2022-02-06:
> Removed the TWCManager hardware and its stopped doing it. Got a moderately complex Home Assistant automation now handling it all via API. It was a good hard hack while it lasted... _sniff_.

Hi, I’m very interested in that. I’m also wrestling with a combination of Home Assistant / Node-Red and TWC-manager to do loadbalancing en solar tracking. Also seeing TWC having strange behavior (in that combination).
Would you be willing to share your set-up/code?  (Perhaps a bit off-topic here, but it could also be in the HASS forum.)

surfarn, 2022-02-08:
I would also be very interested to see how you did this in HASS.

alexeiw123, 2022-08-28:
> I do plan a second option called API Assisted Charge Rate Adjustment

Oh this sounds an ideal use of the API. Being able to use 5A without API priority would be great.

MikeBishop, 2026-10-26:
v1.4.0 adds a Tesla API EVSE controller (6f075db0): a Tesla at home that isn't being served by a TWC becomes a controllable endpoint, and its charge rate is set via the Fleet API. That covers the Mobile Connector case, though it hasn't been tested on a UMC specifically. Closing as done; please re-open if it doesn't work for you.

closed 2026-10-26 by MikeBishop