ngardiner/TWCManager #375: The values in Mqtt topics doesn't match with the dashboard

KirkKirk, 2021-10-03:
Hi @ngardiner,
I believe that rare cases are using 2 or more chargers on one TWCManager so no reports about anomalies.
Recently, My home devices communicated with HASS I move to Mqtt and so I move the TWCManager as well.
The only difference is that:
- I was expecting a much faster refresh on the HA dashboard but it is the same as via API.
- the MQTT information arriving in the MQTT topics is a mixup between two chargers
- when no charging, the value of TWC/XXXX/ampsMax doesn't follow the real value for ampsMax (correct on the dashboard)
- the topic TWC/xxxx/ampsInUse doesn't refresh when the dashboard does (late around 40sec.) and when charging stops it stays frozen on the last value (6.xx A)
About this issue, I didn't know which log data to provide here, so if need it, let me know what and how to do it.
Cheers,
Kirk

ngardiner, 2021-10-03:
Hi @KirkKirk,

Thanks for reporting this. There should be some logs similar to the below, if you increase your debugLevel to 8:

22:41:24 📊 MQTT     13 Publishing MQTT Topic TWC/config/minAmpsPerTWC (value is 6)
22:41:24 📊 MQTT     13 Publishing MQTT Topic TWC/all/maxAmpsForSlaves (value is 0)
22:41:38 📊 MQTT     13 Publishing MQTT Topic TWC/XXXX/carsCharging (value is 0)

If you could capture these plus other state messages around them (such as the green energy, energy offered, etc logs) I could get a feel for where the values might be incorrect.

KirkKirk, 2021-10-04:
Please find attached log from tonight's manually triggered charging. You will see that when charging stops but the value on _ampsInUse_ remains 5.24 and the refresh is really slow. 
[image]

I will replicate the same situation tomorrow when the other car will charging on the second charger.
[logfile.txt](https://github.com/ngardiner/TWCManager/files/7277952/logfile.txt)

ngardiner, 2021-10-05:
Got it, so that's the second part of the 4th issue in the list. I understand why that happens and will fix that in the next release, 1.2.4

In terms of the refresh rate, there are two different approaches here which explains why the refresh rates are different. The API is queried, so the data rate is controlled by the API client. Each time you query the API, you get point in time data.

Status modules are event driven. They change only when the data changes. That said, we may not be triggering on the correct event and if that's the case we can certainly adjust it.

I'll investigate point 3 as that should be something I can find quite easily in the code, but could you help to understand point 2 a bit better - which values/topics do you see getting mixed up between the TWCs?

KirkKirk, 2021-10-05:
Thanks for looking at this.
Sorry for point 2 as just finished experimenting and after more closely looking at the behaviour, I figured out that the values are not mixed up. Previously I got that impression because the values were changing slowly and sometimes the value from first charger matches on the second when the first one takes a different value  (and vv).
If you need more info, please let me know what scenario to perform.

KirkKirk, 2021-10-18:
Thanks @ngardiner. Work like a charm :)

Looking forward to the implementation of the sensor measurements receiving to be via MQTT as well.

closed 2021-10-18 by KirkKirk