ngardiner/TWCManager #385: Why is the Scheduled Charging policy status not visible in the Home Screen?

bikeymouse, 2021-11-08:
After being puzzled by the strange behavior of TWC I started testing with the API. 
Then I found out that while the getStatus API-endpoint nicely shows: 
- "currentPolicy": "Scheduled Charging"
- "maxAmpsToDivideAmongSlaves": "14.00"

I get a Homescreen that says that the current Policy is "Don't Charge" (although the 14Amps offered is correct):

[image]

This seems to be a design decision, as there is no label for Scheduled Charging.
Why is that (even in the old GUI)? I find that is very confusing.

MikeBishop, 2021-11-12:
For some reason, that UI element shows what the selected behavior is during Non-Scheduled Charging, not what the selected behavior is _now_.  Personally, I think showing the current state would make far more sense... in which case, Scheduled Charging would reflect a Fixed Rate.

bikeymouse, 2021-11-14:
> For some reason, that UI element shows what the selected behavior is during Non-Scheduled Charging, not what the selected behavior is _now_. Personally, I think showing the current state would make far more sense... in which case, Scheduled Charging would reflect a Fixed Rate.

Thanks for explaining, I - and probably many others -  would never have guessed this myself ;-) .

So I vouch for a request to change the Home Screen to indeed show the _current_ state, clearly showing  all possible states, including scheduled charging.

bikeymouse, 2022-04-04:
Great job! 
Unfortunately, I don't have time to build and test this right now (and not mess up my production environment). Also have found a way to work around the UI by using the API, but will be glad to test this if I can easily up/downgrade using another Docker-image.

bikeymouse, 2022-04-17:
Hi,
Thanks for the explanation! 

I finally had some time to test soI tried to make the image, but got this error during building:

`fatal: not a git repository (or any of the parent directories): .git`

So it seems it can't get you repository or branch although these seem to exist?

Also when I try to checkout the repository from the comnand line using line 3 from the docker file I get the same error:

```
$ git clone https://github.com/dtiefnig/TWCManager && git checkout dtiefnig-web-policy-update
Cloning into 'TWCManager...
remote: Enumerating objects: 8948, done.
remote: Counting objects: 100% (3830/3830), done.
remote: Compressing objects: 100% (1319/1319), done.
remote: Total 8948 (delta 2248), reused 3490 (delta 1998), pack-reused 5118
Receiving objects: 100% (8948/8948), 8.35 MiB | 23.17 MiB/s, done.
Resolving deltas: 100% (5191/5191), done.
fatal: not a git repository (or any of the parent directories): .git
```

dtiefnig, 2022-04-18:
Ah, my bad, you need to change directory for the git checkout:

`RUN cd /usr/src && git clone https://github.com/dtiefnig/TWCManager && cd TWCManager && git checkout dtiefnig-web-policy-update`

bikeymouse, 2022-04-18:
Hi Daniel,

Thanks, I got it working now! Look good, the interface now indeed reacts reflects the actual situation. Great job!

Still missing is that I can't see if the actual policy is "Scheduled Charging" or "Non Scheduled Charging" as these are both mapped to "Fixed Rate/Amps".  Also "Charge Now" is mapped to this, but a least you can indirectly conclude that this is active by the fact that the "Cancel Charge now" button can be pushed. 

This is all relevant, as I'm keep getting baffled  by the seemingly strange logic and priorities of TWC when setting times for Scheduled Charging, Tracking Solar and Non-Scheduled Charging. Luckily I can use the API to get the status, as this does say what the actual status is, but it would be helpful if this would show in the GUI as well.

Oh, one other thing I noticed: if the "Include ServerTime in Main UI" is set to yes, I now see the UTC-time in the GUI, not the local time like it used to be in the regular version. Something changed there too?

dtiefnig, 2022-04-18:
> Thanks, I got it working now! Look good, the interface now indeed reacts reflects the actual situation. Great job!

Great! You may have missed my latest update (40 minutes ago) which reworks background logic a little and gives more consistent results. Does not change front end functionality though. Would be great still if you could give it a try. Changes have been pushed to the same branch, so test procedure remains the same.

> Still missing is that I can't see if the actual policy is "Scheduled Charging" or "Non Scheduled Charging" as these are both mapped to "Fixed Rate/Amps". Also "Charge Now" is mapped to this, but a [...]

That's true, as the actual action is always the same. Technically there's not much difference. It might be helpful for you if the currently active policy name would be printed on the web page. Or are you looking for a different information?

> Oh, one other thing I noticed: if the "Include ServerTime in Main UI" is set to yes, I now see the UTC-time in the GUI, not the local time like it used to be in the regular version. Something changed there too?

Might be just a container config issue. I usually use something like the following to bring my containers to the local timezone:

```
    environment:
      TZ: 'Europe/Vienna'
```

Cheers,
Daniel

bikeymouse, 2022-04-18:
>Great! You may have missed my latest update (40 minutes ago) which reworks background logic a little and gives more consistent results.
>
I just rebuild the image and did a short test. Looks good! 
Will keep on using it and report if I find strange things.

> That's true, as the actual action is always the same. Technically there's not much difference. It might be helpful for you if the currently active policy name would be printed on the web page. Or are you looking for a different information?
> 
Yes, printing the current active policy would be very helpful!

>    Might be just a container config issue. I usually use something like the following to bring my containers to the local timezone:

    environment:
      TZ: 'Europe/Vienna'
>
I already had this in my docker-compose.yml
```
    volumes:
      - /etc/localtime:/etc/localtime:ro
```
That worked until now, but adding the TZ did do the job, so thanks!

PR #443 (2022-04-20): web: display active charging policy
Fixes #385

I'm passing a numeric charging action (consistent with the `nonScheduledAction` setting) based on our currently active charging policy to the template `Modern/main.html.j2`, which is then highlighting the corresponding action. Highlights are updated via `jsrefresh.html.j2` as it may change when user starts `Charge Now`, or when a policy change is initiated by one of its criteria (e.g. time based).

As far as I can see, a similar change is **not** required for the default theme, as the logic is different there.

Please note this includes a commit  [... cut]
closed 2022-04-20 by ngardiner