ngardiner/TWCManager #387: Please check the permissions on /etc/twcmanager/settings.json

ChutneyMary, 2021-11-14:
I've installed version 1.2.4. The TWCManager appears to be running but I can't save any Settings from the Save Settings tab.

I'm trying to save a manual Tesla API key override Access Token and Refresh Token. This results in a a message -

"We experienced an error trying to save your settings. This mea ns that while your current settings will be retained for the current session, th ey will not be saved if TWCManager exits. Please check the permissions on /etc/t wcmanager/settings.json and try saving your settings again."

I've used to WinSCP to check in the etc/ twcmanager folder. There is no settings.json file.

For the permissions issue, I've tried both -
sudo chown -R pi:pi /etc/twcmanager
chown -R pi:pi /etc/twcmanager

Any suggestions?

ChutneyMary, 2021-11-16:
@ngardiner - I've made progress after starting again and doing a re-install. For info, I'm setting up a new Raspberry Pi W in headless mode with Raspberry Pi OS Lite version 2021-10-30-raspios-bullseye-armhf-lite.img. I'm installing TWCManager 1.2.4 on a new instance of this installation.

I found a suggestion to run the following additional command as well as those specified on your install how-to -

sudo apt-get install python3-pip

That ran the install fine. The Settings.json now saves without the error message.

Sorry for the Linux numpty question, but now I need to change the ownership of /etc/twcmanager back to pi:pi whenever I want to edit the config.json (using WinSCP), then change the ownership back to twcmanager:twcmanager. Am I doing something wrong here?

What is the correct ownership of /etc/twcmanager going forward?

ngardiner, 2021-11-17:
Hi @ChutneyMary,

Are you running ```make install``` as part of the build steps?

This creates all of the necessary directories and applies permissions. 

If you want to be able to edit the configuration as the pi user you can use the following commands:

```
chown twcmanager:pi /etc/twcmanager/config.json
chmod g+w /etc/twcmanager/config.json
```

ChutneyMary, 2021-11-17:
Thanks. I'll give that a go. I followed the commands -

git clone https://github.com/ngardiner/TWCManager
cd TWCManager
git checkout main
make install

closed 2021-11-17 by ChutneyMary