ngardiner/TWCManager #388: [Question] Can I parallel plug multiple tesla charger to set the max current all at once

scarrier92, 2021-11-14:
In a condominium builiding if I want to lower all chargers to 12 Amp at certain time of the day for 50  chargers. 

Can I all wire them in parallele and adress them at the same time? Otherwise Can I relay them individually ( is there some handshake that needs the connection always stay alive? Was hopping not to need 50 raspberry pi and 50 485 t usb adapter ... ? or a 50 usb hub...

FYI I'm thinking of using the Gen 2 J1772 wall connector

Anything to suggest?

ChutneyMary, 2021-11-15:
As best I understand, the Tesla Wall Charger is designed for a maximum of 4 chargers daisy-chained on the RS485 bus. The Raspberry Pi acts as a fake Master device, which means three physical charging devices can then be controlled.

So you would need 17 Raspberry Pi's to control up to 51 Wall Chargers.

MikeBishop, 2021-11-15:
Technically, you could probably run multiple instances of TWCManager with multiple RS458 adapters connected to the same device.  However, I don't know that there's been much testing of parallel instances.

KirkKirk, 2021-11-16:
I am running 2x Gen2 WC on one TWCManager and in most scenarios, it's working fine when 2 cars are charging at the same time.
Not sure what is your idea, but 50 chargers with a single common point of failure, is something that I wouldn't do.  
I believe you have a solar system to satisfy >=384kW charging 50x32A on a single phase.

scarrier92, 2021-11-16:
@ChutneyMary I don't want to daisy chain them because none of them are sharing the same supply, but they are externally constrained by power consumption.

@KirkKirk  I actually have no solar at all.  when you say you have 2 gen 2, do you have a rs485 adapter for each of them or did you just plug the wire in parallel to both charger on 1 rs485 adapter so they both receive the exact sae message at the same time? 
 
(i'll round up numbers here)
We have a electrical supply that can deliver 2000 A. (50 x 40 A). and thats for 50 apartment who all have a parking spot. 

Since people want to add electrical charger to the same electrical infrastructure we want to lower car chargers at peak time 9morning and evening so we never go over 2000 A.  So if the whole building is consuming 1500 A at a specific time we want to lower evrybodys charger at 10 A for a short period of time. This makes more sens than cutting them off completely IMO.
In the first years when most people don't drive electric it might never kick in but soon enough it will start to kick in at 7AM and 6PM

The same system could also apply in a house. If you have a main panel that is 100 A (normal here in canada with electrical heating) and in cold weather when all heaters are on  you don't have enough power to supply heater  + oven + hot water + dish washer + tumble dryer... so at that same peak time instead of cutting off the supply to the car charger I would lower the car charger so the total consumption of the house is lower than 80 A  (80 % of 100 A ) if this still doesn't work I'd hav completely cut out the car charger. So this is almost the same use case as solar but our issue is house supply. Obviously were I live the supply has only delivered 100% renewable for the last 50 years and solar is not worth while because electricity is 0.05$/ kwh

I understand I might have to tweak the library a bit to do what I want to do but this is a use case I think will see poping in the next few years a lot.

MikeBishop, 2021-11-16:
So what you're really looking for is something kind of like #384, where rather than tracking available green energy, you want to cap the grid draw.  That actually doesn't require all that much -- if you have something that can measure current usage, you could just hard-code that 1.5kA of "solar" is available all the time, and TWCManager's normal Track Green Energy code will behave more or less as expected.

The fan-out to a large number of TWCs remains harder, but multiple independent instances of TWCManager looking at the same information should arrive at basically the same result -- each will control 3 spots and see the consumption of all the other spots as part of the usage it's reacting to.

MikeBishop, 2021-11-16:
Also, it's fine to daisy-chain even when they're not constrained -- you just configure TWCManager that the maximum for each individual TWC is (for example) 48A (on a 60A circuit) but the maximum across the three of them is 144A or greater.  Then the individual limits will control and they'll never be throttled for each others' use except to keep under the limit you've configured.

KirkKirk, 2021-11-28:
@scarrier92 both chargers are connected on 1x RS485 adapter.

ccutrer, 2022-03-30:
@ngardiner : can you think of a reason for a hard cap at 4 devices? I see that in the code it's currently capped to 3, but that's just an artificial limitation. Even if Tesla's master implementation only allows 4, I don't see why TWCManager couldn't allow more. I have a (non-TWC) RS-485 network with ~35 devices on it. It's only running at 4800 baud, and while it occasionally drops messages due to collisions, it's by and far ~99% reliable. Even if @scarrier92 could put 10 or 15 wall connectors per network, it would drastically simplify things for him. I myself have 3 TWCs (actually, I have a 4th NIB just-in-case), and I'm extremely happy to see TWCManager allow me to configure 80A per TWC, with 120A total (instead of 80A total), since mine are wired as separate 100A circuits, all from a 150A sub-panel.

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop