ngardiner/TWCManager #390: Config.json not used?

calleman, 2021-11-18:
This might be a pure user error but I installed your fork of TWCmanager in a clean updated raspios-bullseye environment on a RPi Zero, all according to the manual. I needed the cryptographic module to be able to compile without errors though. All fine after that exept that the supplied config.json was not read. To force it to read I changed the "/etc/twcmanager" path in TWCManager.py.

Will TWCManager skip to default settings if the config.json has any errors? Could that be it?

This is awsome, thanks!

ChutneyMary, 2021-11-18:
I've just attempted the same install on a RPi Zero W. I initially used the headless raspios-bullseye Lite variant. That threw errors that were fixed by running the command -

sudo apt-get install python3-pip

before running the **make install** command. I still had errors related to the cryptographic module but I didn't know what cryptographic module was required (any hints?).

There's two config.json files -

etc/twcmanager/config.json
home/pi/TWCManager/etc/twcmanager/config.json

As far as I'm aware, the first one is the one actively used.

I went through a second installation using raspios-bullseye Full (desktop version). That didn't appear to require the line -

sudo apt-get install python3-pip

At least is said it was already installed. Both installation environments appeared to work ok.

calleman, 2021-11-18:
Thank you! All is working good now!

closed 2021-11-18 by calleman